kizniche/Mycodo #786: Error 502: Bad Gateway Error after upgrade to 8.5.8

frodus17, 2020-07-09:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug
I upgraded from 8.5.7 to 8.5.8, get this error, even after reboot:

Error 502: Bad Gateway Error
Something bad happened but it's probably not your fault. Try refreshing the page to see if it will load. If that doesn't work, try rebooting your system, then try refreshing it again. If that still doesn't work, submit a new issue on GitHub with any usefull information that can help determine the cause of this issue.

### Versions:

 - Mycodo Version: 8.5.8
 - Raspberry Pi Version: 3+
 - Raspbian OS Version: Raspbian Buster

### Reproducibility

I upgraded from 8.5.7 to 8.5.8. This immediately happened. This has happened for me multiple times over the last year as well, which is causing me some concern, as I have to re-start setup/restore from scratch.

### Expected behavior

I expected the server to start and Mycodo to operate after an upgrade.

### Screenshots

n/a

### Additional context

n/a

kizniche, 2020-07-09:
I very rarely experience this with my Pis. Let's see what's preventing the frontend from starting. Can you run the following commands and copy the resulting error here?

```bash
sudo service mycodoflask stop
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

frodus17, 2020-07-09:
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/run/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2020-07-09 12:44:46 -0700] [1864] [INFO] Starting gunicorn 20.0.4
[2020-07-09 12:44:46 -0700] [1864] [INFO] Listening at: unix:/var/run/mycodoflask.sock (1864)
[2020-07-09 12:44:46 -0700] [1864] [INFO] Using worker: gthread
[2020-07-09 12:44:46 -0700] [1869] [INFO] Booting worker with pid: 1869
[2020-07-09 12:44:47 -0700] [1869] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
    super().init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
    from mycodo.mycodo_flask.app import create_app
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 19, in <module>
    from flask_talisman import Talisman
ModuleNotFoundError: No module named 'flask_talisman'
[2020-07-09 12:44:47 -0700] [1869] [INFO] Worker exiting (pid: 1869)
[2020-07-09 12:44:47 -0700] [1864] [INFO] Shutting down: Master
[2020-07-09 12:44:47 -0700] [1864] [INFO] Reason: Worker failed to boot.

kizniche, 2020-07-09:
It looks like flask_talisman wasn't installed during your upgrade for some reason. You should be able to get your frontend running again with the following commands.

```bash
sudo ~/Mycodo/env/bin/pip3 install flask_talisman
sudo service mycodoflask restart
```

kizniche, 2020-07-09:
I just had the idea that I should include an excerpt from the web log (which would include the error you just pasted) on the 502 Error page. Not sure why I didn't think of that sooner.

frodus17, 2020-07-09:
That appears to have worked Kyle. Thank you for being so quick.

The excerpt on the web log would be great if that was possible! 502 is pretty generic.

closed 2020-07-09 by frodus17