kizniche/Mycodo #787: Failed to start Mycodo server

Barnickel, 2020-07-14:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

After a recent update faild the mycodo server wasnt starting.
I freshly installed mycodo and im receiving an error

pi@raspberrypi:~ $ sudo /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py --debug
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 40, in <module>
    from Pyro5.api import serve
ModuleNotFoundError: No module named 'Pyro5'

### Versions:

 - Mycodo Version: 8.5.8
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Raspbian 10 Buster

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

I execute the install script and it errors with the error message above

### Expected behavior

### Screenshots

If applicable, add screenshots to help explain your problem.
[image]

### Additional context

Is there anything that should be added to make it easier to address this issue?

dandyfop, 2020-07-14:
I too have had this exact issue and been trying to troubleshoot for several days now.
Also running Mycodo 8.5.8, RPi 3B+, on Buster.

kizniche, 2020-07-14:
Lite or Desktop Raspberry Pi OS image?

kizniche, 2020-07-14:
What version did you upgrade from?

kizniche, 2020-07-14:
This comment has commands to rebuild your python virtualenv: https://github.com/kizniche/Mycodo/issues/776#issuecomment-636606371

After issuing those commands, restart the Pi with ```sudo shutdown now -r```

Barnickel, 2020-07-15:
@kizniche it is the Desktop Raspberry Pi OS Image
I tried upgradeing from 8.5.4 when the Update failed.
After that i tried a fresh mycodo install.

I tried the commands to rebuild the virtualenv. After the reboot i still get the same Error :(

[image]

Barnickel, 2020-07-15:
Here is the Error i found while rebuilding virtualenv

[image]

kizniche, 2020-07-15:
The Desktop image version has an incompatible software version installed ([issue discovered yesterday](https://kylegabriel.com/forum/general-discussion/mycodo-installer-script-on-fresh-default-installation-of-raspbian-buster-fails-to-detect-failure-of-necessary-modules-mycodo-daemon-unstartable)). Uninstall the cffi packages with the following command, then rebuild your virtualenv again with the command in the above comment I linked to.

```bash
sudo apt remove python-cffi-backend python3-cffi-backend
```

Barnickel, 2020-07-15:
Thank you Kyle this fixed the Problem :)
Server is up and running again.
Really appreciate your work, thank you for your Help

closed 2020-07-15 by Barnickel