kizniche/Mycodo #789: EZO Pump Calibration Error

kpslc, 2020-07-16:
Clicked on "Begin Calibration" and the following error occurred.

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 140, in setup_atlas_ezo_pump
    atlas_command = connect_atlas_ezo_pump(selected_output)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 99, in connect_atlas_ezo_pump
    i2c_address=int(str(selected_output.location), 16),
ValueError: invalid literal for int() with base 16: ''

Tried "Clear Calibration". It also displayed the same error:
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 115, in setup_atlas_ezo_pump
    atlas_command = connect_atlas_ezo_pump(selected_output)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 99, in connect_atlas_ezo_pump
    i2c_address=int(str(selected_output.location), 16),
ValueError: invalid literal for int() with base 16: ''

kizniche, 2020-07-16:
Is your pump properly configured and successfully dispenses volumes when instructed to do so from the Output page?

kpslc, 2020-07-16:
Yes. They also work from the terminal using 

~/Mycodo/env/bin/python ~/Mycodo/mycodo/devices/atlas_scientific_i2c.py

and from the mycodo setup/output menu. Using either terminal or the output page the pumps work perfectly.

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com <mailto:kevin.plumb@gmail.com>

kizniche, 2020-07-16:
That's for the feedback. I'll look into it.

kpslc, 2020-07-16:
I calibrated through the terminal. No rush in looking into it. Just thought you ought to know.

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com <mailto:kevin.plumb@gmail.com>

kizniche, 2020-07-16:
From the error, I see what the issue is. I'll commit a fix that will be available in the next release. Thanks for the bug report.

kpslc, 2020-07-17:
Great. Thanks. Awesome system. 

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com

closed 2020-07-23 by kizniche