kizniche/Mycodo #791: Add ADXL343-ADT7410 Sensor Module

Magnum-Pl, 2020-07-18:
Hey Kyle, 

Can you add the ADXL343-ADT7410 sensor module from Analog Devices/Digikey?

I attached the script I used to get it working. 

[ADXL343_ADT7410.ino.zip](https://github.com/kizniche/Mycodo/files/4941289/ADXL343_ADT7410.ino.zip)

[https://www.adafruit.com/product/4147](https://www.adafruit.com/product/4147)

[image]

kizniche, 2020-07-18:
Do you know of any Python code? Arduino code can't run on a Pi, so it would have to be ported if Python code doesn't exist.

kizniche, 2020-07-18:
I'd be inclined to make two input modules, one for each sensor. For instance, the ADXL343 has the most up-to-date code at https://github.com/adafruit/Adafruit_CircuitPython_ADXL34x however I've had trouble in the past attempting to change from the default I2C bus, which is a common thing to do with I2C devices. I'll dive back into the circuitpython code and try again, but if you happen to find a solution for this, let me know.

kizniche, 2020-07-18:
Alright, after some digging, it looks like https://github.com/adafruit/Adafruit_Python_Extended_Bus is what we need to use to be able to select the specific I2C bus the sensor is connected to. I can post a test Input module if you're willing to test.

kizniche, 2020-07-18:
Reference to issue for the Raspberry Pi 4 and its additional I2C buses: https://github.com/adafruit/Adafruit_Blinka/issues/189

Magnum-Pl, 2020-07-18:
Yeah, I can test it.

kizniche, 2020-07-18:
Here's test module v01: [adxl343_test_01.py.txt](https://github.com/kizniche/Mycodo/files/4942946/adxl343_test_01.py.txt)

Rename from .txt to .py and import from the Configuration -> Input menu.

kizniche, 2020-07-19:
And here's the ADT7410 test module v01: [adt7410_test_01.py.txt](https://github.com/kizniche/Mycodo/files/4943066/adt7410_test_01.py.txt)

Magnum-Pl, 2020-07-19:
I was able to get the test module installed and its working great. Can you change the units from "G-Force" to "MS^2" for meters per second squared?

Live data and a graph look like this:

[image]

[image]

kizniche, 2020-07-19:
The measurements are in g-force, not ms^2, correct?

kizniche, 2020-07-19:
If you want to convert to ms^2, you can add a conversion, then select convert from the input measurement options.

Magnum-Pl, 2020-07-19:
The sensor is supposed to measuring acceleration in units of ms^2. In the photo above, the sensor was sitting on a table. You can tell the downward direction is the Z-axis because the 10.003 value is near the 9.8ms^2 or the acceleration due to gravity. If it was measuring g-force, wouldn't the value on the Z-axis would be 1?

Magnum-Pl, 2020-07-19:
I was able to get the ADT7410 test module installed working too.

kizniche, 2020-07-19:
Okay. I was going off the data sheet, and that's the units it was using in the specifications. Typically they use the unit the sensor returns. Guess not in this case. I'll change the units for the modules in the release. Want me to attach them here afterbi make the edits, so you can use them before the release? I'm out of my house right now so won't be able to get to it until later. Thanks for testing!

kizniche, 2020-07-19:
I'll look into this further. It may actually be g and needs to be calibrated. In any case, I'll add options to the input(s) for these special actions/options
Now that I know they work, I feel better about putting in extra work.

Magnum-Pl, 2020-07-19:
I was thinking in terms of ms^2 because the Adafruit code I've been using displays ms^2 by default and I didn't see any conversion. I see in the data sheet that all the units are in g, but since one g is 9.8ms^2 and the z-axis value is within 2% of that, I'm assuming the data is in ms^2. 

I read in the data sheet on pages 28-29 about calibration. Would it be very difficult to add a method for calibration?

I'd appreciate the edited modules when you finish them. 

Thanks a lot for your help.

kizniche, 2020-07-19:
Just verified that the Adafruit library does indeed convert from g to m/s^2. I'll change the units in the Mycodo module.

See https://github.com/adafruit/Adafruit_CircuitPython_ADXL34x/blob/35a1a1bc11407c4db3ab94755a67a58d57deca89/adafruit_adxl34x.py#L179-L185

kizniche, 2020-07-19:
I added a conversion to/from g and m/s/s (available in next release), changed units to m/s/s, and added the option to select the range. There is no calibration in the Adafruit library. ADXL34x test module v02: [adxl34x_test_02.py.txt](https://github.com/kizniche/Mycodo/files/4943195/adxl34x_test_02.py.txt)

If you import the above input module without first making the changes to [mycodo/config_devices_units.py](https://github.com/kizniche/Mycodo/commit/058a57a2b00facecbb95834efdc9c16b0509cdb1#diff-8c6effbabae3548ad4e45fb242bf0a08), you may break your system. So, make sure you have the edits I just pushed before doing so.

There weren't any additional functions in the Adafruit ADT7410 library, so no new options were added to this input module.

Magnum-Pl, 2020-07-20:
I'll probably use it as is until the update. I have it running on an ESP8266 with MQTT right now and that only uses the MQTT input. It's nice to be able to run the sensor directly on the Pi though. 

Thanks for all your help.

closed 2020-07-23 by kizniche