kizniche/Mycodo #793: Joy-IT Explorer 700 OLED SPI

Mark0Mi, 2020-07-19:
Hi 
I have a PI with the Explorer 700 HAT from Joy-IT 
The display is the OLED 128x64 SD1306 
but unfortunately it seems not to use I2C -> instead it uses SPI (and it is fixed wired on the Explorer700) 

Product:
https://joy-it.net/en/products/RB-Explorer700
Code examples:
https://joy-it.net/files/files/Produkte/RB-Explorer700/Explorer-700%20Beispielcodes.zip
in the folder OLED there is the code SSD1306.py to address the display
and three sample codes for displaying text. 

Can you add support for addressing OLED displays over SPI instead of I2C? 
or guide me how to do this? 
thanks!

kizniche, 2020-07-19:
How specifically is it accessed by the Pi (i.e. hardware SPI, software SPI, software device, pins)?

Mark0Mi, 2020-07-19:
[SSD1306.py.txt](https://github.com/kizniche/Mycodo/files/4944345/SSD1306.py.txt)
[dispchar.py.txt](https://github.com/kizniche/Mycodo/files/4944348/dispchar.py.txt)
I have attached the two files to get the display to work 
as far as I understood it it uses the spidev with hardware SPI 
(from dispchar.py) 
# Raspberry Pi pin configuration:
RST = 19
# Note the following are only used with SPI:
DC = 16
bus = 0
device = 0
disp = SSD1306.SSD1306(RST, DC, SPI.SpiDev(bus,device))

the SSD1306 class is defined in the second file SSD1306.py

kizniche, 2020-07-20:
I just pushed a commit that adds the ability to select the LCD interface, and an added SPI interface for the SSD1306 LCDs. If you're feeling brave and want to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), you can test the new code, otherwise you'll have to wait until the next release.

Mark0Mi, 2020-07-20:
I did the test - unfortunately it does not work. 
deamon log: 
2020-07-20 16:20:37,192 - ERROR - mycodo.controllers.controller_lcd_b51387b7 - initialize_variables() Exception: invalid literal for int() with base 16: 'None'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 298, in initialize_variables
    self.lcd_out = LCD_Pioled(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled.py", line 23, in __init__
    self.i2c_address = int(str(lcd_dev.location), 16)
ValueError: invalid literal for int() with base 16: 'None'
2020-07-20 16:20:37,196 - INFO - mycodo.controllers.controller_lcd_b51387b7 - Activated in 625.3 ms
2020-07-20 16:20:37,197 - INFO - mycodo.daemon - All activated LCD controllers started

after I click deactivate 
Error: Could not deactivate LCD controller with ID b51387b7-ee89-4062-9433-675a9b65d5bf, it's not active.
after I click activate it seems it gets activated but in deamon log the following shows up:
2020-07-20 16:30:12,346 - ERROR - mycodo.controllers.controller_lcd_b51387b7 - initialize_variables() Exception: invalid literal for int() with base 16: 'None'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 298, in initialize_variables
    self.lcd_out = LCD_Pioled(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled.py", line 23, in __init__
    self.i2c_address = int(str(lcd_dev.location), 16)
ValueError: invalid literal for int() with base 16: 'None'
2020-07-20 16:30:12,347 - INFO - mycodo.controllers.controller_lcd_b51387b7 - Activated in 541.8 ms

is there any other log you might need? 
BR
marko

kizniche, 2020-07-20:
No, the daemon log is where all the exceptions should be going. I just pushed a fix. You can upgrade to master and test again.

Mark0Mi, 2020-07-20:
display works fine ;-) 
thank you very much! 

one minor thing - after booting it shows 
line1: mycodo 8.5.8
line2: start 128x64 OLED (SD:
and then after 30seconds (first refresh I guess) the configured sensor values

if deactivate is pressed same message with stop in line 2: (it should go blank shouldn't it?)
if activated again I have to wait 30 seconds for the refresh but that's ok.. 

many thank for your very quick fix! 
BR
marko

kizniche, 2020-07-20:
That is the expected behavior.

closed 2020-07-20 by Mark0Mi

Mark0Mi, 2020-08-20:
hi after running now on 8.7.1 finally - my display is not working anymore can you check it? 

2020-08-21 01:42:00,799 - INFO - mycodo.daemon - Mycodo daemon v8.7.1 starting
2020-08-21 01:42:07,440 - INFO - mycodo.devices.atlas_scientific_uart_dev_ttyUSB0 - Atlas Scientific Board: None, Rev: 0, Firmware: None
2020-08-21 01:42:07,440 - INFO - mycodo.outputs.atlas_ezo_pmp_4f5e51b9 - Initialized in 6326.2 ms
2020-08-21 01:42:07,444 - INFO - mycodo.outputs.wired_1fa43ce2 - Output setup on pin 5 and turned OFF (OFF=LOW)
2020-08-21 01:42:07,445 - INFO - mycodo.outputs.wired_1fa43ce2 - Initialized in 3.5 ms
2020-08-21 01:42:07,446 - INFO - mycodo.outputs.wired_9d5a4f4e - Output setup on pin 6 and turned OFF (OFF=LOW)
2020-08-21 01:42:07,446 - INFO - mycodo.outputs.wired_9d5a4f4e - Initialized in 0.5 ms
2020-08-21 01:42:07,446 - INFO - mycodo.controllers.controller_output - Setting Output 1fa43ce2 startup state to OFF
2020-08-21 01:42:07,447 - INFO - mycodo.controllers.controller_output - Setting Output 9d5a4f4e startup state to OFF
2020-08-21 01:42:07,447 - INFO - mycodo.controllers.controller_output - Activated in 6435.3 ms
2020-08-21 01:42:08,236 - INFO - mycodo.controllers.controller_conditional_460e1353 - Activated in 74.1 ms
2020-08-21 01:42:08,237 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-08-21 01:42:08,434 - INFO - mycodo.controllers.controller_trigger_317ab7f5 - Activated in 59.2 ms
2020-08-21 01:42:08,693 - INFO - mycodo.controllers.controller_trigger_e05919f9 - Activated in 62.2 ms
2020-08-21 01:42:08,958 - INFO - mycodo.controllers.controller_trigger_15e84940 - Activated in 58.6 ms
2020-08-21 01:42:09,272 - INFO - mycodo.controllers.controller_trigger_59b29264 - Activated in 76.8 ms
2020-08-21 01:42:09,276 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-08-21 01:42:09,717 - INFO - mycodo.controllers.controller_input_dd1b392c - Activated in 174.5 ms
2020-08-21 01:42:12,456 - INFO - mycodo.devices.atlas_scientific_ftdi_DT04DC2N - Atlas Scientific Board: ORP, Rev: 2, Firmware: 2.11
2020-08-21 01:42:12,525 - ERROR - mycodo.inputs.atlas_orp_7e37e204 - Calibration measurement not found within the past 120 seconds
2020-08-21 01:42:13,831 - INFO - mycodo.controllers.controller_input_7e37e204 - Activated in 3909.5 ms
2020-08-21 01:42:13,880 - ERROR - mycodo.inputs.atlas_orp_7e37e204 - Calibration measurement not found within the past 120 seconds
2020-08-21 01:42:14,163 - INFO - mycodo.controllers.controller_input_13487ed8 - Activated in 141.8 ms
2020-08-21 01:42:15,924 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 23.125, 'timestamp_utc': None}}
2020-08-21 01:42:16,854 - INFO - mycodo.devices.atlas_scientific_ftdi_DT04DD8O - Atlas Scientific Board: pH, Rev: 2, Firmware: 2.14
2020-08-21 01:42:16,878 - ERROR - mycodo.inputs.atlas_ph_ea2abd1f - Calibration measurement not found within the past 120 seconds
2020-08-21 01:42:18,181 - INFO - mycodo.controllers.controller_input_ea2abd1f - Activated in 3880.1 ms
2020-08-21 01:42:18,256 - ERROR - mycodo.inputs.atlas_ph_ea2abd1f - Calibration measurement not found within the past 120 seconds
2020-08-21 01:42:18,442 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The water temp value is None
2020-08-21 01:42:18,446 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The outside temp value is 23.125
2020-08-21 01:42:18,573 - INFO - mycodo.controllers.controller_input_551be49e - Activated in 130.0 ms
2020-08-21 01:42:18,574 - INFO - mycodo.daemon - All activated Input controllers started
2020-08-21 01:42:18,574 - INFO - mycodo.daemon - All activated Math controllers started
2020-08-21 01:42:18,574 - INFO - mycodo.daemon - All activated PID controllers started
2020-08-21 01:42:19,438 - ERROR - mycodo.controllers.controller_lcd_b51387b7 - initialize_variables() Exception: No module named 'Adafruit_BBIO'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 292, in initialize_variables
    self.lcd_out = LCD_Pioled(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled.py", line 50, in __init__
    spi=SPI.SpiDev(lcd_dev.spi_bus, lcd_dev.spi_device))
  File "/home/pi/Mycodo/env/src/adafruit-ssd1306/Adafruit_SSD1306/SSD1306.py", line 243, in __init__
    gpio, spi, i2c_bus, i2c_address, i2c)
  File "/home/pi/Mycodo/env/src/adafruit-ssd1306/Adafruit_SSD1306/SSD1306.py", line 85, in __init__
    self._gpio = GPIO.get_platform_gpio()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_GPIO/GPIO.py", line 420, in get_platform_gpio
    import Adafruit_BBIO.GPIO
ModuleNotFoundError: No module named 'Adafruit_BBIO'
2020-08-21 01:42:19,441 - INFO - mycodo.controllers.controller_lcd_b51387b7 - Activated in 745.6 ms
2020-08-21 01:42:19,441 - INFO - mycodo.daemon - All activated LCD controllers started
2020-08-21 01:42:19,441 - INFO - mycodo.daemon - All activated Custom controllers started
2020-08-21 01:42:19,514 - INFO - mycodo.controllers.controller_widget - Activated in 73.0 ms
2020-08-21 01:42:20,516 - INFO - mycodo.daemon - Mycodo daemon started in 19.717 seconds
[... cut ...]
2020-08-21 01:44:39,179 - INFO - mycodo.controllers.controller_lcd_b51387b7 - Activated in 808.5 ms
2020-08-21 01:44:39,179 - INFO - mycodo.daemon - All activated LCD controllers started
2020-08-21 01:44:39,180 - INFO - mycodo.daemon - All activated Custom controllers started
2020-08-21 01:44:39,252 - INFO - mycodo.controllers.controller_widget - Activated in 72.1 ms
2020-08-21 01:44:40,254 - INFO - mycodo.daemon - Mycodo daemon started in 23.226 seconds
2020-08-21 01:44:40,260 - INFO - mycodo.daemon - 50.38 MB RAM in use

kizniche, 2020-08-20:
```
ModuleNotFoundError: No module named 'Adafruit_BBIO'
```

```
~/Mycodo/env/bin/pip install Adafruit_BBIO
```

Mark0Mi, 2020-08-21:
2020-08-21 09:16:35,509 - ERROR - mycodo.controllers.controller_lcd_b51387b7 - initialize_variables() Exception: argument 1 must be str, not int
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 292, in initialize_variables
    self.lcd_out = LCD_Pioled(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled.py", line 50, in __init__
    spi=SPI.SpiDev(lcd_dev.spi_bus, lcd_dev.spi_device))
  File "/home/pi/Mycodo/env/src/adafruit-ssd1306/Adafruit_SSD1306/SSD1306.py", line 243, in __init__
    gpio, spi, i2c_bus, i2c_address, i2c)
  File "/home/pi/Mycodo/env/src/adafruit-ssd1306/Adafruit_SSD1306/SSD1306.py", line 89, in __init__
    self._gpio.setup(self._rst, GPIO.OUT)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_GPIO/GPIO.py", line 278, in setup
    pull_up_down=self._pud_mapping[pull_up_down])
TypeError: argument 1 must be str, not int
2020-08-21 09:16:35,518 - INFO - mycodo.controllers.controller_lcd_b51387b7 - Activated in 832.0