kizniche/Mycodo #794: Error 502: Bad Gateway Error

eliaskill, 2020-07-21:
Something bad happened but it's probably not your fault. Try refreshing the page to see if it will load. If that doesn't work, try rebooting your system, then try refreshing it again. If that still doesn't work, submit a new issue on GitHub with any usefull information that can help determine the cause of this issue.

kizniche, 2020-07-22:
Can you execute the command ```journalctl -u mycodoflask | tail -n 60``` and copy the output here?

eliaskill, 2020-07-22:
pi@raspberrypi:~ $ journalctl -u mycodoflask | tail -n 60
-- Logs begin at Tue 2020-07-21 19:48:50 -03, end at Tue 2020-07-21 21:12:13 -03. --
jul 21 20:01:01 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
jul 21 20:01:01 raspberrypi systemd[2117]: mycodoflask.service: Failed to execute command: No such file or directory
jul 21 20:01:01 raspberrypi systemd[2117]: mycodoflask.service: Failed at step EXEC spawning /var/mycodo-root/env/bin/gunicorn: No such file or directory
jul 21 20:01:01 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=203/EXEC
jul 21 20:01:01 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.

kizniche, 2020-07-22:
What is the output of these commands?

```bash
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

eliaskill, 2020-07-22:
pi@raspberrypi:~ $ cd ~/Mycodo/mycodo
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/run/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
sudo: /var/mycodo-root/env/bin/gunicorn: comando não encontrado

kizniche, 2020-07-22:
There seems to be something critically wrong with your install. Did you just install Mycodo? if so, can you attach the setup.log located in ~/Mycodo/install/ ?

eliaskill, 2020-07-22:
### Mycodo installation initiated ter jul 21 19:55:49 -03 2020
#### Mycodo installation began ter jul 21 19:56:36 -03 2020

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Updating apt repositories
Atingido:1 http://archive.raspberrypi.org/debian buster InRelease
Atingido:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Atingido:3 https://download.sublimetext.com apt/stable/ InRelease
Lendo listas de pacotes...

#### Uninstalling apt version of pip (if installed)
Lendo listas de pacotes...
Construindo árvore de dependências...
Lendo informação de estado...
O pacote 'python-pip' não está instalado, por isso não será removido
Os seguintes pacotes foram instalados automaticamente e já não são necessários:
  libjs-sphinxdoc libjs-underscore libmicrodns0 libpython-all-dev python-all
  python-all-dev python-configparser python-dbus python-entrypoints
  python-keyring python-keyrings.alt python-secretstorage python-wheel
  python-xdg rpi-eeprom-images
Utilize 'sudo apt autoremove' para os remover.
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 16 não atualizados.

#### Installing prerequisite apt packages and update pip
Lendo listas de pacotes...
Construindo árvore de dependências...
Lendo informação de estado...
O pacote 'apache2' não está instalado, por isso não será removido
Os seguintes pacotes foram instalados automaticamente e já não são necessários:
  libjs-sphinxdoc libjs-underscore libmicrodns0 libpython-all-dev python-all
  python-all-dev python-configparser python-dbus python-entrypoints
  python-keyring python-keyrings.alt python-secretstorage python-wheel
  python-xdg rpi-eeprom-images
Utilize 'sudo apt autoremove' para os remover.
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 16 não atualizados.
Lendo listas de pacotes...
Construindo árvore de dependências...
Lendo informação de estado...
gawk is already the newest version (1:4.2.1+dfsg-1).
gcc is already the newest version (4:8.3.0-1+rpi2).
git is already the newest version (1:2.20.1-2+deb10u3).
libffi-dev is already the newest version (3.2.1-9).
libi2c-dev is already the newest version (4.1-1).
logrotate is already the newest version (3.14.0-4).
moreutils is already the newest version (0.62-1+b1).
nginx is already the newest version (1.14.2-2+deb10u1).
python3 is already the newest version (3.7.3-1).
python3-dev is already the newest version (3.7.3-1).
python3-pylint-common is already the newest version (0.2.5-2).
python3-setuptools is already the newest version (40.8.0-1).
python3-smbus is already the newest version (4.1-1).
rng-tools is already the newest version (2-unofficial-mt.14-1).
sqlite3 is already the newest version (3.27.2-3).
wget is already the newest version (1.20.1-1.1).
Os seguintes pacotes foram instalados automaticamente e já não são necessários:
  libjs-sphinxdoc libjs-underscore libmicrodns0 libpython-all-dev python-all
  python-all-dev python-configparser python-dbus python-entrypoints
  python-keyring python-keyrings.alt python-secretstorage python-wheel
  python-xdg rpi-eeprom-images
Utilize 'sudo apt autoremove' para os remover.
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 16 não atualizados.
Searching for pip
Best match: pip 20.1.1
Adding pip 20.1.1 to easy-install.pth file
Installing pip script to /usr/local/bin
Installing pip3 script to /usr/local/bin
Installing pip3.8 script to /usr/local/bin

[... 284 lines cut ...]
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/scripts/restart_daemon.py", line 12, in <module>
    from mycodo.config import DAEMON_PID_FILE
  File "/home/pi/Mycodo/mycodo/config.py", line 10, in <module>
    from flask_babel import lazy_gettext
ModuleNotFoundError: No module named 'flask_babel'
Job for mycodo.service failed because the control process exited with error code.
See "systemctl status mycodo.service" and "journalctl -xe" for details.
#### Mycodo Installer finished ter jul 21 20:11:11 -03 2020
#### Total install time: 14 minutes and 35 seconds

************************************
** Mycodo successfully installed! **
************************************

The full install log is located at:
/home/pi/Mycodo/install/setup.log

Go to https://192.168.100.31/, or whatever your Raspberry Pi's
IP address is, to create an admin user and log in.

kizniche, 2020-07-22:
This appears to be a duplicate of #787/#792. To fix, run these commands and you should have everything fixed:

```bash
sudo apt remove python-cffi-backend python3-cffi-backend
sudo service mycodo stop
sudo service mycodoflask stop
mv ~/Mycodo/env ~/Mycodo/env-old
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh setup-virtualenv
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-dependencies
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions
sudo shutdown now -r
```

eliaskill, 2020-07-22:
thank you so much for the help now it's working !! I even did the loggin :) thanks again

eliaskill, 2020-07-22:
[image]

kizniche, 2020-07-22:
Glad it was an easy fix. I need to get the next release out soon, as it has the fix to this issue already.

eliaskill, 2020-07-22:
very happy to be able to speak creator of this wonderful tool! pardon my English! I'm Brazilian using a translator!

kizniche, 2020-07-22:
Nice to meet you. I'm sorry you had an issue installing the software. It normally doesn't break so early in its use. haha. Let me know if you find any really bad translation errors with the Portuguese translation. You can change the language to Portuguese on the General Settings page after clicking the gear icon at the top right of the page.

closed 2020-07-23 by kizniche