kizniche/Mycodo #795: if a sensor breaks - lcd output

Mark0Mi, 2020-07-22:
hi 
while it seems that functions and methods do nothing if a sensor stops working it is different with the LCD output. 
e.g. I have a temp sensor and it shows the temp on the display - if I disconnect the sensor (e.g. if it breaks) the LCD still shows the last fetched temp. (even if this was 10 hours ago) a reboot solves the problem (e.g. I see an error on the display line) 

can I somehow change this (e.g. write to the display no sensor data - if for a period of time no value could be fetched?) - so that I immediately recognize that something is not working?

kizniche, 2020-07-22:
What is your Max Age setting for that particular display line?

Mark0Mi, 2020-07-22:
first I did not set any - now I have 30 seconds - still no change on the display 
also I have set to display the temp-timestamp which is updated! (even if the sensor is disconnected) 

[image]

here is the log:
  measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 16:14:30,737 - ERROR - mycodo.controllers.controller_input_072dcb6e - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 16:14:31,401 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2020-07-22 16:14:45,495 - ERROR - mycodo.controllers.controller_input_30a6e21f - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 16:14:45,762 - ERROR - mycodo.controllers.controller_input_072dcb6e - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 16:14:46,443 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2020-07-22 16:15:00,421 - ERROR - mycodo.controllers.controller_input_30a6e21f - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 16:15:00,792 - ERROR - mycodo.controllers.controller_input_072dcb6e - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 16:15:01,384 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.

kizniche, 2020-07-22:
Let's back up. I need more information about your environment. What Mycodo version?

Mark0Mi, 2020-07-22:
I was the guy for whom you implemented the SPI display 
Mycodo Version: 8.5.8 (with the special fetch from the SPI integration) 
Python Version: 3.7.3 (default, Dec 20 2019, 18:57:59) [GCC 8.3.0]
Database Version: 4ea0a59dee2b

I use a 1-wire temp sensor - DS18B20
fresh reboot of the system - sensor attached 
I use a function if temp > 27 then switch on relay for 10 seconds
2020-07-22 16:33:32,042 - INFO - mycodo.outputs.python_8c826e37 - Initialized in 213.3 ms
2020-07-22 16:33:32,087 - INFO - mycodo.outputs.python_8abf9422 - Initialized in 42.9 ms
2020-07-22 16:33:32,130 - INFO - mycodo.outputs.python_65ab79dc - Initialized in 42.2 ms
2020-07-22 16:33:32,131 - INFO - mycodo.controllers.controller_output - Activated in 540.6 ms
2020-07-22 16:33:32,894 - INFO - mycodo.controllers.controller_conditional_460e1353 - Activated in 53.0 ms
2020-07-22 16:33:32,895 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-07-22 16:33:32,895 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-07-22 16:33:33,217 - INFO - mycodo.controllers.controller_input_dd1b392c - Activated in 194.3 ms
2020-07-22 16:33:33,532 - INFO - mycodo.controllers.controller_input_13487ed8 - Activated in 137.2 ms
2020-07-22 16:33:33,533 - INFO - mycodo.daemon - All activated Input controllers started
2020-07-22 16:33:33,533 - INFO - mycodo.daemon - All activated Math controllers started
2020-07-22 16:33:33,534 - INFO - mycodo.daemon - All activated PID controllers started
2020-07-22 16:33:35,039 - INFO - mycodo.controllers.controller_lcd_b51387b7 - Activated in 1383.7 ms
2020-07-22 16:33:35,039 - INFO - mycodo.daemon - All activated LCD controllers started
2020-07-22 16:33:35,039 - INFO - mycodo.daemon - All activated Custom controllers started
2020-07-22 16:33:35,540 - INFO - mycodo.daemon - Mycodo daemon started in 4.219 seconds
2020-07-22 16:33:35,545 - INFO - mycodo.daemon - 43.93 MB RAM in use
2020-07-22 16:33:42,949 - INFO - mycodo.controllers.controller_conditional_460e1353 - This INFO log entry will appear in the Daemon Log
2020-07-22 16:33:42,950 - ERROR - mycodo.controllers.controller_conditional_460e1353 - This ERROR log entry will appear in the Daemon Log
2020-07-22 16:33:43,041 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The value is 30.375
2020-07-22 16:33:43,094 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 16:33:43: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: ON
2020-07-22 16:33:53,138 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 16:33:53: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: OFF
2020-07-22 16:34:07,305 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 16:33:50
2020-07-22 16:34:07,320 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30.375 @ 2020-07-22 16:33:50
2020-07-22 16:34:07,338 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 16:33:50
2020-07-22 16:34:07,350 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30 @ 2020-07-22 16:33:50
2020-07-22 16:34:43,134 - INFO - mycodo.controllers.controller_conditional_460e1353 - This INFO log entry will appear in the Daemon Log
2020-07-22 16:34:43,135 - ERROR - mycodo.controllers.controller_conditional_460e1353 - This ERROR log entry will appear in the Daemon Log
2020-07-22 16:34:43,208 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The value is 30.5625
2020-07-22 16:34:43,259 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 16:34:43: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: ON

16:36 I removed the 1-wire sensor 
in the condition sensor data must not be older than 360 seconds 
[image]

it seems this is also NOT working - the relay switches on even if the temp sensor is next to the PI disconnected... 

2020-07-22 16:44:10,856 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30.4375 @ 2020-07-22 16:44:05
2020-07-22 16:44:10,868 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 16:44:05
2020-07-22 16:44:10,877 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30 @ 2020-07-22 16:44:05
2020-07-22 16:44:20,556 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2020-07-22 16:44:35,501 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2020-07-22 16:44:43,017 - INFO - mycodo.controllers.controller_conditional_460e1353 - This INFO log entry will appear in the Daemon Log
2020-07-22 16:44:43,017 - ERROR - mycodo.controllers.controller_conditional_460e1353 - This ERROR log entry will appear in the Daemon Log
2020-07-22 16:44:43,109 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The value is 30.4375
2020-07-22 16:44:43,175 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 16:44:43: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: ON
2020-07-22 16:44:50,546 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
[... cut ...]
2020-07-22 16:46:11,561 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30.4375 @ 2020-07-22 16:46:05
2020-07-22 16:46:11,574 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 16:46:05
2020-07-22 16:46:11,586 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30 @ 2020-07-22 16:46:05
2020-07-22 16:46:20,514 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.

Mark0Mi, 2020-07-22:
also note the system knows the sensor is not present
[image]

kizniche, 2020-07-22:
I cannot replicate the issue.

When a Max Age of X is set, if there hasn't been a measurement stored in the database in the past X seconds, "NO DATA" is displayed on the LCD.

It was an oversight on my part that allowed LCD controllers to be activated without Max Age or Decimal Places set in the first place. I just pushed a fix that prevents this.

kizniche, 2020-07-22:
Example: LCD with Period of 5 seconds and one line set to an Input with a Max Age of 30 seconds. The LCD is activated, then the Input is deactivated.

Here is my Daemon log:

```
2020-07-22 11:19:39,900 - INFO - mycodo.controllers.controller_lcd_0416bc43 - Activated in 860.5 ms
2020-07-22 11:19:44,969 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - Latest temperature: 12.14 @ 2020-07-22 11:19:40
2020-07-22 11:19:55,211 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - Latest temperature: 12.14 @ 2020-07-22 11:19:40
2020-07-22 11:19:58,314 - INFO - mycodo.controllers.controller_input_fcf05cd7 - Deactivated in 91.6 ms
2020-07-22 11:20:05,389 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - Latest temperature: 12.2 @ 2020-07-22 11:19:55
2020-07-22 11:20:15,582 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - Latest temperature: 12.2 @ 2020-07-22 11:19:55
2020-07-22 11:20:25,771 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - No data returned from influxdb
2020-07-22 11:20:35,973 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - No data returned from influxdb
2020-07-22 11:20:46,173 - DEBUG - mycodo.controllers.controller_lcd_0416bc43 - No data returned from influxdb
```

Mark0Mi, 2020-07-22:
I understand 
in my point of view it seems my input does not get disabled it seems it tries to set a value of none which is not allowed? 
So it is not a display issue - it is an issue with my input 

2020-07-22 20:55:39,237 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 20:55:35
2020-07-22 20:55:39,250 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30.4375 @ 2020-07-22 20:55:35
2020-07-22 20:55:39,262 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 20:55:35
2020-07-22 20:55:39,274 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30 @ 2020-07-22 20:55:35
2020-07-22 20:55:48,512 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The value is 30.4375
2020-07-22 20:55:48,571 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 20:55:48: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: ON
2020-07-22 20:55:50,496 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.
2020-07-22 20:55:58,580 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 20:55:58: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: OFF
2020-07-22 20:56:05,540 - ERROR - mycodo.inputs.ds18b20_13487ed8 - Cannot set a value of None. Must be a float or string representing a float. Check the sensor and Input module is working correctly.

If I go to setup -> data and disable the sensor and enable it again I get exactly what you get -"No data returned from influxdb"
LOG:
w1thermsensor.errors.NoSensorFoundError: 
Could not find sensor of type DS18B20 with id 00000ab604cf
Please check cabling and check your /boot/config.txt for
dtoverlay=w1-gpio
2020-07-22 20:59:25,208 - INFO - mycodo.controllers.controller_input_13487ed8 - Activated in 143.3 ms
2020-07-22 20:59:25,208 - ERROR - mycodo.controllers.controller_input_13487ed8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 20:59:40,239 - ERROR - mycodo.controllers.controller_input_13487ed8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-22 20:59:40,675 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 20:59:06
2020-07-22 20:59:40,689 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 20:59:40,702 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest time: 30 @ 2020-07-22 20:59:06
2020-07-22 20:59:40,716 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 30 @ 2020-07-22 20:59:06
2020-07-22 20:59:48,416 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The value is 30.4375
2020-07-22 20:59:48,498 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 20:59:48: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: ON

kizniche, 2020-07-22:
"No data returned from influxdb" is the result of the LCD controller querying a measurement for the selected Input measurement for that particular LCD line. Since there is no measurement to be found for that particular Input measurement within the set Max Age, none is returned and "NO DATA" is displayed on the LCD.

Regarding the DS18B20 error "Cannot set a value of None", this is unrelated and is not causing the issue. Also, I pushed a fix for this today. Attempting to store a None value merely doesn't store anything, so could nto possibly be causing the issue.

kizniche, 2020-07-22:
Since I am unable to reproduce the issue, can you provide the exact steps to induce the issue on your system?

I see you have other devices operating. I would recommend you turn ***everything*** off so you can clearly see and understand what is going on. Create an LCD controller and set all lines Blank except one, which should point to the measurement of your sensor. Only activate that single sensor, and observe the result.

Mark0Mi, 2020-07-22:
just a second - I have the following configured: 
Setup -> Data: 
one TEMP sensor DS18B20 
one RPi Temp sensor 

I have four outputs configured R1-R4 /python code
I have one conditional function 
which is more or less your default example
[image]
[image]

and the LCD
[image]

So 
if I boot up the pi and have the sensor attached everything is working 
the sensor reads approx 30° and R1 is activated for 10 seconds (like it should be) 
LCD shows correct readings

If I remove the sensor nothing changes - R1 gets activated LCD shows the values like the sensor would work 
If I go to setup -> data and disable the sensor -> enable the sensor then the correct behavior is shown (No data on the LCD screen // R1 does not get activated) 

I will now do a update of the system (to get the fixes from today) and reboot

kizniche, 2020-07-22:
You are not using my recommendations. You are also introducing a completely unrelated issue. To diagnose the LCD (and only the LCD issue), disable ***everything*** and follow my recommendations [here](https://github.com/kizniche/Mycodo/issues/795#issuecomment-662647491).

> I will now do a update of the system (to get the fixes from today) and reboot

Like I said, there is nothing in the code I pushed that would fix your issue. As I alluded to earlier, you need to have Max Age set, and I notice you do not have Max Age set for several LCD lines.

Mark0Mi, 2020-07-22:
I did not want to add another unrelated issue - as in my point of view this is related. 
I know I don't have the max age set but it was the same measurement (to see if there is a difference) 

after disabling EVERYTHING except the temp sensor and the lcd, with one line configured -> upgrade to the latest and a reboot  everything works as expected - here is the log - exactly like it should be
21:40 boot - up 
21:41 read sensor and display it on LCD 
21:41 disconnect sensor 
21:42 No data is displayed (this is the correct behavior) 
now I activate the condition 
21:44  - and the LCD suddenly displays a temp (the sensor is still disconnected since 21:41)

[image]

2020-07-22 21:40:36,748 - INFO - mycodo.daemon - 41.75 MB RAM in use
2020-07-22 21:40:37,186 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.625, 'timestamp_utc': None}}
2020-07-22 21:41:02,645 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.9375, 'timestamp_utc': None}}
2020-07-22 21:41:06,143 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 29.938 @ 2020-07-22 21:41:02
2020-07-22 21:41:07,046 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:41:22,880 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:41:36,308 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 21:41:37,910 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:41:52,936 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:42:06,473 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 21:42:07,869 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:42:22,900 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:42:36,636 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 21:42:37,940 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:42:52,880 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:43:06,800 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 21:43:07,320 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:43:22,261 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:43:36,966 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 21:43:37,301 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:43:52,342 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:44:07,128 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - No data returned from influxdb
2020-07-22 21:44:07,282 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:44:22,321 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
[... cut ...]
2020-07-22 21:45:37,325 - DEBUG - mycodo.controllers.controller_input_13487ed8 - Adding measurements to InfluxDB with ID 13487ed8-c14e-4c47-8da6-124028f42280: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 29.875, 'timestamp_utc': None}}
2020-07-22 21:45:37,615 - DEBUG - mycodo.controllers.controller_lcd_b51387b7 - Latest temperature: 29.875 @ 2020-07-22 21:45:37
2020-07-22 21:45:43,434 - INFO - mycodo.controllers.controller_conditional_460e1353 - Check this measurement in the Daemon Log. The value is 29.875
2020-07-22 21:45:43,486 - INFO - mycodo.outputs.python_c0a5d193 - 2020-07-22 21:45:43: ID: c0a5d193-ea31-4d10-b446-7fa055741cb0: ON

kizniche, 2020-07-22:
Does your sensor happen to be [counterfeit](https://github.com/cpetrich/counterfeit_DS18B20)?

kizniche, 2020-07-22:
> Does your sensor happen to be counterfeit?

This doesn't affect the issue, I'm just curious since you're using the DS18B20.

With regard to the issue, I think I might have found what's causing the recording of measurements after an input crash or disconnect of an input. I'm investigating now.

Mark0Mi, 2020-07-22:
Might be possible but I don't think so - I bought it through one of the biggest distributors in Austria. 
Also it would not be logical for me if the sensor is NOT present why a value would show up? 

What I don't get is - if this ghost value comes up - it is added to the database (of course) but where is the value coming from? is this maybe an older read of the last value that gets added? 
also what I noticed when I look at the input - 1-wire ID - it clearly states None found - which should logically mean there is no sensor present

kizniche, 2020-07-22:
I figured out the issue. This is actually a very significant bug! It came down to how the Input modules were copying the measurement dictionary template to add measurements to, prior to passing it to the Input Controller. It was in fact performing a shallow copy rather than a deep copy. As a result, the original dictionary was being modified when it shouldn't have been. I just pushed a fix that changes how all Inputs behave with their measurement dictionary copying. If you upgrade to master, you can test if it fixes your issue.

I'm actually very surprised this bug survived this long without being noticed.

Mark0Mi, 2020-07-22:
thanks for figuring it out 
I'm honestly not a programming guy - so whatever it was glad it got fixed  ;-) 

- it is working now! thanks a lot!

closed 2020-07-22 by Mark0Mi