kizniche/Mycodo #796: DS18S20 - not showing any temp values

Mark0Mi, 2020-07-23:
Hi Kyle 

I'm terrible sorry but I need your help again
I have multiple 1-wire temp sensors 
currently the DS18B20 works fine 
I also have two DS18S20 (don't know why I ordered these instead of the ones working but they should work on mycodo as well) 
My problem is they get recognized as Input in mycodo and I can add them to the dashboards but mycodo does not show any value on the dashboards 

mycodo is on the latest version

on pi cli I can query the sensor so it is working basically 
cat /sys/bus/w1/devices/10-000803a2cf1b/w1_slave
[image]
[image]
deamon log
2020-07-23 21:59:00,696 - ERROR - mycodo.controllers.controller_input_4fecc312 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-23 21:59:15,721 - ERROR - mycodo.controllers.controller_input_4fecc312 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-23 21:59:30,747 - ERROR - mycodo.controllers.controller_input_4fecc312 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

thanks for looking into this

kizniche, 2020-07-23:
>  'NoneType' object has no attribute 'next'

This error occurs after the input has already crashed. For clarity in your log, deactivate all inputs, reboot, then only activate the input you want to view the error for. Then, immediately deactivate this input, and check the log. You need to find the first error before the " 'NoneType' object has no attribute 'next'" errors begin, as these errors are not useful.

kizniche, 2020-07-23:
> on pi cli I can query the sensor so it is working basically

Verified, as in you have successfully retrieved a valid measurement from the sensor?

Mark0Mi, 2020-07-23:
hi 
here is the log 
2020-07-23 22:18:52,404 - INFO - mycodo.controllers.controller_output - Activated in 422.1 ms
2020-07-23 22:18:52,908 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-07-23 22:18:52,909 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-07-23 22:18:52,909 - INFO - mycodo.daemon - All activated Input controllers started
2020-07-23 22:18:52,909 - INFO - mycodo.daemon - All activated Math controllers started
2020-07-23 22:18:52,910 - INFO - mycodo.daemon - All activated PID controllers started
2020-07-23 22:18:52,910 - INFO - mycodo.daemon - All activated LCD controllers started
2020-07-23 22:18:52,910 - INFO - mycodo.daemon - All activated Custom controllers started
2020-07-23 22:18:53,410 - INFO - mycodo.daemon - Mycodo daemon started in 1.655 seconds
2020-07-23 22:18:53,416 - INFO - mycodo.daemon - 39.32 MB RAM in use
2020-07-23 22:21:55,773 - ERROR - mycodo.controllers.controller_input_4fecc312 - initialize_variables() Exception: 'custom_options'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 350, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/ds18s20.py", line 70, in __init__
    INPUT_INFORMATION['custom_options'], input_dev)
KeyError: 'custom_options'
2020-07-23 22:21:55,780 - INFO - mycodo.controllers.controller_input_4fecc312 - Activated in 136.7 ms
2020-07-23 22:21:55,780 - ERROR - mycodo.controllers.controller_input_4fecc312 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-23 22:22:06,249 - ERROR - mycodo.daemon - Could not deactivate Input controller with ID 4fecc312-2ffa-4bd2-8a1f-e3185d7b9a4c: 'NoneType' object has no attribute 'stop_input'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 365, in controller_deactivate
    self.controller[cont_type][cont_id].stop_controller()
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 111, in stop_controller
    self.pre_stop()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 564, in pre_stop
    self.measure_input.stop_input()
AttributeError: 'NoneType' object has no attribute 'stop_input'
2020-07-23 22:22:10,808 - ERROR - mycodo.controllers.controller_input_4fecc312 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

@working yes on the cli 
cat /sys/bus/w1/devices/10-000803a2cf1b/w1_slave
36 00 4b 46 ff ff 0c 10 7e : crc=7e YES
36 00 4b 46 ff ff 0c 10 7e t=27000
which shows 27°

kizniche, 2020-07-23:
Thanks. I just pushed a fix. You can upgrade to master to test.

Mark0Mi, 2020-07-23:
what should I say - thanks it works! 
one more question - if I switch to parasitic power supply (not sure if this is the right word) where I only use two wires and a pull-up resistor - do I need to change anything in mycodo to make it aware of the pullup resistor?

kizniche, 2020-07-23:
You're welcome. You don't need to do anything in Mycodo. The operating system (kernel) is responsible for detecting the sensor. Mycodo merely communicates with the kernel device.

Mark0Mi, 2020-07-23:
thank you very much!!

closed 2020-07-23 by Mark0Mi