kizniche/Mycodo #799: Output Dosing Pump Duration with 8.6.1

kpslc, 2020-07-25:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

Since updating to 8.6.1 our system will not dose any pump from the conditional statement. Researching the problem I found that each individual pump on the output menu now has two manual operation options: Duration and Volume. 
-Volume works. Entering 10 causes 10ml to dispense.
-Duration does not work. Entering any value displays a success popup, but the pump doesn't activate.

This is causing the conditional statement to also not pump any pH or EC chemicals as they are using a duration valye

### Versions:

 - Mycodo Version: 8.6.1
 - Raspberry Pi Version: 4
 - Raspbian OS Version: 5.4.51-v7l+ #1326 SMP 

Mycodo Version: 8.6.1 
Python Version: 3.7.3 (default, Dec 20 2019, 18:57:59) [GCC 8.3.0] 
Database Version: 4ea0a59dee2b
Daemon Status: Running 
Daemon Process ID: 946 
Daemon RAM Usage: 67.66 MB 
Daemon Virtualenv: Yes 
Frontend Process ID: 465 
Frontend RAM Usage: 52.968 MB 
Frontend Virtualenv: Yes
uptime
 12:30:19 up 31 min,  2 users,  load average: 0.12, 0.19, 0.14
Kernel Information: uname -a
Linux raspberrypi 5.4.51-v7l+ #1326 SMP Fri Jul 17 10:51:18 BST 2020 armv7l GNU/Linux

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to setup/output
2. Click on any output pump
3. Enter a duration value such as 1 or 5 and click on "send"
4. The pump doesn't respond

Entering a volume amount and clicking send, does cause the pump to cycle.

### Expected behavior

A clear and concise description of what you expected to happen.
An entered value in "duration" should run the pump for that duration.

### Screenshots

If applicable, add screenshots to help explain your problem.
[image]

This image shows the pumps dosing. Then the system was updated to 8.6.1 and the dosing from the conditional statement stopped. The dosing events to the far right of the image were manually completed with a volume amount.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2020-07-25:
Thank you for the bug report and the detailed information. I was able to replicate the issue and develop a fix. I just committed the fix and it will be available in the next release, which should be sometime today.

kpslc, 2020-07-25:
You are awesome 

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com

kizniche, 2020-07-25:
Just released v8.6.2

closed 2020-07-25 by kizniche