kizniche/Mycodo #800: lcd bug when blank line used

Mark0Mi, 2020-07-25:
hi kyle 
I'm on the latest version my LCD display is on Explorer Hat 700 (you know my LCD is the one with SPI) - but I think this is not the problem here. 

in the display set if I use Blank line (like in this example) I get only the first 4 lines printed to the display
[image]

it does not matter where the blank line is (e.g. line 5,6,7,8) I get 4 lines printed lines 5-8 are dark.
If I have a blank line in line 1-4 it does not show such an effect 
If I select custom text and add a blank I have a work-around but I think you should know that there is something odd going on
unfortunately the logs do not show anything regarding this issue.

kizniche, 2020-07-25:
I wasn't able to reproduce the issue. For simplicity sake, and so you don't have confounding variables, deactivate your current LCD controller, create a new LCD controller with one set of lines, and set all lines to Custom Text with a unique string (e.g. "TEST 1", "TEST 2", etc.). Activate and verify all lines display each string. Deactivate and change a line in the middle to Blank, save, and activate. Please report the results.

kizniche, 2020-07-25:
I don't have an 8-line display available to test, but I think I see the issue after looking at the code. I'll have a test fix you can try in a moment.

Mark0Mi, 2020-07-25:
Hi 
I can reproduce the issue 
did the set with custom text 1-8 and it shows 1-8 on the display
as soon as I set line 5 to blank line 5-8 are gone (they should show 6,7,8)
[image]

kizniche, 2020-07-25:
Open Mycodo/mycodo/devices/lcd_pioled.py and change these lines:

https://github.com/kizniche/Mycodo/blob/0f14212e0d6e06776d382a2ceed86c9b74eceecc/mycodo/devices/lcd_pioled.py#L90-L94

To this:

```python
        if message_line_5 is not None:
            draw.text((x, top + 32), message_line_5, font=font, fill=255)
        if message_line_6 is not None:
            draw.text((x, top + 40), message_line_6, font=font, fill=255)
        if message_line_7 is not None:
            draw.text((x, top + 48), message_line_7, font=font, fill=255)
        if message_line_8 is not None:
            draw.text((x, top + 54), message_line_8, font=font, fill=255)
```

Mark0Mi, 2020-07-25:
it works ;-)

kizniche, 2020-07-25:
Great, I'll include the fix in the next release. Thanks for the big report. You caught me right as I was finishing up a bunch of other code changes and had my IDE open.

Mark0Mi, 2020-07-25:
one more question - I know I should probably open a new ticket for it: 
regarding temp sensors e.g. DS18S20 over 1-wire interface - I can't calibrate them - do you think it would be possible e.g. on the input sensors to have a field for adjusting the temp-value e.g. if the sensor reads 25° but I know the sensor is off by 1° I'd like to correct that by entering e.g. -1 and it should then show 24°

kizniche, 2020-07-25:
This was discussed in #608. I have no plans to add an offset option, but you're welcome to copy the DS18B20 Input module and add a [custom integer option](https://github.com/kizniche/Mycodo/wiki/Building-a-Custom-Input-Module#custom-options), which will allow you to apply any offset you wish to the received measurement prior to storing in the measurement database.

closed 2020-07-25 by kizniche