kizniche/Mycodo #801: Data error with 8.6.3

kpslc, 2020-07-25:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

The dashboard graph is setup to show pH, EC... and the pump actions. After upgrading to 8.6.3, the dispense by volume bug is resolved, but the manual dosing that was done to test the system did not show in the graph.

Looked at the Daemon log and noticed the following errors:

2020-07-25 16:05:23,332 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:05:38,259 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:05:53,291 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:06:08,328 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:06:23,264 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:06:38,291 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

### Versions:

 - Mycodo Version: 8.6.3
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2020-07-25:
```'NoneType' object has no attribute 'next'``` occurs after the controller has already crashed, and is not informative. You need to deactivate the offending controller, reactivate it, and copy the first error after the controller's activation line. This will be the error of interest.

kpslc, 2020-07-25:
Just focused on the pH probe, even though all of them seem to be affected. Here is the dameon log after deactivating and reactivating it.

AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:50:26,774 - INFO - mycodo.controllers.controller_input_d94716a6 - Deactivated in 48.3 ms
2020-07-25 16:50:34,253 - INFO - mycodo.devices.atlas_scientific_i2c_99_1 - Atlas Scientific Board: pH, Rev: 2, Firmware: 2.14
2020-07-25 16:50:36,820 - INFO - mycodo.controllers.controller_input_d94716a6 - Activated in 3207.9 ms
2020-07-25 16:50:38,276 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

kizniche, 2020-07-25:
I don't see any error for input d94716a6, only the activation line:

```2020-07-25 16:50:36,820 - INFO - mycodo.controllers.controller_input_d94716a6 - Activated in 3207.9 ms```

kpslc, 2020-07-25:
Tried it again: If this doesn't have what you need, please let me know the steps to follow to get you the data.

2020-07-25 16:58:08,257 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:58:09,887 - INFO - mycodo.controllers.controller_input_d94716a6 - Deactivated in 86.5 ms
2020-07-25 16:58:19,069 - INFO - mycodo.devices.atlas_scientific_i2c_99_1 - Atlas Scientific Board: pH, Rev: 2, Firmware: 2.14
2020-07-25 16:58:21,632 - INFO - mycodo.controllers.controller_input_d94716a6 - Activated in 3197.7 ms
2020-07-25 16:58:23,282 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:58:38,311 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:58:53,236 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-25 16:59:08,272 - ERROR - mycodo.controllers.controller_input_740ff562 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

kizniche, 2020-07-25:
Again, I only see one line for input d94716a6:

```2020-07-25 16:58:21,632 - INFO - mycodo.controllers.controller_input_d94716a6 - Activated in 3197.7 ms```

As far as I can tell, Input d94716a6 is working fine.

kpslc, 2020-07-25:
1. Why are there a the errors "attempting to read"
2. Why does the dosing no longer show up in the chart? The pH is going down, but the dosing isn't being charted.
[image]

kizniche, 2020-07-25:
Can you verify if input d94716a6 is or isn't working?

kizniche, 2020-07-25:
Input d94716a6 doesn't seem to have an issue. You need to be focusing on the input that is causing the errors. That is the input to follow my [recommendation](https://github.com/kizniche/Mycodo/issues/801#issuecomment-663912627) for.

kizniche, 2020-07-25:
> manual dosing that was done to test the system did not show in the graph.

volume or duration? How was the dispensing performed?

kpslc, 2020-07-25:
Tested 1ml (volume) and 5 (seconds) duration.

I'm not sure how to identify input d94716a6. Sorry to be a dork.

kizniche, 2020-07-25:
On the Data page, you can hover your mouse over the Input type button and the popup will show the full input ID. the abbreviated ID (above) is the first few characters.

I think I might have found your graph issue. Investigating now.

kpslc, 2020-07-25:
Thanks for the info on identifying objects. I'll check back in a few hours. Coming up on my tee time.

kizniche, 2020-07-26:
Fixed in 8.6.4 release

closed 2020-07-26 by kizniche