kizniche/Mycodo #802: Duration pump events not charting

kpslc, 2020-07-27:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

Updated to 8.6.4. The duration events from any of the pumps displays in a dashboard individual measure widget but not in the chart widget

### Versions:

 - Mycodo Version: [e.g. 8.6.4]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

In the attached screenshot you can see that 5ml (volume) and 3 seconds (duration) of pH down were manually pumped from the data/output screen for the pH pump.

Also notice in the chart that it was zoomed to 1 minute after both doses were manually added. Only the 5ml volume event is charted. The second event (duration) occurred 15 seconds after the volume event.

Also since we are still testing before going into production, we refreshed our 55 gal supply barrel with clean water. Then restarted the system. The nutrient pump A has been dosing every 360 second while we did the manual test on pH down. None of the duration events from the conditional statement using the nutrient are showing in the chart.
[image]  

This second screen shot was taken 13 minutes after the first. The widgets for nutrient a show no volume and do so duration. I didn't do a manual volume test on Nutrient A, but allowed the conditional to run. The pH is below the danger zone so per the code, it is only pumping Nutrient till the nutrient level is green.

Hope this helps.

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2020-07-27:
You have only selected to view the pump volumes. You did not select the pump duration measurements.

kpslc, 2020-07-27:
I'll add it. Thanks. Here is the log errors I think you were looking for yesterday. Is this error a problem?
2020-07-27 11:25:21,355 - INFO - mycodo.daemon - Mycodo daemon started in 17.647 seconds
2020-07-27 11:25:21,359 - INFO - mycodo.daemon - 49.34 MB RAM in use
2020-07-27 11:30:17,201 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0
2020-07-27 11:35:17,167 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0
2020-07-27 11:40:17,129 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0
2020-07-27 11:45:17,110 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0
2020-07-27 11:50:17,361 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0
2020-07-27 11:55:17,335 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0
2020-07-27 11:57:12,934 - INFO - mycodo.controllers.controller_conditional_bcbe5d75 - Refreshing conditional settings
2020-07-27 11:57:23,351 - ERROR - mycodo.outputs.atlas_ezo_pmp_53b19669 - Invalid parameters: State: on, Output Type: sec, Volume: None, Flow Rate: 50.0

kizniche, 2020-07-27:
You'll have to give me some more context. I'm not sure what issue are you referring to. It's best you don't cross-post. I see dozens of issues a day.

kizniche, 2020-07-27:
It looks like you have something set to do something with the pump every 5 minutes. Can you provide more information?

kpslc, 2020-07-27:
Sorry, I agree.  I'll add it to a new thread. The duration was my error. It is working great now. We are purchasing the items now to run a hydroponics setup. This setup is for aeroponics towers. BTW, we have integrated a relay, optical sensors and a control valve for water level control. Let me know if you want to see information on it. It is working great.

closed 2020-07-27 by kpslc

kpslc, 2020-07-28:
You gave me a clue and I corrected the setting on my end.

Thanks.

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com <mailto:kevin.plumb@gmail.com>

kizniche, 2020-07-28:
Can you explain so others can avoid this issue or a patch can be made?

kpslc, 2020-07-29:
Kyle,

It was completely my error during setup. I was following the manual adding input devices ph, EC… I also added the float sensor per the manual. At the time I left the GPIO pin blank as it wasn’t yet connected. 

Then I created a new circuit with a relay, two optical sensors and a motorized ball valve to control the water level in our 55 gal supply tank. That was connected to  GPIO pin 19 and activated. BTW it is working perfectly. I can send a diagram of the circuit and more details if you are interested.

Once you mentioned how to identify object IDs with the mouse hover, I was able to find the original float sensor input device that was active but not fully setup.

Unrelated:

I’ve looked in the forum and manual to see how people may be handling a possible error condition that would be created if the water pump stops working. Then if the pH (or EC) happens to be out of range, the system would keep dosing based on the water currently in the sample container. Is this discussed in the manual or on the forum?

If not, I may add a bit of code to the function that doses the pumps to look for this error.

Thanks again for a great system. BTW: Our automated greenhouse is making a lot of positive waves with groups of long-time grower organizations in UT. Before their annual conference was canceled due to Covid-19, they asked if I would present along with their keynote speaker. Our greenhouse has quite a bit of tech including an earth battery to keep the environment cooler in the summer and warm on 10 degree days in the winter without electric or gas heaters.

Kevin Plumb
C: 801.414.8643
E: kevin.plumb@gmail.com <mailto:kevin.plumb@gmail.com>

kizniche, 2020-07-29:
There are a few ways to detect whether the pump has failed. You could use a current sensor to detect if current is flowing to the pump, however this will not tell you if the impeller has broken and the pump is still drawing power but not moving water. A safer method would be to put a flow meter on the line with the pump (as done in my hydroponic build), which you can then use to determine if there is low or no flow, and act accordingly (deactivate EC/pH conditional, email notification, etc.).

I'm always interested in projects that use Mycodo. Send as much info and photos as you can! Though, it might be better to make a new issue out of it and I'll create and add a tag like "Project Feature".