kizniche/Mycodo #803: Feature requests

trevhub, 2020-07-27:
I'm a huge fan of this project. If any of these features are already possible I apologize. 

Feature requests:
- ability to have a Day/week counter on the dashboard. I have a nutrient schedule that changes depending on what week of the grow I am in. It would be neat to have (week 3, day 6) as a reference. 
- ability to add a quick note from a dashboard text field. 
- energy usage since (date) on the dashboard. Id love to see a running power cost since starting a new grow.
- ability to insert an image on the dashboard to have a diagram overview of the system so people viewing various sensors will have an idea where they are and what they do. 
- save mycodo backups to a different location. USB, Google drive, samba share, somewhere off of the SD card.  
- camera time lapse disk space full notification.  

Thanks for you time, keep up the great work.

kizniche, 2020-07-27:
Hi. I've put a lot of effort implementing the ability to import custom Input, Output, and Controller modules so users can create these themselves rather than forking the repository and trying to figure out how to modify the system. There are other parts of Mycodo I'd like to make modular, including dashboard widgets... As it looks like most of your suggestions are dashboard widgets. I wouldn't likely use these widgets, which isn't to say they're bad, but it does make me not particularly want to spend the time needed to develop them. However, I do see utility in putting time into refactoring the widget system to allow importing custom widget modules. I'll take a look at it and assess the feasibility.

> save mycodo backups to a different location. USB, Google drive, samba share, somewhere off of the SD card.

You can already export the Mycodo settings database and the measurements database from the Import/Export menu.

> camera time lapse disk space full notification.

There is a Free Disk Space Input you can add, then create a Conditional Controller and have it monitor the Input. When it detects a measurement below a certain value, have it activate an Email Notification Action.

kizniche, 2020-07-28:
> save mycodo backups to a different location

My mistake. I misread this as settings backup. Yes, downloading of a backup would be a good feature. importing a backup would be a logical complimentary component as well. This shouldn't be too much work to implement.

trevhub, 2020-07-30:
I figured out the week counter. Used the date +%s command as an input and subtracted the grow start date from that, then converted it to weeks. 

I have a 'current disk space remaining gauge' I didn't even think to add an alert when it gets to a specific size remaining. I will do that.

Thanks for getting back to me.

kizniche, 2020-07-30:
Clever!

I just had the idea for a Python Code Dashboard Widget that will display output from Python code. That would open up a lot more possibilities for displaying data.

kizniche, 2020-07-30:
Sneak peak:

[image]

kizniche, 2020-07-30:
Had to refactor to include loop code and refresh code, since the process runs separate from the browser session (daemon). This allows the loop code to continue running when no browser sessions are open on the dashboard, and the refresh code to execute whenever the widget refreshes on the dashboard (in any browser session the widget is displayed).

[image]

kizniche, 2020-07-31:
As a test, I created a note entry with an image attachment, then created a Python Code widget with the following code using the note image URL, and I'll be darned, it worked flawlessly! Not quite the intended use, but it shows how versatile this new widget is.

[image]

kizniche, 2020-07-31:
In another test, I was able to acquire energy usage statistics! You can now format the data any way you wish, such as developing HTML to display a table, for instance. In the example, below, I'm merely displaying the raw data, but HTML tags will render in the widget boxes, as the <img> tag did, above.

[image]

kizniche, 2020-07-31:
Alright, this is the last successful experiment of the night. All these took some refactoring, but they now all work. This test was to see if I could generate energy usage for a particular time frame, then format the raw output to something more pretty than just the dictionaries (as seen above). So, I think this now checks off 4 of your initial feature requests... all possible with one widget!

[image]

kizniche, 2020-07-31:
Although these aren't very user-friendly, it shows that having access to a Python Code Widget makes most things possible without having to edit the Mycodo codebase. I'd consider these methods intermediate to advanced, but at least they can now be done.

trevhub, 2020-07-31:
absolutely blown away, thank you so much for taking the time to add this. Genuinely excited to implement this.

emuehlstein, 2020-08-02:
Python widget is awesome.  Thank you!

  Here's my new favorite addition to a PID tuning dash:

'''
log_lines = 30
log = "/var/log/mycodo/mycodo.log"

with open(log) as f:
    content = f.read().splitlines()

dashstring = ""
count = len(content)
for i in range(count-log_lines, count):
    dashstring += (content[i] + "<br/>")
return dashstring
'''

I should probably add some filtering to grep just the autotune lines.

kizniche, 2020-08-03:
Very nice! I've been working on refactoring the widget system so custom widgets can be imported. It's proving very difficult due to how complex the current widgets are and their current integration into Mycodo. Though, I am making progress. If I'm successful, this should be an incredibly useful feature.

kizniche, 2020-08-04:
I just had the Python Code Widget successfully work as a single-file module, in all parts of the system. It's a huge milestone for the proof of concept. Now I need to convert all the other widgets to modules and add a Widget Import page, then figure out how I'm going to properly upgrade everyone's database to the new format.

Here's the single-file Widget module if anyone is curious to see this monstrosity: [widget_python_code.py.txt](https://github.com/kizniche/Mycodo/files/5019485/widget_python_code.py.txt)

kizniche, 2020-08-05:
Converting the current widgets to the single-file module format has been surprisingly smooth. I just converted 3 in the past hour (screenshot). Though I suspect the graph widget will be especially tough, considering all the custom options I'll have to add for that particular widget.

[image]

infinisean, 2020-08-09:
Trevhub, I am also a huge fan of this project and I do I like these feature suggestions as well.

If I may add onto your existing list here, here are a couple more which are somewhat related (hopefully I'm not too late and you haven't put in a ton of work which would be a pain to re/un-do)

1.  A function to be able to "duplicate" a whole dashboard (including all widgets, locations, settings/colors/ranges, etc) exactly.

i.e.  when one spends a fair bit of time getting a dashboard all configured exactly the way we want it with many widgets and all their corresponding settings, we can simply duplicate the whole dashboard to a new one and then modify it as needed. Even if the two dashboards only have a few widgets in common, I believe having them pre-created (via duplication) would save a fair bit of time re-configuring all the widgets/settings and also guarantee that they are identical between the two dashboards for consistency purposes.

2. Dynamic line color in the graphs based on the current value, much like the angular gauge colors for value-ranges.

Not sure if this would be possible/feasible but I think it would be visually helpful (and pretty darn slick, to boot) to have the line color of a graph change dynamically based on the current value. 
Ideally, the color would change X-pixel by X-pixel, but if that isn't feasible, updating the color of the entire line based on the current Y value (vs. defined value-range / color relationships) would make it easier to interpret (at a glance) the severity  "Good / Mediocre / Bad / Oh-my-God-we're-all-going-to-die!"  of the reading from the sensor in question.

kizniche, 2020-08-09:
Duplication is something that's been discussed several times in the past and is not something I'm going to implement.

Multi-color graph series lines are not possible with Highcharts/Highstock, which is used to display graphs.

infinisean, 2020-08-09:
Understood.

I figured there was a pretty good chance the dynamic line functionality 
wouldn't be supported by whichever library you were using since It's not 
something I've really seen implemented anywhere else

kizniche, 2020-08-09:
Actually, this is possible: http://jsfiddle.net/BlackLabel/jmckag4q/

infinisean, 2020-08-09:
Oh wow, that is pretty slick

kizniche, 2020-08-10:
v8.7.0 was just released. With it, there's the ability to create and import custom widgets. I've shown how to create the ones listed in the original post of features, but have not attempted creating the note widget, though I know it to be now possible. I'm closing this issue, since the ability now exists for users to create their own widgets. The manual has been updated and a simple example widget exists in the examples folder along with the other widget modules that can be used as a reference (see manual for locations). If I find free time and have the desire, I may take a crack at creating an Add Note Widget, but at the moment I don't have the desire to put in the effort since the new widget system was such a large amount of work.

closed 2020-08-10 by kizniche

emuehlstein, 2020-08-11:
New release is awesome - thanks again Kyle!

I've searched the issues a bit for the rationale behind not implementing
dashboard duplication but I haven't found what I'm looking for.
Infiniseans' request speaks to one of my frustrations with Mycodo; creating
a series of dashboards that look the same is really tedious exercise.  For
example, if I want to create dashboards with the same layout for daily,
weekly, and monthly views it takes ages to make sure everything matches.

After some deliberation, I think what I'm really after is a way to script
creation of dashboards, inputs, outputs, etc.  This would provide a
mechanism for repeatable deployments while maintaining a level of
customization not possible with a full export/import and save a ton of
clicks.  Also, it would be nice to be able to share dashboard designs with
other users.  As I explore this possibility, do you have any
recommendations on approach or feasibility of extending the API or command
line client to include dashboard creation?

kizniche, 2020-08-11:
I came up with a simple method to duplicate a dashboard and all its widgets (tested and working):

```python
def clone_model(model, **kwargs):
    """Clone an arbitrary sqlalchemy model object without its primary key values."""
    # Ensure the model’s data is loaded before copying.
    try:
        model.id
    except Exception:
        return

    table = model.__table__
    non_pk_columns = [k for k in table.columns.keys() if k not in table.primary_key]
    data = {c: getattr(model, c) for c in non_pk_columns}
    data.update(kwargs)

    clone = model.__class__(**data)
    db.session.add(clone)
    db.session.commit()
    return clone

dashboard = Dashboard.query.filter(Dashboard.unique_id == form.dashboard_id.data).first()
widgets = Widget.query.filter(Widget.dashboard_id == dashboard.unique_id).all()

new_dashboard = clone_model(dashboard, unique_id=set_uuid(), name="New Dashboard")

for each_widget in widgets:
    clone_model(each_widget, unique_id=set_uuid(), dashboard_id=new_dashboard.unique_id)
```

It was easier than I imagined. I'll implement it later.

kizniche, 2020-08-11:
Regarding dashboard/widget creation via the API, I'm not opposed, but it would take effort. The API already has a lot of endpoints that could serve as examples if one would want to take on developing endpoints to create dashboards/widgets/inputs/outputs/etc.

kizniche, 2020-08-11:
I main focused on GET endpoints for querying data when I introduced the REST API to Mycodo, as this is most crucial for developing communication within/between Mycodo components or developing external dashboards (e.g. command line data monitor). I did intend to add more POST endpoints, but I haven't really had any interest shown for this until now. I really only develop what I find interesting or useful, or what's of popular demand (and if I'm also interested in developing it). I rarely have users submit pull requests for features, which is a point of contention, as I can't let all my time be taken up to write code and refractor the system for every request. This is why some requests, even if I think they're a good idea, take a long time to (or never) materialize.

kizniche, 2020-08-11:
I just created a new issue for this spacific topic. Please post any new messages regarding this topic there.