kizniche/Mycodo #805: PID Controller Output: Min On Duration, Max On Duration, Min Off Duration not saved

emuehlstein, 2020-07-28:
Unable to save PID controller output attributes (Min On Duration, Max On Duration, Min Off Duration).

### Versions:

 - Mycodo Version: 8.6.4
 - Raspberry Pi Version:  3B
 - Raspbian OS Version: Buster Lite

### Reproducibility

1. Deactivate existing PID controller
2. Edit controller Output Min, Max durations
2. Save
3. View saved form and observe system behavior
4. Durations not saved.

### Expected behavior

PID controller out durations should be saved.

### Daemon Log
```
Jul 28 22:51:30 martha gunicorn[465]: 2020-07-28 22:51:30,771 Exception during reset or similar
Jul 28 22:51:30 martha gunicorn[465]: Traceback (most recent call last):
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/pool/base.py", line 697, in _finalize_fairy
Jul 28 22:51:30 martha gunicorn[465]:     fairy._reset(pool)
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/pool/base.py", line 893, in _reset
Jul 28 22:51:30 martha gunicorn[465]:     pool._dialect.do_rollback(self)
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 543, in do_rollback
Jul 28 22:51:30 martha gunicorn[465]:     dbapi_connection.rollback()
Jul 28 22:51:30 martha gunicorn[465]: sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was created in thread id 1949430880 and this is thread id 1938551904.
Jul 28 22:51:30 martha gunicorn[465]: 2020-07-28 22:51:30,776 Exception closing connection <sqlite3.Connection object at 0x730a4920>
Jul 28 22:51:30 martha gunicorn[465]: Traceback (most recent call last):
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/pool/base.py", line 697, in _finalize_fairy
Jul 28 22:51:30 martha gunicorn[465]:     fairy._reset(pool)
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/pool/base.py", line 893, in _reset
Jul 28 22:51:30 martha gunicorn[465]:     pool._dialect.do_rollback(self)
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 543, in do_rollback
Jul 28 22:51:30 martha gunicorn[465]:     dbapi_connection.rollback()
Jul 28 22:51:30 martha gunicorn[465]: sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was created in thread id 1949430880 and this is thread id 1938551904.
Jul 28 22:51:30 martha gunicorn[465]: During handling of the above exception, another exception occurred:
Jul 28 22:51:30 martha gunicorn[465]: Traceback (most recent call last):
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/pool/base.py", line 270, in _close_connection
Jul 28 22:51:30 martha gunicorn[465]:     self._dialect.do_close(connection)
Jul 28 22:51:30 martha gunicorn[465]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 549, in do_close
Jul 28 22:51:30 martha gunicorn[465]:     dbapi_connection.close()
Jul 28 22:51:30 martha gunicorn[465]: sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was created in thread id 1949430880 and this is thread id 1938551904.

```

kizniche, 2020-07-28:
What version Mycodo?

kizniche, 2020-07-28:
I see the settings are not being saved, but I don't think the errors you posted are related.

emuehlstein, 2020-07-28:
I tried changing line 11 of database.py to read:
```
MYCODO_DB_PATH = 'sqlite:///' + SQL_DATABASE_MYCODO + '?check_same_thread=False'
```
and continued to experience the issue.

kizniche, 2020-07-28:
Okay, not I'm getting those errors. I'll investigate.

kizniche, 2020-07-28:
Thanks for the bug report. I just committed a fix that will be available in the next release. You can manually edit the changes and restart the frontend for it take effect.

not5, 2020-08-04:
I upgraded to the master and it's still not saving the min on duration max on duration and min off duration for a lowering PID

kizniche, 2020-08-04:
I see the issue. I'll push a fix. If you don't want lirc to attempt to update during an upgrade, you'll need to uninstall it and make sure no Mycodo controllers/Actions have been aded that depend on it.

Magnum-Pl, 2020-08-04:
I just upgraded to 8.6.4 and the Min On Duration, Max On Duration and Min Off Duration options aren't even available for me. Do you think the fix you mentioned will take care of this too?

Magnum-Pl, 2020-08-05:
I had to delete the PID controllers I had set up and reinstall them to get those options to show back up. Weird.

kizniche, 2020-08-05:
I suspect just saving once would have revealed the options. Since there are multiple different types out outputs now (on/off, volume, PWM), there needed to be different options for PID controllers for each type. The output needs to be first saved before the proper options can be displayed. I tried to do my best to update databases for all potential use cases, but that doesn't always work as expected. In any case, just pressing save once should have fixed it.

kizniche, 2020-08-05:
I haven't had a chance to commit the fix, but it's merely adding this line:

```html
<input type="hidden" id="lower_output_type" name="lower_output_type" value="{{each_pid.lower_output_type}}">
```

After the following line in Mycodo/mycodo/mycodo_flask/templates/pages/function_options/pid.html:

https://github.com/kizniche/Mycodo/blob/edbce90c8e1b524c9f87681ae7f812e6497b8f09/mycodo/mycodo_flask/templates/pages/function_options/pid.html#L270

Just mentioning it in case you need to make the edit before I can push the fix/make a release. I'm in the middle of a large refactor that could delay release for a few weeks.

not5, 2020-08-05:
Thanks. I'll make that change

closed 2020-08-10 by kizniche