kizniche/Mycodo #806: Backup function broken in latest release

infinisean, 2020-08-04:
When attempting to create a backup, it says started successfully but no backups are ever created and nothing shows up in the logs, even with daemon logging enabled.

kizniche, 2020-08-08:
I was not able to replicate this issue. Backups are working fine for me on version 8.6.4. Additionally, backup log lines will appear in the Backup Log, not the Daemon log.

infinisean, 2020-08-08:
Ah, gotcha. Here is the relevant log entry then. Looks like it is looking for /home/pi/Mycodo instead of /home/pi/Mycodo-master. Could it be hard-coded somewhere?

[2020-08-02 08:04:36] 
[2020-08-02 08:04:36] #### ERROR ####
[2020-08-02 08:04:36] There was an error detected while creating the backup. Please review the log at /var/log/mycodo/mycodobackup.log[2020-08-03 17:51:02] 
[2020-08-03 17:51:02] #### Create backup initiated 2020-08-03_17-51-02 ####
[2020-08-03 17:51:02] Backing up current Mycodo from /home/pi/Mycodo to /var/tmp/Mycodo-backup-2020-08-03_17-51-02-/home/pi/Mycodo-master/mycodo/scripts/mycodo_backup_create.sh: line 19: /home/pi/Mycodo/env/bin/python: No such file or directory...rsync: link_stat "/home/pi/Mycodo" failed: No such file or directory (2)
[2020-08-03 17:51:02] rsync: change_dir#3 "/var/tmp/Mycodo-backup-2020-08-03_17-51-02-/home/pi/Mycodo-master/mycodo/scripts/mycodo_backup_create.sh: line 19: /home/pi/Mycodo/env/bin" failed: No such file or directory (2)
[2020-08-03 17:51:02] rsync error: errors selecting input/output files, dirs (code 3) at main.c(720) [Receiver=3.1.3]
[2020-08-03 17:51:02] Failed: Error while trying to back up current Mycodo install from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2020-08-03_17-51-02-/home/pi/Mycodo-master/mycodo/scripts/mycodo_backup_create.sh: line 19: /home/pi/Mycodo/env/bin/python: No such file or directory.
[2020-08-03 17:51:02] Mon 03 Aug 2020 05:51:02 PM EDT

infinisean, 2020-08-08:
More output:

pi@MushPi2:~ $ ln -s Mycodo-master/ Mycodo
pi@MushPi2:~ $ tail -f /var/log/mycodo/mycodobackup.log
[2020-08-08 17:34:44] #### ERROR ####
[2020-08-08 17:34:44] There was an error detected while creating the backup. Please review the log at /var/log/mycodo/mycodobackup.log[2020-08-08 17:40:57]
[2020-08-08 17:40:57] #### Create backup initiated 2020-08-08_17-40-57 ####
[2020-08-08 17:40:57] Backing up current Mycodo from /home/pi/Mycodo to /var/tmp/Mycodo-backup-2020-08-08_17-40-57-8.6.4...Done.
[2020-08-08 17:40:57] Moving /var/tmp/Mycodo-backup-2020-08-08_17-40-57-8.6.4/Mycodo to /var/Mycodo-backups/Mycodo-backup-2020-08-08_17-40-57-8.6.4...mv: cannot stat '/var/tmp/Mycodo-backup-2020-08-08_17-40-57-8.6.4/Mycodo': No such file or directory
[2020-08-08 17:40:57] Failed: Error while trying to move /var/tmp/Mycodo-backup-2020-08-08_17-40-57-8.6.4/Mycodo to /var/Mycodo-backups/Mycodo-backup-2020-08-08_17-40-57-8.6.4.
[2020-08-08 17:40:57] Sat 08 Aug 2020 05:40:57 PM EDT
[2020-08-08 17:40:57]
[2020-08-08 17:40:57] #### ERROR ####
[2020-08-08 17:40:57] There was an error detected while creating the backup. Please review the log at /var/log/mycodo/mycodobackup.log

ls /var/tm      ^C
pi@MushPi2:~ $ ls /var/tmp/
Mycodo-backup-2020-08-08_17-40-57-8.6.4  systemd-private-35e0068e0330440b9f9325b44c65f0a0-systemd-timesyncd.service-uPakYg
pi@MushPi2:~ $ ls /var/tmp/Mycodo-backup-2020-08-08_17-40-57-8.6.4
.bash_history              Bookshelf/                 Desktop/                   .gnupg/                    Music/                     Mycodo-master/             Public/                    Templates/                 .Xauthority
.bash_logout               .cache/                    Documents/                 .local/                    Mycodo/                    Pictures/                  Raspi-Driver-SCD30-master/ Videos/                    .xsession-errors
.bashrc                    .config/                   Downloads/                 master.zip                 Mycodo.git                 .profile                   .ssh/                      .wget-hsts                 .xsession-errors.old
pi@MushPi2:~ $ ls /var/tmp/Mycodo-backup-2020-08-08_17-40-57-8.6.4 ^C
pi@MushPi2:~ $ ls /var/tmp -a;l
.  ..  Mycodo-backup-2020-08-08_17-40-57-8.6.4  systemd-private-35e0068e0330440b9f9325b44c65f0a0-systemd-timesyncd.service-uPakYg
-bash: l: command not found
pi@MushPi2:~ $ ls /var/tmp -al
total 12
drwxrwxrwt  3 root root 4096 Aug  8 17:40 .
drwxr-xr-x 13 root root 4096 Jul 28 10:30 ..
lrwxrwxrwx  1 pi   pi     14 Aug  8 17:40 Mycodo-backup-2020-08-08_17-40-57-8.6.4 -> Mycodo-master/
drwx------  3 root root 4096 Aug  8 16:55 systemd-private-35e0068e0330440b9f9325b44c65f0a0-systemd-timesyncd.service-uPakYg
pi@MushPi2:~ $

kizniche, 2020-08-08:
If you have Mycodo installed in /home/pi/Mycodo-master, you likely didn't install it by the recommended method in the README.

infinisean, 2020-08-08:
You're probably right... but shouldn't it be able to install in any directory and not break the backup function?

kizniche, 2020-08-08:
No, and I'll only support issues when installed by the recommended method.

closed 2020-08-08 by kizniche

infinisean, 2020-08-08:
Fair enough.  I moved it to Mycodo and that resolved the issue.

Thanks