kizniche/Mycodo #807: Measurement Conversion Causes pigpiod to Break on 8.6.4

Magnum-Pl, 2020-08-04:
Hey Kyle, 

Converting the temperature from C to F on a DHT22 caused gpiod to stop working. I am unable to view any data after the conversion. I am still unable to view any data after reversing the conversion or deleting the input and making a new one. 

The Daemon Log shows this:

`ERROR - mycodo.inputs.dht22_1a0d7d62 - Could not connect to pigpiod. Ensure it is running and try again.`
`ERROR - mycodo.controllers.controller_input_1a0d7d62 - StopIteration raised 3 times. Possibly could not read input`

kizniche, 2020-08-04:
The conversion system has no impact on pigpiod. Try uninstalling and reinstalling pigpio from Configure -> Pi.

Magnum-Pl, 2020-08-05:
That fixed it. Thanks.

kizniche, 2020-08-05:
I'm not sure how it got messed up during the upgrade, but good to hear that fixed it.

closed 2020-08-10 by kizniche