kizniche/Mycodo #808: New Input: VL53L1X Time-of-Flight (Distance) Sensor

cedmo8, 2020-08-06:
I can see you currently support the VL53L0X, would it be possible to add support for the newer version (VL53L1X)?

The VL53L1X has a significantly longer range, my hope is to use it to measure and calculate the volume of a kerosene tank using either surface reflection (hopefully) or a PVC tube & float if that fails.

I have one on hand and tried it with the config for a VL53L0X sensor but it just shows -1mm

VL53L1X Datasheet: https://www.st.com/resource/en/datasheet/vl53l1x.pdf
Resources: https://www.st.com/en/imaging-and-photonics-solutions/vl53l1x.html#resource

kizniche, 2020-08-07:
Here's an Input module I threw together. Rename it from .py.txt to .py and import it into Mycodo from the Configuration -> Input menu, and give it a test.

VL53L1X Input Module: [vl53l1x.py.txt](https://github.com/kizniche/Mycodo/files/5039030/vl53l1x.py.txt)

kizniche, 2020-08-07:
If you encounter any errors, could you paste then from the Daemon log ([Gear Icon] -> Mycodo Logs -> Daemon Log)?

kizniche, 2020-08-10:
I went ahead and added this to Mycodo in the latest release, just released. It's currently untested, so if you get around to testing, please let me know if it works or if it doesn't, include the errors from the Daemon Log so I may try to fix it.

cedmo8, 2020-08-10:
I have tested it on a pi 4 and a pi zero over a few days and it works perfectly so far, furthermore, I was as hoped able to measure fluid level changes in a bucket of water by mounting it above the bucket at 90 degrees, the sensor does reflect off the surface of the water and with a bot of math can be used to report the volume of liquid in the bucket.

It is worth mentioning that the default address is 0x29 and I don't believe it is re-addressable like its predecessor.

kizniche, 2020-08-10:
The VL53L1X Python module has functions to change the address. Can you attempt to change the address with the included Change Address function in the Input module? Also, the datasheet states the I2C address is 0x52 (although the library uses 0x29 in its examples). The datasheet also doesn't mention that the I2C address can be changed. Pretty confusing.

kizniche, 2020-08-10:
Also, very good use of a proximity sensor. You can add a new measurement "Volume" with an id such as volume_proximity, and create a conversion From distance to volume_proximity, with an equation such as "x*50" (or whatever the amount per unit distance is). Then you can use that conversion in the Input to have it automatically convert and display the units in your desired volume.

cedmo8, 2020-08-11:
I tried several times to change the address and it has no effect, no errors in log.

Good pointer on the measurements I will look at that, I had just added them as `Data -> Math` 

I passed the mm reading from the sensor through a past, average first to smooth out fluctuations caused by surface ripples and the ~5mm margin of error the sensor sometimes reports and then into an equation to get the value, I settled on `([Distance from sensor to bottom of bucket]-x)*[Volume in liters for 1mm of my bucket]` my bucket is a perfect cylinder, I imagine the formula would be much more complicated for a tapered bucket or reservoir.

kizniche, 2020-08-11:
Ah, I wasn't thinking wen I wrote my equation. Volume increases as distance decreases. Thanks for the feedback. I've removed the option to set the address and changed the default I2C address.

closed 2020-08-11 by kizniche