kizniche/Mycodo #811: Lower PID functions not working, Raise works fine

infinisean, 2020-08-08:
Latest Mycodo release.

I created 3 PID functions as follows:
Measurement: Temp,  Direction: Lower,  Output: Water pump
Measurement: CO2,  Direction: Lower,  Output: Air Pump (through a co2 scrubber)
Measurement: Humidity,  Direction: Raise,  Output: Humidifiers

After configuring all three, enabling debug logs and starting the auto-tune, the only one showing up in the logs is the Humidity... the two "lower" functions never trigger even though the measurement is above the setpoint.

kizniche, 2020-08-08:
Autotune is an experimental feature and should not be relies on to tune the PID controllers.

infinisean, 2020-08-08:
> Autotune is an experimental feature and should not be relied upon to

Understood.

If the functions are set to active, is there any reason why they 
wouldn't trigger the output when the measurement is higher than the 
setpoint?

kizniche, 2020-08-08:
If autotune is not active, your PID tuning will determine how the controller operates the output.

infinisean, 2020-08-08:
ok, I found one issue (my fault) and corrected it... then when I 
configured the PIDs manually, I am seeing the below 3 errors in the logs 
repeating every few seconds for each of the 3 PIDs.
Notice the character it references during its failure to parse the query 
is unique to the PID.   i.e. this group of 3 errors keeps repeating in 
order.

Please let me know what else I can get for you to help troubleshoot 
this.   Thanks!

2020-08-08 18:22:50,197 - ERROR - mycodo.controllers.controller_pid_a9c10f0c - Exception executing check_pid() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 122, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 297, in check_pid
     duration_sec=self.setpoint_tracking_max_age)
   File "/var/mycodo-root/mycodo/utils/influx.py", line 468, in read_last_influxdb
     measure=measure, duration_sec=duration_sec, value='LAST')
   File "/var/mycodo-root/mycodo/utils/influx.py", line 440, in read_influxdb_single
     last_measurement = client.query(query).raw
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 518, in query
     expected_response_code=expected_response_code
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 369, in request
     raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 400: {"error":"error parsing query: found s, expected ; at line 1, char 151"}

2020-08-08 18:22:51,495 - ERROR - mycodo.controllers.controller_pid_537fe0c7 - Exception executing check_pid() on attempt 2 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 122, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 297, in check_pid
     duration_sec=self.setpoint_tracking_max_age)
   File "/var/mycodo-root/mycodo/utils/influx.py", line 468, in read_last_influxdb
     measure=measure, duration_sec=duration_sec, value='LAST')
   File "/var/mycodo-root/mycodo/utils/influx.py", line 440, in read_influxdb_single
     last_measurement = client.query(query).raw
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 518, in query
     expected_response_code=expected_response_code
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 369, in request
     raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 400: {"error":"error parsing query: found s, expected ; at line 1, char 148"}

2020-08-08 18:22:55,871 - ERROR - mycodo.controllers.controller_pid_41d40219 - Exception executing check_pid() on attempt 3 of 3.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 122, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 297, in check_pid
     duration_sec=self.setpoint_tracking_max_age)
   File "/var/mycodo-root/mycodo/utils/influx.py", line 468, in read_last_influxdb
     measure=measure, duration_sec=duration_sec, value='LAST')
   File "/var/mycodo-root/mycodo/utils/influx.py", line 440, in read_influxdb_single
     last_measurement = client.query(query).raw
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 518, in query
     expected_response_code=expected_response_code
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 369, in request
     raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 400: {"error":"error parsing query: found s, expected ; at line 1, char 142"}

kizniche, 2020-08-08:
Can you provide a screenshot of one of the PID controller's options? Keep in mind you can only attach images when making a post from a browser on github.com (email attachments will not work).

infinisean, 2020-08-08:
Sure thing.  Here you go.
Let me know what else I can do to help...   Or if it's something obvious/stupid that I overlooked.   Thanks again

[image]
[image]

kizniche, 2020-08-08:
There are two issues here:

1. You have setpoint tracking enabled to follow Channel 0 of Input 01, which is also the same channel and input for your measurement input. This is essentially like saying "I want to regulate temperature from input X with a setpoint that is X". that is your setpoint will always equal exactly what your input measurement is, and the PID controller will never actuate without output because your setpoint will never deviate from your input value. You need to disable setpoint tracking or change the setpoint tracking to any other measurement than your PID input measurement.

2. There is a minor bug in the input/math setpoint tracking code that I just pushed a fix for and will be available in the next release.

infinisean, 2020-08-08:
ok, I misunderstood what setpoint tracking was then. I thought having that disabled was the issue before that I "fixed" but apparently not. So, before (and now), with that off and everything else the same, I believe it should be activating the pump but that is not the case.

Also, when checking the logs (after disabling setpoint tracking) I noticed this:   (though it could have been when I did a manual disable, as opposed to the PID controller...)

2020-08-08 19:17:30,493 - DEBUG - mycodo.controllers.controller_output - output_on_off(5da78815-4725-44a8-9527-4630d2210c8d, on, sec, 10.0, 0.0, True)
2020-08-08 19:17:30,494 - DEBUG - mycodo.controllers.controller_output - Output 5da78815-4725-44a8-9527-4630d2210c8d (Humidifier) on for 10.0 seconds.
2020-08-08 19:17:31,629 - ERROR - mycodo.trigger_action_81ae39b2 - Error triggering action:
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/function_actions.py", line 1017, in trigger_action
    message = action_activate_controller(cond_action, message)
  File "/var/mycodo-root/mycodo/utils/function_actions.py", line 637, in action_activate_controller
    id=controller_entry.id,
AttributeError: 'Query' object has no attribute 'id'

kizniche, 2020-08-08:
> AttributeError: 'Query' object has no attribute 'id'

That error is likely because your Activate Controller action doesn't have the Controller ID option set.

> I believe it should be activating the pump but that is not the case

What led you to this conclusion? How are you verifying the output is not activating?

infinisean, 2020-08-08:
The pump is right next to me. It turns on manually but it has never 
turned on without me activating it manually.

I also never see the indicator light go on or anything in the debug logs 
indicating it thinks it is activating.

kizniche, 2020-08-08:
Can you add a PID Control widget to your dashboard and take a screenshot of it?

kizniche, 2020-08-08:
You should be receiving debug lines in the log since you have debug logging enabled, but I just found another minor bug preventing the logging level from switching to debug (fix being pushed shortly). You should be seeing lines such as this that help with figuring out issues:

```
2020-08-08 19:44:21,081 - DEBUG - mycodo.controllers.controller_pid_517b658e - Latest (CH0, Unit: F): 138.3296 @ 2020-08-08 19:44:08
2020-08-08 19:44:21,082 - DEBUG - mycodo.controllers.controller_pid_517b658e - PID: Input: 138.3296, Output: P: -53.3296, I: -0.0, D: 0.0, Out: -53.3296
2020-08-08 19:44:21,337 - DEBUG - mycodo.controllers.controller_pid_517b658e - Setpoint: 85.0 Output: -53.3296 to output bbdba17d-4203-45b4-9c14-b11852edaed5
```

infinisean, 2020-08-08:
I just lost power due to a storm in my area.  
I’ll get back to you as soon as I can. 

Thanks again for the assistance!

infinisean, 2020-08-09:
I removed all the PID functions and re-added them without touching the 
auto-tune this time and they appear to be working.

I did notice that the Min-On, Max-On, Min-off values don't seem to stick 
after saving inside a PID config. (raise or lower).   I can open a new 
bug for that, if you like.

In case you are interested, this is the error I was seeing when 
attempting to start auto-tuning (I'd love to see this working to play 
with it a bit, if not to rely on it so let me know if I can help further 
test/troubleshoot with you):

2020-08-08 19:56:22,247 - ERROR - mycodo.controllers.controller_pid_e19a5e5e - Exception executing check_pid() on attempt 3 of 3.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 122, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 313, in check_pid
     if not self.autotune.run(self.last_measurement):
AttributeError: 'NoneType' object has no attribute 'run'

Thanks again for all the help!

-Sean

kizniche, 2020-08-09:
Again, autotune is an experimental feature and I'm not supporting any issues related to it.

kizniche, 2020-08-09:
> the Min-On, Max-On, Min-off values don't seem to stick after saving inside a PID config. (raise or lower). I can open a new bug for that, if you like.

This has already been opened in #805 and subsequent duplicate issues. Always look in the open and recently closed issues for duplicate issues before opening one.

infinisean, 2020-08-09:
Completely Understood.

I didn’t realize the min/max values were part of the auto-tune.
I thought they were part of the normal PID config because they are located
above the “Autotune” label.

I definitely wouldn’t have bothered you with it at all if I had known it
wasn’t supported due to your previous, understandable reply. So, my sincere
apologies.

In any case, If there’s anything I can do (when it comes to testing or
whatever) to help you improve/finish the auto-tune feature, please don’t
hesitate to ask.
Seems like a really cool feature once completed.

Everything else is running like a charm now, so thanks again for all your
help and work on this project. It really is an amazing piece of work.

Cheers, mate!

infinisean, 2020-08-09:
Will do.

kizniche, 2020-08-09:
> I didn’t realize the min/max values were part of the auto-tune.

They're not autotune options. I've already committed fixes for the saving bug, which won't be available until the next release. Alternatively, you can manually make the edits linked in the above issue, or install an older version that doesn't have this issue.

infinisean, 2020-08-09:
Ahh, I see what you were referring to now...  in the logs: 
"self.autotune.run"

Again, my apologies, I missed that line so I didn't realize this was 
auto-tune and not regular PID related. I'll look more closely next time 
(and check for duplicate issues) before sending anything your way.

Thanks man. Enjoy your weekend!

kizniche, 2020-08-09:
Thanks for the bug reports. I've been thinking about pulling the autotune feature out of the PID controller and adding it to its own controller for simplicity sake. Not something I'll be doing soon, but it's on my radar.

infinisean, 2020-08-09:
Sure thing.

And Yea, that's not a bad idea... and you might want to throw some 
"Unsupported / Beta feature" verbiage around it (Big, Bright, Blinking 
Blue/red lettering)  to Bar Bozos like me from Bombarding you with Banal 
Bug Bulletins. Bazinga!

:)

kizniche, 2020-08-09:
Well, that didn't last long. Once I realized how easy it would be to pull the PID autotune into its own controller, I just had to do it. So, it's been done and will be available in the next release. It currently only operates with an output to raise a condition, but at least it's now self-contained in it's [own controller](https://github.com/kizniche/Mycodo/blob/33f2658087fcc44f8b7864b8500474645733bea1/mycodo/functions/function_pid_autotune.py) and can be experimented with.

infinisean, 2020-08-09:
That is awesome!

kizniche, 2020-08-10:
For those interested in the initial issue started for when I implemented the autotune function, see #508. It has some really nice graphs from when I tested the autotune-generated gains.

closed 2020-08-10 by kizniche