kizniche/Mycodo #813: Add support for multi-channel outputs

kizniche, 2020-08-12:
When Mycodo was conceived, the only outputs supported were turning individual GPIO pins on and off. This allowed Outputs to be very simple in their design. As the software developed, though, more complex Outputs were introduced, necessitating a more robust framework to allow their use. Currently, an output controller that is responsible for actuating each output will send a command to the output Python module to either turn the output on (optionally along with a value) or off. If a value is sent, a value type must also be sent, indicating what the value is (e.g. 25 seconds, 30 ml, 55 % duty cycle). However, these output modules were still operating with one output device per output module.

Currently, an output module cannot operate a single device with multiple output channels. Example of devices that allow control of multiple output channels would be with the PCF8574, MCP23008, MCP23017, and DS2408 chips, to name a few. These chips enable the control of turning up to 8 output pins high or low, enabling the control of 8 devices from one chip. Although it's possible to merely add 8 output modules and individually configure each one to communicate with the chip to command one of the 8 pins, this is inefficient and costly (time, RAM, CPU, etc.). A more elegant solution is to build one output module that enables control all 8 pins. This requires a refactor of the output system to enable a "output channel" parameter to be passed to the output module to indicate which particular output channel of the output device is to be actuated.

Since there can be multiple measurement channels per output already (e.g. the peristaltic pump output has measurement Channel 0: Pump On Duration, measurement Channel 1: Dispense Volume, and measurement Channel 3: Dispense Duration, all for the same common output of dispensing water), using the measurement channel designator will not work, since multiple measurement channels can be assigned to each specific output channel of the output module. The most logical solution I can come up with to maintain compatibility and allow a smooth upgrade process for users is to merely add an option to the OUTPUT_INFORMATION dictionary of the output module that indicates how many output channels the output module is in control of. Then, in the Output Controller, an "output channel" parameter will be passed to the output module to indicate which output channel to actuate. The output module can then determine which measurement channels to save measurements to based on which output channel is being operated.

Reference: https://kylegabriel.com/forum/general-discussion/review-of-a-nice-8-channel-i2c-solid-state-relay-board-available

ticomyco, 2020-08-12:
Here is an example bash script (similar to the previous bash script I used to keep track of the state for PCF8574-based boards like the Krida relay board).

https://github.com/ticomyco/mcp23017-mcp23008-scripts/blob/master/mcp23017_relay_v1.sh

Whereas the PCF8574 (and probably 8575 too) can't have its state accurately read when a relay is connected to it (because that relay will tend to pull the pin low regardless of whether the PCF8574 is instructed to hold the pin high or low), thankfully the MCP23017 and MCP23008 chips include a separately accessible GPIO register (to read the state of a pin if it's configured as an input) and an OLAT register to write *and* read the desired state for the output drivers to set pin levels, so with those chips it's not necessary to keep a statefile or state variable.

kizniche, 2020-08-12:
You touched on another benefit of working from a single output module to control all output channels, rather an a different output module for each channel, and that is a single module can more easily keep track of the output channel states than separate output modules. I'll work on refactoring the output framework to allow multi-channel output modules and develop a multi-channel output module for the PCF8574, since this would really benefit more than other chips from this type of module, since the pin states aren't able to be read, and hence would need to be stored within the module.

ticomyco, 2020-08-12:
I agree 100%  -- especially for deciding initial relay states at power-on, or for chips that require an initialization of some sort (though I suppose even with the MCP230xx you could just set the IODIR to an output before every pin change, but that seems sloppy and I suppose there may be some chips where that approach isn't feasible/possible). Thanks again for all the help!
I'm sure you already got the notification, but I posted on your forum some hastily-written directions on how to use these chips in conjunction with the ULN2803a and my shell scripts. Once I get things looking a bit less ugly I'll try to take some pictures and put up a wiring schematic on a blog.

kizniche, 2020-08-14:
It took a few days, but I finally got the new output system working, at least on the Outputs Page where the user can turn the output channels on and off, and have it successfully communicate with the daemon to then instruct the output module to turn the appropriate channel on/off.

[image]

ticomyco, 2020-08-15:
Oh awesome! I was actually just about to ask if I should try to write a module for the mcp230xx as a "some single output for each channel" in python the way it has worked before and using your pcf8574 code as a template, or wait for an example multichannel output module and associated framework to exist before starting....
I'd be happy to take a crack at the mcp23017 and mcp23008, however the 16bit chip handles the outputs as two separate 8bit banks, and I suppose a user might possibly want to use one bank as a set of inputs and the other as a set of outputs. Do you have an opinion of whether we should try to handle that condition, or just treat them as outputs only?

kizniche, 2020-08-15:
There's still a lot of changes happening to the output system, so I wouldn't advise attempting to create any type of output module. The new pcf8574 output module will be a good template once I commit the changes I've been working on. I don't want to commit it just yet, as there are many parts of the output system that are still broken and need updating. I also need to think about how certain parts should operate before deciding on a particular method, as I want to do this once the most logical way rather than realizing I should have done it another way had I taken more time to consider the options.

I don't think it's a good idea to mix inputs with outputs. They should remain separate and we can experiment with developing an Input module with an Output module so they can play nicely together. Both the Output and Input system allow modules to be imported, so we should be able to figure something out.

ticomyco, 2020-08-15:
That makes sense. I'll hold off on trying to piece something together for
now, until you've got that pcf8574 module working as a multichannel output.

After more reading, it appears that with the mcp230xx chips it is possible
to initialize the iodir register(s) with a bitmask to specify individual
pins as inputs or outputs, rather than having to set an entire 8bit/channel
group as entirely outputs or inputs.

But I imagine, that if we have one half of an mcp23008 chip intialized with
0x0f as the iodir mode, so the lowest four pins are inputs and there's a
matching 4-channel Mycodo input module for those bits, and then another
output module using the same iodir mask to get the highest four bits to be
output pins, that there would have to be a bit of additional sanity
checkingbin Mycodo to make sure that the separate input and output modules
talking to the same chip don't clobber each other with initializing the
iodir register with conflicting masks.

kizniche, 2020-08-15:
I'm beginning to see this is a lot more involved and the effects from the changes are farther-reaching than I anticipated. I'm now considering scrapping what I've done so far and work within the current framework to attempt to build a multi-channel output module. There already exists a locking function, which can be used to gain exclusive access to an output device (e.g. PCF8574) and switch a pin and change an array of pin states to a central file or database entry that all PCF8574 modules (for the same chip) read/write. Configuration would simply entail adding 8 PCF8574 outputs and configuring them all for the same I2C address and bus, and they would then essentially access the same file/DB entry.

kizniche, 2020-08-15:
I just pushed changes to the multi_channel_output branch that essentially contains a pcf8574 output module that should work in unison with each other. They create a file /var/lock/pcf8574_bus_i2caddress.states that contains the state of each channel. Each output module will gain a lock of the device with a separate lock file, read the array, create a new array by changing only one channel, writing to the device, saving the file, and releasing the lock. I made some additional changes to the output controller, so that's why I created a new branch instead of merely pushing just the pcf8574 output module to master or attaching it here. So, if you want to test this, you'll have to clone the Mycodo repo, checkout the branch, then install normally by executing setup.sh in the install directory. You can move your existing install from ~/Mycodo to something like ~/Mycodo-backup to be able to easily so back to using it. Here's how I would do the test:

```bash
sudo service mycodo stop
sudo service mycodoflask stop
sudo mv ~/Mycodo ~/Mycodo-backup
clone https://github.com/kizniche/Mycodo
cd Mycodo
git checkout alt_output_mod
sudo /bin/bash install/setup.sh
```

Then you'll have the latest version of Mycodo from that branch installed and running. To go back to your previous backup you just made, do this:

```bash
sudo service mycodo stop
sudo service mycodoflask stop
sudo mv ~/Mycodo ~/Mycodo-backup-branch
sudo mv ~/Mycodo-backup~/Mycodo
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
```

And after it finished your original Mycodo install will be running.

ticomyco, 2020-08-15:
> clone https://github.com/kizniche/Mycodocd Mycodo

Awesome! I'll take a look when I get back to my devices, and I'll give you
an update and I'm sure a handful more questions. Thanks!

ticomyco, 2020-08-16:
Ok, so I've got Mycodo running on your alt_output_mod branch, and the PCF8574 output is working successfully (ie, enabling/disabling one channel does only affect one channel and leaves all other channels in their previous state), with one exception -- that the output order is reversed (relay channel 8 = mycodo channel 1, etc).
I double-checked my bash script that I had been using to control them via the i2cset commands, just to make sure that I hadn't done any swapping of bit order. 

If you'll forgive my bash, this is what I've used to control the pins in the correct order on both the PCF8574 and the MCP23008 chips, which both use active=1,inactive=0, and in this code snippet, the $1 variable is the channel number from 0-7:
---
             (( AMASK =  2 ** $1  ))
             case $2 in 
                "on") 
                    # use an OR with a positive bit mask to enable only the
                    # bit for the desired relay and leave others unmodified 
                    (( STATE |= $((AMASK)) ))
                    printf -v HEXSTATESTR '%x' $STATE # need to make a text
                        # string of the new state in order to call i2cset:
                    /usr/sbin/i2cset -y $BUS $ADDR 0x$HEXSTATESTR
                    #/usr/sbin/i2cset -y $BUS $ADDR $OLAT 0x$HEXSTATESTR
                    ;; # done enabling a relay
                "off")
                    # use an AND with a negated bit mask to disable only the
                    #  bit for the desired relay and leave others unmodified
                    (( STATE &= $((~ AMASK & 255)) ))
                    printf -v HEXSTATESTR '%x' $STATE # need to make a text
                        # string of the new state in order to call i2cset:
                    /usr/sbin/i2cset -y $BUS $ADDR 0x$HEXSTATESTR
                   #  /usr/sbin/i2cset -y $BUS $ADDR $OLAT 0x$HEXSTATESTR
                    ;;
                *)
---

ticomyco, 2020-08-16:
Okay, I found the issue -- I changed the order here and everything is working totally now.

pi@raspmush2:~/Mycodo $ diff /tmp/pcf8574.py mycodo/outputs/pcf8574.py 
206c206,207
<             ret.append(bool(state & 1 << 7 - i))
---
>             #ret.append(bool(state & 1 << 7 - i))
>             ret.append(bool(state & 1 << i))
245c246,247
<                 new_state |= 1 << 7 - i
---
>                 new_state |= 1 << i
>                 #new_state |= 1 << 7 - i
252c254,255
<         bit = 1 << 7 - output_number
---
>         bit = 1 << output_number
>         #bit = 1 << 7 - output_number

kizniche, 2020-08-16:
Great. Most importantly, though, when two outputs are used, do thet work together nicely or are their conflicts or other issues?

ticomyco, 2020-08-16:
They work correctly and do not conflict with each other --
I can  set all off, then just 1 on,
Then just 8 on (while 1 remains already enabled), then turn whichever
individual channel off, etc.

The only issue I had was my own user error-- when I was adding outputs for
8 channels, I forgot to select the channel# in the drop down on one, and
then on another I forgot to keep changing the bus #, since on my system the
pcf8574 is on bus 19. Is there a way to add 8 outputs at a time and have
the channel# autoincrement? (Or select a batch of 8 outputs and change the
bus# for all of them to match a desired number?)

Are you still inclined towards sticking with the "one channel per output"
model, versus the multichannel output model you'd started working on?

Unless you say otherwise I'll assume this output is usable, since I can
always fallback to my shell script outputs if something starts breaking..

kizniche, 2020-08-16:
Ideally a multi-channel output system would be ideal, but is looking more and more like I'll need to perform major changes in many parts of the system to implement it. I need to think on it before I start making those kinds of changes so I don't waste my time pursuing deadends.

kizniche, 2020-08-16:
> I don't know if your question about using two outputs specifically referred to two channels of the same PCF8574 chip, which was my original assumption, or if you meant the possibility of using two PCF8574 chips. At the moment I only have this one PCF8574 chip, but I assume that if I had a second PCF8574 chip running on a different bus or different address, that the code would handle it fine, since they would each have a unique lockfile.

Well, in this codebase, each output is a channel, so I was meaning when you add two outputs, which each would control a separate channel. Also, since the name of the lock and state files contains the I2C bus and address, there shouldn't be conflicts with multiple chips.

kizniche, 2020-08-16:
> Okay, I found the issue -- I changed the order here and everything is working totally now.

Thanks. I've incorporated your changes.

> Are you still inclined towards sticking with the "one channel per output"
model, versus the multichannel output model you'd started working on?

I haven't completely given up on the multi-channel option. I'm going to keep working on it and see how it goes. I understand now it's going to take a lot longer to complete than I initially anticipated. Now that I have a working model for the current system (alt-output branch), I'm more at ease that I have a fall-back plan. I'm going to keep progressing the multi-channel branch and at some point in the near future, I'll decide whether which branch I want to merge.

ticomyco, 2020-08-16:
Awesome -- that all makes sense. 
Two remaining questions -- 
1) if I wanted to keep running the alt-output branch, but on my production Pi (I have two at this point), is there a safe way to upgrade the production system from the current release to the alt-output branch, and still keep my previous sensor data history intact?

2) should I still hold off on trying to make a single-channel output for the MCP230xx chips based on your working PCF8574 code, and wait until you've made a decision about which path makes more sense, or should I give it a shot? As I posted on the forum, I've got working bash shell scripts for those chips, and I know I've got limited time to try to recode things in Python (which I'm still rusty at) but it seems like it should be fairly straightforward and possibly easier than the PCF8574 since the output states can be read directly from the chips. Either way I'm not in a rush since I have a working system, but I didn't want to waste effort either. 

I still haven't ordered any DS2408 or DS2413 chips, but those look pretty attractive to me based on my need/desire to control a handful of relays at a bit of a distance (and in what's apparently an EMI/RFI noisy environment), and I may bite the bullet and order some of those in the near future if I can get my temp-controlled soldering iron working again in order to be able to solder surface-mount chips successfully (versus my current fall-back iron which limits me to through-hole components for the most part). Does any of your rethinking of output models affect output modules based on a 1-wire interface vs I2C currently?

Also, I recently acquired a "HiLetgo 20x4  LCD display" and it appears to be based on the PCF8574 chip as well, however my initial attempts to get mycodo to output any text on it have failed entirely, however I haven't yet been able to eliminate the 5v-3.3v logic level converter as being a source of problems, and I'm also still searching for documentation for this product. Is the current LCD output module in mycodo based on that chip, or one of the others used to drive LCDs? I made a minimal search through the codebase but I wasn't totally sure one way or another what chips the current LCD output module supports yet, tho I could do a bit more homework.

kizniche, 2020-08-16:
> if I wanted to keep running the alt-output branch, but on my production Pi (I have two at this point), is there a safe way to upgrade the production system from the current release to the alt-output branch, and still keep my previous sensor data history intact?

The pcf8574 output module ***should*** be able to be imported into the latest release version and work fine. So, you can restore your production version, then take the output module from the alt-output branch and import it.

> should I still hold off on trying to make a single-channel output for the MCP230xx chips based on your working PCF8574 code?

I don't think I'll have something ready for release for at least another week, or longer, and even then, the version won't be ready for release for another week or more. I'd say I have a 90% chance of going with the multi-channel branch and will ultimately decide to use the multi-channel module version over the alt-channel module version. I wouldn't start making any module unless you're eager to experiment. If your current system is working, I don't see any benefit to building something new unless you need additional features your scripts can't accomplish.

> Is the current LCD output module in mycodo based on that chip, or one of the others used to drive LCDs?

You can find the modules for the LCDs here ([1](https://github.com/kizniche/Mycodo/blob/49caaaec7cee4230d5f075c3fd7f7a553a3e19f3/mycodo/devices/lcd_generic.py), [2](https://github.com/kizniche/Mycodo/blob/49caaaec7cee4230d5f075c3fd7f7a553a3e19f3/mycodo/devices/lcd_pioled.py)).

kizniche, 2020-08-16:
> Does any of your rethinking of output models affect output modules based on a 1-wire interface vs I2C currently?

I'm not sure what you're asking.

ticomyco, 2020-08-17:
Thanks! I'll hold off on doing anything else with the Mcp230xx chips for
now then.

Regarding the ds24xx 1-wire chips, I didn't know if the fact that they're
1-wire bus devices as opposed to i2c devices would factor in at all into
your redesigning multichannel outputs or the alt branch -- it wasn't a
specific question, in retrospect mostly thinking out loud I suppose, and
trying to determine that if I was to get one/some of those chips if I
should try to just write more bash scripts to control them or wait for a
python model that I could follow.

Anyhow probably making a big deal out of nothing. Is there anything I can
do to help at this point?
I was thinking about trying to put together a schematic for other Mycodo
users to use to be able to utilize the multichannel chips we've been
talking about in conjunction with some of the common relay boards and the
Darlington transistor arrays with the Mcp or pcf chips but I didn't know if
that would be useful documentation or what exactly would be most helpful
for your username.  I suspect that many users might not be building their
own circuits.

kizniche, 2020-08-17:
> I didn't know if the fact that they're 1-wire bus devices as opposed to i2c devices

The communication interface should have no effect on the module, since it's a module-level setting rather than a channel-level setting. For example, "turn channel on at startup" would be a channel-level setting that would need a separate setting for each channel.

> I was thinking about trying to put together a schematic for other Mycodo users to use to be able to utilize the multichannel chips we've been talking about in conjunction with some of the common relay boards and the Darlington transistor arrays with the Mcp or pcf chips

This would be useful for anyone

ticomyco, 2020-08-17:
Okay -- sounds good. I'll see what I can put together. Since my circuit building has so far been constrained to the MCP230xx chips I'll see what I can put together regarding those chips, whereas with the PCF8574 the only pics or schematics I could try to put together would be based on the Krida 8-channel relay board I've mentioned previously that uses the PCF8574 chip, or linking to other projects randomly available on the web. For text I'm happy to post on the forum or here on github, but for images what would be ideal for your workflow?

kizniche, 2020-08-17:
I have no real circuit-building workflow. I've heard good things about circuitlab.com ([link](https://github.com/kizniche/Mycodo/issues/184#issuecomment-344825791))

kizniche, 2020-08-17:
Also check out https://kizniche.github.io/Mycodo/Outputs/#schematics-for-dc-fan-control

kizniche, 2020-08-18:
Major progress on the multi-channel output system. I have outputs able to be created, settings saved (for output-specific and each channel-specific option), and deleted properly from the frontend. The frontend buttons (on, off, etc.) also work to control specific channels. The only thing left to complete is integrating the channels in the rest of the system, which includes form dropdown selections for outputs to be expanded to show every output channel (and all the processing of those selections by controllers), energy usage statistics calculations, and various other parts that reference output devices.

Perhaps the most difficult part may still be developing the upgrade script to upgrade user databases to use the new schema, since I will have to add a single channel for all current outputs, change all DB references for an output to include the first channel, then copy the settings that were once for the output itself to the single channel.

It seems I'm past the 50% point, things are working, and the morale is high.

[image]

ticomyco, 2020-08-18:
Wow! That's amazing!! You're staying busy!! Unfortunately I've not been
able to touch circuits or code today and most of yesterday at all, but
hopefully tomorrow I can at the very least create some schematics that
aren't drawn on notebook paper and might be usable documentation for
others...

Regarding the upgrades and the new schema, perhaps I'm being naive, but
would it be possible to leave the old data as it is, with outputs that were
"single output single channel" as one type of output, and then the new
multichannel outputs as a different output type, the same way that on/off
outputs are different than PWM outputs? Anyhow, I don't know if that would
make things easier or more difficult, or for that matter if that would
necessitate changing tons of other parts of the codebase, but the thought
just popped in my mind.

Anyhow, glad the morale is high! If you were nearby I'd offer home cooked
beer and mushrooms but for now I'll wish the you the best with your coding
:)

Btw I was going to order some lm317 regulators to build some DC pwm
controllers, but I found that the cost for complete regulated lm317 power
supplies (including heatsink, tiny PCB with capacitor and potentiometer,
and screw terminals) cost less for a pack of 5 on Amazon vs buying
individual components so I may see if the traces on the PCB are amenable to
easy modification to accept an external pwm signal versus just using the
on-board potentiometer to set voltage thresholds. If this works it could be
nice addition to your dc pwm fan control circuit.  We'll see...

Cheers for now!

kizniche, 2020-08-18:
> would it be possible to leave the old data as it is, with outputs that were "single output single channel" as one type of output, and then the new multichannel outputs as a different output type

No, because under the new system, all outputs, even if they have one channel, need to have both an output device and an output channel in order to be manipulated by the system. I could incorporate a check that if a channel isn't supplied by something desiring to manipulate it, it will default to channel 0, but I do not want to have to maintain two different database entry formats. If I ever need to update a user's database (which [happens often](https://github.com/kizniche/Mycodo/tree/49caaaec7cee4230d5f075c3fd7f7a553a3e19f3/databases/alembic/versions)), I want to know exactly what the entry format is, otherwise I'm adding complexity that could cause issues.

ticomyco, 2020-08-18:
Áhh okay, that makes sense. Definitely not a fan of unnecessarily adding
complexity

kizniche, 2020-08-19:
I've updated all references and refactored the code for PID controllers, Output Widgets, and Output Actions. Also converted the GPIO PWM Output.

Still need to work on Conditional and Trigger Functions, Energy Usage Statistics, pre-output selections (for Cameras, Inputs, etc.), and finally, the database upgrade script.

kizniche, 2020-08-20:
Completed refactoring Triggers and Energy Usage Statistics. I believe I have the database schema finalized, and everything should be in a working state, so if you wanted to test the branch, you can follow the commands in [this comment](https://github.com/kizniche/Mycodo/issues/813#issuecomment-674456153), but checkout the multi_channel_output branch instead of alt_output_mod. I'm now in the home stretch.

ticomyco, 2020-08-20:
Fantastic! I'm slammed at the moment, so I can't try that branch out quite yet, but I'll give it a shot as soon as I can. Unfortunately, I also just upgraded the production Pi from 8.6.1 to 8.7.1 and I noticed that the dashboard widgets are broken, as I believe a couple of other folks have experienced as well.

ticomyco, 2020-08-30:
Hi, I got around to tinkering with this again, and I ran into a problem. On my testing pi, I was running the alt_output_mod branch still, I tried to restore the original installation from the ~/Mycodo-backup directory using your directions in the previous comment. Unfortunately, when running the upgrade_post.sh script, it seems that it was unable to reactivate mycodoflask, and also it apparently failed to build lirc as well. 

Since this is a testing machine, I can blow it all away and start over if needed, but if it's possible to restore the backup I'd be interested in that. On a side note, I've got a handful of MCP23008 and MCP23017 chips driving several relay boards successfully using the ULN2803a darlington array, with the current being supplied from a separate +5V power supply so as not to tax the Pi's 5V rail, and it's been rock solid for the past couple weeks. None of the I2C bus interference problems I'd experienced earlier, so this looks like a useful combination for other Mycodo users. 
[restore1.txt](https://github.com/kizniche/Mycodo/files/5147482/restore1.txt)
[restore2.txt](https://github.com/kizniche/Mycodo/files/5147483/restore2.txt)

kizniche, 2020-09-09:
I must have missed your last comment. lirc doesn't currently successfully build, so that's not an issue. Also, after running upgrade_post.sh, mycodoflask won't automatically start. You'll need to run ```sudo service mycodoflask restart``` or reboot. I also just released v8.8.0 that has the new multi-channel outputs, and the PCF8574 output module included in the built-in set.