kizniche/Mycodo #814: Out of Disk Space

DarthPleurotus, 2020-08-13:
I have had Mycodo running for a while without any issues. I haven't been keeping an eye on it. Today I connected to my Raspberry Pi where Mycodo is installed and found that I am completely out of disk space. I am using Rasbian Lite and can't imagine what besides Mycodo would have filled up all the disk space. I have tried deleting some old Mycodo backups but even after doing so, the RPi reports that no space is available (when I use "df -h"). What could be causing this issue?

kizniche, 2020-08-13:
Try using ```ncdu```

DarthPleurotus, 2020-08-13:
Ok thanks! I gave that a go. Here are the first few results. At minimum, I will delete Simplicity Studio (which I no longer need). Could any data from Mycodo have accumulated over the past few weeks and contributed to the disk space filling up?

kizniche, 2020-08-13:
I don't see any results. It could be, but why speculate when you can determine exactly what the cause is? Free disk space issues are easily solved by determining what's taking up the space. ```ncdu``` is how you should determine that.

DarthPleurotus, 2020-08-13:
Got it. Are you able to see the screenshot I attached to my previous post?

kizniche, 2020-08-13:
You're only looking in your home directory. Start your search at the root of the filesyate (/) and work your way deeper in the directories that are taking up the largest amount of space.

DarthPleurotus, 2020-08-13:
Ok! Thanks for walking me through this

DarthPleurotus, 2020-08-13:
Ah ok, it looks like by far the majority of space is being taken up by /var/log/mycodo/mycodo.log.1

DarthPleurotus, 2020-08-13:
Do you know what this is?

kizniche, 2020-08-13:
That's a backup of the Mycodo Daemon Log that's created when the main log (*.log) exceeds 10 MB. What size is it? What are the persistent errors in in that are likely making it so large? This is why it's a good to both set up a free disk space Input and have email notification enabled for when the free space drops below a certain value, and to periodically check the Daemon log for issues. Any software that you have a constant stream of errors will eventually cause space issues unless there's a mechanism to delete logs after a certain size. logrotate is used to rotate Mycodo logs, but I should probably include a maximum size option as well.

DarthPleurotus, 2020-08-13:
It is 23.4 GiB. How do I view its contents?

Sorry for the delayed response, my wifi is having some issues.

kizniche, 2020-08-13:
tail -n 100 /var/log/mycodo/mycodo.log

DarthPleurotus, 2020-08-13:
`for each_set in string_sets.split(';'):
                ts_value = each_set.split(',')
                list_ts_values.append({'time': ts_value[0], 'value': float(ts_value[1])})
            return list_ts_values
        return None

    def stop_conditional(self):
        self.running = False

    def conditional_code_run(self):
        # Example code for learning how to use a Conditional. See the manual for more information.

        self.logger.info("This INFO log entry will appear in the Daemon Log")
        self.logger.error("This ERROR log entry will appear in the Daemon Log")

        # Replace "asdf1234" with a Condition ID
        measure1 = self.condition("c5becf2b-3229-4099-a5d8-2e141c534047")
        measure2 = self.condition("c2fe2ae2-d24d-4c7b-8d04-187c6c47f173")
        measure3 = self.condition("d4e5b6f0-9fe2-40a1-8ba9-7001b14e3bc7")
        measure4 = self.condition("3c7f3f52-0e65-4f6a-8774-18189fa2e882")

        if measure1 == measure2 == measure3 == measure4 == "off":
            self.run_all_actions(message=self.message)  # Run all actions sequentially

2020-08-13 18:34:19,861 - INFO - mycodo.controllers.controller_conditional_4a392d2e - This INFO log entry will appear in the Daemon Log
2020-08-13 18:34:19,862 - ERROR - mycodo.controllers.controller_conditional_4a392d2e - This ERROR log entry will appear in the Daemon Log
2020-08-13 18:34:19,973 - DEBUG - mycodo.controllers.controller_conditional_033596fa - Conditional Statement (pre-replacement):
# Example code for learning how to use a Conditional. See the manual for more information.

self.logger.info("This INFO log entry will appear in the Daemon Log")
self.logger.error("This ERROR log entry will appear in the Daemon Log")

# Replace "asdf1234" with a Condition ID
measure1 = self.condition("{597b827d}")
measure2 = self.condition("{52f9808f}")
measure3 = self.condition("{a12254c6}")
measure4 = self.condition("{be8afef5}")

if measure1 == measure2 == measure3 == measure4 == "off":
    self.run_all_actions(message=self.message)  # Run all actions sequentially
2020-08-13 18:34:19,976 - DEBUG - mycodo.controllers.controller_conditional_033596fa - Conditional Statement (post-replacement):
import os
import sys
sys.path.append(os.path.abspath('/var/mycodo-root'))
from mycodo.mycodo_client import DaemonControl
control = DaemonControl()

class ConditionalRun:
    def __init__(self, logger, function_id, message):
        self.logger = logger
        self.function_id = function_id
        self.variables = {}
        self.message = message
        self.running = True

    def run_all_actions(self, message=None):
        if message is None:
            message = self.message
        control.trigger_all_actions(self.function_id, message=message)

    def run_action(self, action_id, message=None):
        if message is None:
            message = self.message
        control.trigger_action(action_id, message=message, single_action=True)

    @staticmethod
    def condition(condition_id):
        return control.get_condition_measurement(condition_id)

    @staticmethod
    def condition_dict(condition_id):
        string_sets = control.get_condition_measurement_dict(condition_id)
        if string_sets:
            list_ts_values = []
            for each_set in string_sets.split(';'):
                ts_value = each_set.split(',')
                list_ts_values.append({'time': ts_value[0], 'value': float(ts_value[1])})
            return list_ts_values
        return None

    def stop_conditional(self):
        self.running = False

    def conditional_code_run(self):
        # Example code for learning how to use a Conditional. See the manual for more information.

        self.logger.info("This INFO log entry will appear in the Daemon Log")
        self.logger.error("This ERROR log entry will appear in the Daemon Log")

        # Replace "asdf1234" with a Condition ID
        measure1 = self.condition("597b827d-f0e9-4b9c-94c8-d20ff1fd92db")
        measure2 = self.condition("52f9808f-bf1a-401e-b168-d1e8858086e7")
        measure3 = self.condition("a12254c6-71c3-4510-9214-0cb4598fdce2")
        measure4 = self.condition("be8afef5-281a-4063-940b-37647be40166")

        if measure1 == measure2 == measure3 == measure4 == "off":
            self.run_all_actions(message=self.message)  # Run all actions sequentially

2020-08-13 18:34:19,978 - INFO - mycodo.controllers.controller_conditional_033596fa - This INFO log entry will appear in the Daemon Log
2020-08-13 18:34:19,980 - ERROR - mycodo.controllers.controller_conditional_033596fa - This ERROR log entry will appear in the Daemon Log`

DarthPleurotus, 2020-08-13:
As far as I can tell this might be because of unconfigured outputs I have in my list. I will get rid of those. In the meantime, is it ok for me to delete some of the old log files?

kizniche, 2020-08-13:
> this might be because of unconfigured outputs

There's nothing in the log to indicate anything about an Output. There are 5 log lines, and they all originate from two different conditional controllers.

If you read each line of the log, you can see what the issue is.

1. You have your conditional controller in debug mode, which adds an extremely large amount of log lines. Two, you have unnecessary log lines in your conditional: "This INFO log entry will appear in the Daemon Log" and "This ERROR log entry will appear in the Daemon Log".

There doesn't seem to be any issue with Mycodo, you just merely instructed it to generate a large amount of log lines for a long period of time.

DarthPleurotus, 2020-08-13:
Ah ok I understand. I will take the controller out of debug mode and delete the unnecessary lines. I'll also go ahead and delete the old log files.

DarthPleurotus, 2020-08-13:
That seems to have done the trick. Everything is working well now.

DarthPleurotus, 2020-08-13:
As always, I really appreciate your quick and helpful responses!

closed 2020-08-13 by DarthPleurotus

kizniche, 2020-08-14:
After checking the Mycodo logrotate config, it should have started deleting logs after 50 MB. I've never tested whether it works beyond merely rotating them, but I'll see if I can reproduce a runaway conditition that logrotate doesn't rectify.