kizniche/Mycodo #815: Graph output duration period (Not feature request, informational)

BirdBare, 2020-08-14:
Hi,

I'm trying to monitor my PID heater with greater resolution. In the PID overview you show a graph with a 20sec period between output duration measurements. I can't figure out how to change the period. Can you point me in the right direction?

[image]

BirdBare, 2020-08-14:
As additional information. I have a 1440 x axis duration and I am monitoring an On/Off python command. Right now the period is every 5 minutes.

kizniche, 2020-08-14:
The Period option determines the interval for which PID and Conditional Controllers operate.

BirdBare, 2020-08-14:
Thanks for the reply. I think I am still missing something. If I increase the temp the number shoots past the PID period as you can see at the end of the attached graph. Additionally, I really like the bar look you had in your graph. Are you using a custom widget or is that possible with the builtin widgets?

[image]

BirdBare, 2020-08-14:
In the case where it shoots up I get a PID output duration of 1000 seconds.

BirdBare, 2020-08-14:
To clarify, I know how to change the period in the PID. But not the period in the graph as you have done. If I use the PID output duration I get crazy high numbers. If I use the raw output duration the period is always set to 60 seconds.

BirdBare, 2020-08-14:
Here is a graph with the raw output duration. I get the output time as bars which is what I want. But the period is every 60 seconds. Is there an easy way to change the period to 20 seconds like you have above?

[image]

kizniche, 2020-08-14:
Yes, you change the PID Period to 20.

kizniche, 2020-08-14:
If what you mean is to turn off data grouping, you must check "Disable Data Grouping" for the particular graph Output series.

BirdBare, 2020-08-14:
Oh my gosh that's what it was. Thanks so much! I don't know how I missed that in the manual. Thank you for your time.

closed 2020-08-14 by BirdBare

kizniche, 2020-08-14:
Your PID output is the result of your PID tuning (gains and integrator). If the output is higher than you would like, this indicates your Kp and/or Ki and/or integrator may be too high.