kizniche/Mycodo #816: after upgrade dashboard not working 8.7.1

Mark0Mi, 2020-08-19:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1278, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 593, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: widget.log_level_debug

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 563, in page_dashboard
    dashboard = Widget.query.all()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3341, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3503, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3528, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1014, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 298, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1133, in _execute_clauseelement
    distilled_params,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1318, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1512, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 178, in raise_
    raise exception
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1278, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 593, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: widget.log_level_debug
[SQL: SELECT widget.id AS widget_id, widget.unique_id AS widget_unique_id, widget.graph_type AS widget_graph_type, widget.dashboard_id AS widget_dashboard_id, widget.name AS widget_name, widget.log_level_debug AS widget_log_level_debug, widget.font_em_name AS widget_font_em_name, widget.enable_drag_handle AS widget_enable_drag_handle, widget.position_x AS widget_position_x, widget.position_y AS widget_position_y, widget.width AS widget_width, widget.height AS widget_height, widget.custom_options AS widget_custom_options, widget.enable_header_buttons AS widget_enable_header_buttons, widget.period AS widget_period, widget.refresh_duration AS widget_refresh_duration, widget.x_axis_duration AS widget_x_axis_duration, widget.custom_yaxes AS widget_custom_yaxes, widget.decimal_places AS widget_decimal_places, widget.enable_status AS widget_enable_status, widget.enable_value AS widget_enable_value, widget.enable_name AS widget_enable_name, widget.enable_unit AS widget_enable_unit, widget.enable_measurement AS widget_enable_measurement, widget.enable_channel AS widget_enable_channel, widget.enable_timestamp AS widget_enable_timestamp, widget.pid_ids AS widget_pid_ids, widget.output_ids AS widget_output_ids, widget.math_ids AS widget_math_ids, widget.note_tag_ids AS widget_note_tag_ids, widget.input_ids_measurements AS widget_input_ids_measurements, widget.enable_navbar AS widget_enable_navbar, widget.enable_rangeselect AS widget_enable_rangeselect, widget.enable_export AS widget_enable_export, widget.enable_title AS widget_enable_title, widget.enable_auto_refresh AS widget_enable_auto_refresh, widget.enable_xaxis_reset AS widget_enable_xaxis_reset, widget.enable_manual_y_axis AS widget_enable_manual_y_axis, widget.enable_start_on_tick AS widget_enable_start_on_tick, widget.enable_end_on_tick AS widget_enable_end_on_tick, widget.enable_align_ticks AS widget_enable_align_ticks, widget.use_custom_colors AS widget_use_custom_colors, widget.custom_colors AS widget_custom_colors, widget.disable_data_grouping AS widget_disable_data_grouping, widget.max_measure_age AS widget_max_measure_age, widget.stops AS widget_stops, widget.range_colors AS widget_range_colors, widget.y_axis_min AS widget_y_axis_min, widget.y_axis_max AS widget_y_axis_max, widget.option_invert AS widget_option_invert, widget.font_em_value AS widget_font_em_value, widget.font_em_timestamp AS widget_font_em_timestamp, widget.enable_output_controls AS widget_enable_output_controls, widget.show_pid_info AS widget_show_pid_info, widget.show_set_setpoint AS widget_show_set_setpoint, widget.camera_id AS widget_camera_id, widget.camera_image_type AS widget_camera_image_type, widget.camera_max_age AS widget_camera_max_age 
FROM widget]
(Background on this error at: http://sqlalche.me/e/13/e3q8)

kizniche, 2020-08-19:
Looks like your database didn't upgrade. Try executing in a terminal:

```bash
cd ~/Mycodo/databases
../env/bin/alembic upgrade head
```

Mark0Mi, 2020-08-19:
I don't get any results when executing: 
pi@raspberrypi:~/Mycodo/databases $ ../env/bin/alembic upgrade head
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.

kizniche, 2020-08-19:
Then there seems to have been an error during the database upgrade, not that it didn't upgrade. Have you ever upgraded the the master branch in the history of this Mycodo install?

Mark0Mi, 2020-08-20:
as I have no idea how to do this - the answer is no. 
I started approx 1 month ago by installing 8.5.8 - then a couple of minor upgrades due to bugs and feature enhancements (LCD,..) 
I did not upgrade immediately to 8.7.x as I had no time - so I guess I upgraded from 8.6.x to 8.7.1 -> the upgrarde itself seemed to have run through...

Mark0Mi, 2020-08-20:
upgrade log:
2020-08-19 20:31:38] Installed version: 8.6.4
[2020-08-19 20:31:38] Latest version: 8.7.1
 ERROR: Failed building wheel for python-lirc is the only thing that sticks out. is there anything else you might need? 

2020-08-19 20:32:35] #### Compiling mycodo_wrapper
[2020-08-19 20:32:36] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function âupgrade_commandsâ:
[2020-08-19 20:32:36] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:17:5: warning: âstrncatâ specified bound 255 equals destination size [-Wstringop-overflow=]
[2020-08-19 20:32:36]      strncat(full_cmd, "/upgrade_commands.sh ", sizeof(full_cmd));
[2020-08-19 20:32:36]      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[2020-08-19 20:32:36] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function âmainâ:
[2020-08-19 20:32:36] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:35:4: warning: âstrncatâ specified bound 255 equals destination size [-Wstringop-overflow=]
[2020-08-19 20:32:36]     strncat(restoreScript, "/upgrade_commands.sh backup-create", sizeof(restoreScript));
[2020-08-19 20:32:36]     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[2020-08-19 20:34:54] dpkg-query: no packages found matching liblircclient-dev
[2020-08-19 20:34:59] debconf: unable to initialize frontend: Dialog
[2020-08-19 20:34:59] debconf: (TERM is not set, so the dialog frontend is not usable.)
[2020-08-19 20:34:59] debconf: falling back to frontend: Readline
[2020-08-19 20:34:59] debconf: unable to initialize frontend: Readline
[2020-08-19 20:34:59] debconf: (This frontend requires a controlling tty.)
[2020-08-19 20:34:59] debconf: falling back to frontend: Teletype
[2020-08-19 20:35:00] dpkg-preconfigure: unable to re-open stdin: 
[2020-08-19 20:35:35]   ERROR: Command errored out with exit status 1:
[2020-08-19 20:35:35]    command: /home/pi/Mycodo/env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-u7dz06fo/python-lirc/setup.py'"'"'; __file__='"'"'/tmp/pip-install-u7dz06fo/python-lirc/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-6conq8d1
[2020-08-19 20:35:35]        cwd: /tmp/pip-install-u7dz06fo/python-lirc/
[2020-08-19 20:35:35]   Complete output (13 lines):
[2020-08-19 20:35:35]   running bdist_wheel
[2020-08-19 20:35:35]   running build
[2020-08-19 20:35:35]   running build_ext
[2020-08-19 20:35:35]   building 'lirc' extension
[2020-08-19 20:35:35]   creating build
[2020-08-19 20:35:35]   creating build/temp.linux-armv7l-3.7
[2020-08-19 20:35:35]   creating build/temp.linux-armv7l-3.7/lirc
[2020-08-19 20:35:35]   arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/lirc/ -I/home/pi/Mycodo/env/include -I/usr/include/python3.7m -c lirc/lirc.c -o build/temp.linux-armv7l-3.7/lirc/lirc.o
[2020-08-19 20:35:35]   lirc/lirc.c:276:10: fatal error: lirc/lirc_client.h: No such file or directory
[2020-08-19 20:35:35]    #include "lirc/lirc_client.h"
[2020-08-19 20:35:35]             ^~~~~~~~~~~~~~~~~~~~
[2020-08-19 20:35:35]   compilation terminated.
[2020-08-19 20:35:35]   error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
[2020-08-19 20:35:35]   ----------------------------------------
[2020-08-19 20:35:35]   ERROR: Failed building wheel for python-lirc
[2020-08-19 20:35:37] DEPRECATION: Could not build wheels for python-lirc which do not use PEP 517. pip will fall back to legacy 'setup.py install' for these. pip 21.0 will remove support for this functionality. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
[2020-08-19 20:35:39]     ERROR: Command errored out with exit status 1:
[2020-08-19 20:35:39]      command: /home/pi/Mycodo/env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-u7dz06fo/python-lirc/setup.py'"'"'; __file__='"'"'/tmp/pip-install-u7dz06fo/python-lirc/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-uvukynvx/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python3.7/python-lirc
[2020-08-19 20:35:39]          cwd: /tmp/pip-install-u7dz06fo/python-lirc/
[2020-08-19 20:35:39]     Complete output (13 lines):
[2020-08-19 20:35:39]     running install
[2020-08-19 20:35:39]     running build
[2020-08-19 20:35:39]     running build_ext
[2020-08-19 20:35:39]     building 'lirc' extension
[2020-08-19 20:35:39]     creating build
[2020-08-19 20:35:39]     creating build/temp.linux-armv7l-3.7
[2020-08-19 20:35:39]     creating build/temp.linux-armv7l-3.7/lirc
[2020-08-19 20:35:39]     arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/lirc/ -I/home/pi/Mycodo/env/include -I/usr/include/python3.7m -c lirc/lirc.c -o build/temp.linux-armv7l-3.7/lirc/lirc.o
[... cut ...] '"'"'; __file__='"'"'/tmp/pip-install-u7dz06fo/python-lirc/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-uvukynvx/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python3.7/python-lirc Check the logs for full command output.
[2020-08-19 20:35:40] Reading package lists...

kizniche, 2020-08-20:
Nothing about lirc is the issue. It's your database. Try downgrading one version, then uograding:

```bash
cd ~/Mycodo/databases
../env/bin/alembic downgrade -1
../env/bin/alembic upgrade head
```

kizniche, 2020-08-20:
Doing so may require you delete all dashboards and widgets from the configure -> diagnostic page, if your dashboards don't work correctly after the downgrade/upgrade.

kizniche, 2020-08-20:
It would be better for you to attach (as txt file, don't paste the whole thing) the entire upgrade log for this version (not the entire log, just this version's excerpt), rather than trying to find the issue yourself.

Mark0Mi, 2020-08-20:
pi@raspberrypi:/sys/bus/w1/devices $ cd ~/Mycodo/databases
pi@raspberrypi:~/Mycodo/databases $ ../env/bin/alembic downgrade -1
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
INFO  [alembic.runtime.migration] Running downgrade d66e33093e8e -> 4d3258ef5864, add Python Code widget options
Traceback (most recent call last):
  File "../env/bin/alembic", line 8, in <module>
    sys.exit(main())
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 577, in main
    CommandLine(prog=prog).main(argv=argv)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 571, in main
    self.run_cmd(cfg, options)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 551, in run_cmd
    **dict((k, getattr(options, k, None)) for k in kwarg)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/command.py", line 339, in downgrade
    script.run_env()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/script/base.py", line 489, in run_env
    util.load_python_file(self.dir, "env.py")
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/util/pyfiles.py", line 98, in load_python_file
    module = load_module_py(module_id, path)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/util/compat.py", line 184, in load_module_py
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "alembic/env.py", line 69, in <module>
    run_migrations_online()
  File "alembic/env.py", line 64, in run_migrations_online
    context.run_migrations()
  File "<string>", line 8, in run_migrations
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/runtime/environment.py", line 846, in run_migrations
    self.get_context().run_migrations(**kw)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/runtime/migration.py", line 520, in run_migrations
    step.migration_fn(**kw)
  File "/home/pi/Mycodo/databases/alembic/versions/d66e33093e8e_add_python_code_widget_options.py", line 67, in downgrade
    batch_op.drop_column('log_level_debug')
  File "/usr/lib/python3.7/contextlib.py", line 119, in __exit__
    next(self.gen)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/operations/base.py", line 354, in batch_alter_table
    impl.flush()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/operations/batch.py", line 114, in flush
    fn(*arg, **kw)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/operations/batch.py", line 491, in drop_column
    del self.columns[column.name]
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/_collections.py", line 344, in __delitem__
    dict.__delitem__(self, key)
KeyError: 'period'

does not seem to work.. 
$ ../env/bin/alembic upgrade head
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.

@diagnostic: I can't delete them when I run diag -> delete all dashboards I get
Error: Delete Dashboard: (sqlite3.OperationalError) no such column: widget.log_level_debug
[SQL: SELECT widget.id AS widget_id, widget.unique_id AS widget_unique_id, widget.graph_type AS widget_graph_type, widget.dashboard_id AS widget_dashboard_id, widget.name AS widget_name, widget.log_level_debug AS widget_log_level_debug, widget.font_em_name AS widget_font_em_name, widget.enable_drag_handle AS widget_enable_drag_handle, widget.position_x AS widget_position_x, widget.position_y AS widget_position_y, widget.width AS widget_width, widget.height AS widget_height, widget.custom_options AS widget_custom_options, widget.enable_header_buttons AS widget_enable_header_buttons, widget.period AS widget_period, widget.refresh_duration AS widget_refresh_duration, widget.x_axis_duration AS widget_x_axis_duration, widget.custom_yaxes AS widget_custom_yaxes, widget.decimal_places AS widget_decimal_places, widget.enable_status AS widget_enable_status, widget.enable_value AS widget_enable_value, widget.enable_name AS widget_enable_name, widget.enable_unit AS widget_enable_unit, widget.enable_measurement AS widget_enable_measurement, widget.enable_channel AS widget_enable_channel, widget.enable_timestamp AS widget_enable_timestamp, widget.pid_ids AS widget_pid_ids, widget.output_ids AS widget_output_ids, widget.math_ids AS widget_math_ids, widget.note_tag_ids AS widget_note_tag_ids, widget.input_ids_measurements AS widget_input_ids_measurements, widget.enable_navbar AS widget_enable_navbar, widget.enable_rangeselect AS widget_enable_rangeselect, widget.enable_export AS widget_enable_export, widget.enable_title AS widget_enable_title, widget.enable_auto_refresh AS widget_enable_auto_refresh, widget.enable_xaxis_reset AS widget_enable_xaxis_reset, widget.enable_manual_y_axis AS widget_enable_manual_y_axis, widget.enable_start_on_tick AS widget_enable_start_on_tick, widget.enable_end_on_tick AS widget_enable_end_on_tick, widget.enable_align_ticks AS widget_enable_align_ticks, widget.use_custom_colors AS widget_use_custom_colors, widget.custom_colors AS widget_custom_colors, widget.disable_data_grouping AS widget_disable_data_grouping, widget.max_measure_age AS widget_max_measure_age, widget.stops AS widget_stops, widget.range_colors AS widget_range_colors, widget.y_ax [... cut ...] S widget_option_invert, widget.font_em_value AS widget_font_em_value, widget.font_em_timestamp AS widget_font_em_timestamp, widget.enable_output_controls AS widget_enable_output_controls, widget.show_pid_info AS widget_show_pid_info, widget.show_set_setpoint AS widget_show_set_setpoint, widget.camera_id AS widget_camera_id, widget.camera_image_type AS widget_camera_image_type, widget.camera_max_age AS widget_camera_max_age 
FROM widget]
(Background on this error at: http://sqlalche.me/e/13/e3q8)

Mark0Mi, 2020-08-20:
[upgradelog.txt](https://github.com/kizniche/Mycodo/files/5104077/upgradelog.txt)

kizniche, 2020-08-20:
```
[2020-08-19 20:34:47] #### Upgrading Mycodo database with alembic (if needed) [2020-08-19 20:34:48] INFO [alembic.runtime.migration] Context impl SQLiteImpl. [2020-08-19 20:34:48] INFO [alembic.runtime.migration] Will assume non-transactional DDL.
```

This indicates the database wasn't upgraded, for whatever reason. I would suggest [restoring](https://kizniche.github.io/Mycodo/Upgrade-Backup-Restore/#backup-restore) a previous version, then attempting an upgrade again.

ticomyco, 2020-08-20:
I also am having dashboard problems after upgrading from 8.6.1 -> 8.7.1:
Clicking on the dashboard link pulls up the page with all of the frames for the various graphs and their titles, however there are no graphs displayed. PID controls appear and work okay, but gauges and graphs do not. I tried creating a new graph and selecting a few random sensors to see if it was just old graphs that were affected, however that resulted in just another blank frame.
I don't know if my issue is the exact same as Mark0Mi's issue, so I haven't tried reverting to a previous version or upgrading the database. 
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/5104971/mycodoupgrade.log)

kizniche, 2020-08-20:
@ticomyco Your issue does not appear to be a database issue, as your dashboard is actually loading. Perhaps what you should be posting in is #817.

ticomyco, 2020-08-20:
@kizniche -- Gotcha, will do.