kizniche/Mycodo #817: Dashboard problem

LilianV, 2020-08-20:
Hi,

After having upgraded from 8.6.4 to 8.7.1, everything looked fine except some dashboards a bit messed up. When I tried to repair them (moving and resizing the widgets that needed), it was impossible.

I tried to recreate after having deleted ("Diagnostics > Delete All Dashboards"). For the first 6/7 widgets, no problem but, at a point I can create new widgets but not drag or resize.

Example in the following picture, the last widget is created with a full width size, impossible to resize or move.

[image]

It seems that the widgets begin to misbehave when I add a measurement concerning a math sum for a particular output (Peristaltic Pump from Atlas Scientific). I tried to recreate the output, the math sum, nothing worked. I have another pump with the same math sum and widget, that works fine.

If you have an idea...

Regards,
Lilian

kizniche, 2020-08-20:
Can you provide detailed steps to reproduce the issue?

LilianV, 2020-08-20:
Hi !

1) I create a new dashboard
2) I add widgets
3) After a number of widgets added correctly, there is one that is created with a (correct) value but not resizable and impossible to move. 
4) Every widget after this one is created with a (correct) value but not resizable and impossible to move. 

I can give an access to my system if you want to see (I have a NAT/PAT with a public address).

Regards,
Lilian

kizniche, 2020-08-20:
These instructions aren't detailed enough. Can you be more specific about which widgets to add and with which configurations that are pertinent to inducing the issue?

LilianV, 2020-08-20:
I understand and I'm really sorry but I did not find a logic...

I can add an number of widgets, and maybe the 6th or 12th will make a mess and after all the new added widgets will behave asame.

I did not find a particular widget type or configuration, neither a number of widgets...

My default dashboard messed up at widget 12 (Measurement - Math Sum Past/Single channel - based on volume in ml dispensed by an Atlas scientific peristaltic pump in the last 24 hours):

[image]

LilianV, 2020-08-20:
Maybe I need to make a full clean install.

ticomyco, 2020-08-20:
Re-posting here --
I also am having dashboard problems after upgrading from 8.6.1 -> 8.7.1:
Clicking on the dashboard link pulls up the page with all of the frames for the various graphs and their titles, however there are no graphs displayed. PID controls appear and work okay, but gauges and graphs do not. I tried creating a new graph and selecting a few random sensors to see if it was just old graphs that were affected, however that resulted in just another blank frame.

I restored my last backup to 8.6.1 and everything went back to working normally. 

Then I tried running the upgrade to 8.7.1 and once again I get the same broken dashboard, as shown in the screenshot. 
[image]
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/5105611/mycodoupgrade.log)

Looking at the logfiles, it appears that widgets.py is having some Unicode encoding errors, but I'm not familiar with the errors, and for the record. I don't know if it matters at all, but none of the sensors or outputs names contain any characters that aren't in latin-1 or ascii.

kizniche, 2020-08-20:
Can you save the source of the dashboard to a txt file and attach it?

kizniche, 2020-08-20:
FYI, there are over 50 systems on 8.7.1, and so far you two are the only ones experiencing this type of display issue.

ticomyco, 2020-08-20:
Here's the source. Good to hear that 8.7.1 is working well for others! 
[raspberrymush - Default Dashboard - Mycodo 8.7.1.html.txt](https://github.com/kizniche/Mycodo/files/5105661/raspberrymush.-.Default.Dashboard.-.Mycodo.8.7.1.html.txt)

LilianV, 2020-08-20:
Just a notice: my system langage is not english by default and you ticomyco ? I notice that some dashboards became blank when I switch Mycodo in french

LilianV, 2020-08-20:
> If you can send instructions for how to access your web interface to kylegabriel.com/contact I'll take a look.

Done.

ticomyco, 2020-08-20:
@kizniche I've also forwarded my system access credentials to your contact page in case that's helpful.
@LilianV  -- interesting. My system is set for english, however when I try switching to Spanish or Russian I also still have the same result as when it is in English.

kizniche, 2020-08-20:
@LilianV I found there's an extra </div> closing tag that's causing the issue. I'm looking what's causing that.

@ticomyco You are likely experiencing a different issue, since you have different symptoms. I'll look at your system if/when I finish with Lilian's, if I have time.

ticomyco, 2020-08-20:
Thanks! -- And no worries, I can always revert to a backup again if it you run out of time or just want to call it a day :)

kizniche, 2020-08-20:
Good news, I was able to reproduce the issue on my system. It occurs when you create two Measurement widgets using a Math controllers. Still trying to find the exact cause.

kizniche, 2020-08-20:
I just pushed a fix, which you should see the reference above this comment. If you want to test the fix with your system, you'll need to manually make the change, then run the following commands:

```bash
sudo sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh generate-widget-html
sudo service mycodoflask restart
```

ticomyco, 2020-08-20:
That's great! 
However, it seems that something is still not quite right on my system:

pi@raspberrymush:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh  generate-widget-html

#### Generating widget HTML files
2020-08-20 18:02:37,939 Generating widget HTML for widget: widget_gauge_solid
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/utils/widgets.py", line 116, in generate_widget_html
    fw.write(html_js_ready_end)
UnicodeEncodeError: 'latin-1' codec can't encode character '\u25cf' in position 5099: ordinal not in range(256)
2020-08-20 18:02:37,951 Generating widget HTML for widget: widget_gauge_angular
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/utils/widgets.py", line 116, in generate_widget_html
    fw.write(html_js_ready_end)
UnicodeEncodeError: 'latin-1' codec can't encode character '\u25cf' in position 4867: ordinal not in range(256)
2020-08-20 18:02:37,962 Generating widget HTML for widget: widget_graph_synchronous
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/utils/widgets.py", line 105, in generate_widget_html
    fw.write(html_js_ready)
UnicodeEncodeError: 'latin-1' codec can't encode characters in position 1942-1943: ordinal not in range(256)

kizniche, 2020-08-20:
Okay, you two are having very similar issues because it seems @ticomyco's system wasn't proerly generating the HTML either. I was a moment away from attempting to generate it on his system. I didn't have SSH access to your system, so I couldn't test.

kizniche, 2020-08-20:
Just pushed a fix for the issue. If you make the edits manually, you should be able to then successfully run the HTML generation command.

kizniche, 2020-08-20:
Whoops, missed one. There's another commit with the remaining fix.

kizniche, 2020-08-20:
Actually, that didn't work as expected: 
[image]

kizniche, 2020-08-20:
Hah, sorry for all the commits, but I decided to use just use the ```o``` character, but then realized it wasn't even necessary for the gauges, so took it out completely. That should be the last fix to get it all working.

ticomyco, 2020-08-21:
Fantastic! I just took a peak at it and it looks great! Thanks a bunch.

kizniche, 2020-08-21:
Alright, that was really the last commit for this issue. I found setting ```useHTML: true,``` let me use the character I originally wanted to use.

LilianV, 2020-08-21:
Hi guys !

I applied the commits early in the morning (in France) when I woke up and widgets are working fine now !

.... and it looks like my tomatoes are growing faster now ! :)

I understand that it's ok for ticomyco too, so I close the issue.

Thanks for night support ! 👍

closed 2020-08-21 by LilianV