kizniche/Mycodo #821: dashboard graph report

Mark0Mi, 2020-08-23:
[image]
a pic shows more than words - the specific time point when the pump is switched on is logged 
the pump was on for 16 000 seconds -> 4.4 hours 
the graph shows one line at a time -> can I change this so that the graph shows a "high" as long as the pump is running? 

### Versions:

 - Mycodo Version: 8.7.1
 - Raspberry Pi Version: [e.g. 4B+]

### Reproducibility

kizniche, 2020-08-23:
You can set minimum and maximum for specific y-axes in the graph options.

Mark0Mi, 2020-08-23:
Hi 
I have seen this but I want to scale on x-axes (time)
I have drawn into the screenshot in yellow -> this is what I'd like to see (the seconds are counted -> I want this to be on X axes not y. is this possible at all? 
[image]
thanks!

kizniche, 2020-08-23:
Well, the "X-axis" option let's you set the x axis. You can also enable the range selector.

kizniche, 2020-08-23:
Oh, I see what you're asking. No, it is not possible.

kizniche, 2020-08-23:
Though, you can now create and import your own widgets, so it's possible for you to copy and modify the graph widget to do what you want.

closed 2020-08-23 by kizniche