kizniche/Mycodo #822: fresh install - some errors in log - #### Generating widget HTML files

Mark0Mi, 2020-08-23:
### Versions:

 - Mycodo Version: [e.g. 8.7.1]
 - Raspberry Pi Version: [e.g. 4B+]
- 3rd party OS
### Reproducibility
fresh install on a new OS (not raspian but I guess it does not matter) 

the database is initialized later on (if I remember correctly) but the setup tries to access table misc upfront see attached log 
[installlog.txt](https://github.com/kizniche/Mycodo/files/5114548/installlog.txt)

again everything seems to be working so I think it is a "cosmetic" issue... 
BR and keep up the work I really like it ;-)

kizniche, 2020-08-23:
Good catch. Thanks. I'll look into fixing it.

kizniche, 2020-08-23:
Just released 8.7.2 that should fix the issue.

closed 2020-08-23 by kizniche