kizniche/Mycodo #824: Issue not loading the front end

cajun1689, 2020-08-25:
I am having an issue with loading the front end. I am not sure what changed one moment was working and then it stopped.

`Aug 25 00:19:07 raspberrypi gunicorn[482]: [2020-08-25 00:19:07 -0400] [482] [INFO] Listening at: unix:/var/run/mycodoflask.sock (482)
Aug 25 00:19:07 raspberrypi gunicorn[482]: [2020-08-25 00:19:07 -0400] [482] [INFO] Using worker: gthread
Aug 25 00:19:07 raspberrypi gunicorn[482]: [2020-08-25 00:19:07 -0400] [597] [INFO] Booting worker with pid: 597
Aug 25 00:19:11 raspberrypi gunicorn[482]: [2020-08-25 00:19:11 -0400] [597] [ERROR] Exception in worker process
Aug 25 00:19:11 raspberrypi gunicorn[482]: Traceback (most recent call last):
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
Aug 25 00:19:11 raspberrypi gunicorn[482]:     worker.init_process()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
Aug 25 00:19:11 raspberrypi gunicorn[482]:     super().init_process()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
Aug 25 00:19:11 raspberrypi gunicorn[482]:     self.load_wsgi()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
Aug 25 00:19:11 raspberrypi gunicorn[482]:     self.wsgi = self.app.wsgi()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
Aug 25 00:19:11 raspberrypi gunicorn[482]:     self.callable = self.load()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
Aug 25 00:19:11 raspberrypi gunicorn[482]:     return self.load_wsgiapp()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
Aug 25 00:19:11 raspberrypi gunicorn[482]:     return util.import_app(self.app_uri)
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
Aug 25 00:19:11 raspberrypi gunicorn[482]:     mod = importlib.import_module(module)
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Aug 25 00:19:11 raspberrypi gunicorn[482]:     return _bootstrap._gcd_import(name[level:], package, level)
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
Aug 25 00:19:11 raspberrypi gunicorn[482]:     from mycodo.mycodo_flask.app import create_app
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 32, in <module>
Aug 25 00:19:11 raspberrypi gunicorn[482]:     from mycodo.mycodo_flask import routes_page
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 89, in <module>
Aug 25 00:19:11 raspberrypi gunicorn[482]:     from mycodo.mycodo_flask.forms import forms_input
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_input.py", line 34, in <module>
Aug 25 00:19:11 raspberrypi gunicorn[482]:     class InputAdd(FlaskForm):
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_input.py", line 36, in InputAdd
Aug 25 00:19:11 raspberrypi gunicorn[482]:     dict_inputs = parse_input_information()
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 108, in parse_input_information
Aug 25 00:19:11 raspberrypi gunicorn[482]:     input_custom = load_module_from_file(full_path, 'inputs')
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "/home/pi/Mycodo/mycodo/utils/modules.py", line 17, in load_module_from_file
Aug 25 00:19:11 raspberrypi gunicorn[482]:     module_custom = importlib.util.module_from_spec(spec)
Aug 25 00:19:11 raspberrypi gunicorn[482]:   File "<frozen importlib._bootstrap>", line 580, in module_from_spec
Aug 25 00:19:11 raspberrypi gunicorn[482]: AttributeError: 'NoneType' object has no attribute 'loader'
Aug 25 00:19:11 raspberrypi gunicorn[482]: [2020-08-25 00:19:11 -0400] [597] [INFO] Worker exiting (pid: 597)
Aug 25 00:19:11 raspberrypi gunicorn[482]: [2020-08-25 00:19:11 -0400] [482] [INFO] Shutting down: Master
Aug 25 00:19:11 raspberrypi gunicorn[482]: [2020-08-25 00:19:11 -0400] [482] [INFO] Reason: Worker failed to boot.
Aug 25 00:19:11 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Aug 25 00:19:11 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.`

kizniche, 2020-08-25:
What version? Are you using any custom modules you imported?

I see where I need to modify the code to prevent this type of error from crashing the frontend, but that doesn't fix this particular issue.

cajun1689, 2020-08-25:
i have no custom modules just a couple conditionals. i updated the system to the latest version this afternoon. the backend is still running because my pumps are turning on and off like they should

cajun1689, 2020-08-25:
i dont know if this is related in any way but the system also freezes when i go to the camera page. i have been working on some python outputs that will send i2c commands to an arduino to turn on and off relays for some additional peristaltic pumps

cajun1689, 2020-08-25:
any thoughts on what to do here? Should i just start with a fresh install and set everything back up?

kizniche, 2020-08-25:
What is the output from ```ls ~/Mycodo/mycodo/inputs```?

cajun1689, 2020-08-25:
[image]
this is what i'm getting

kizniche, 2020-08-26:
Nothing looks out of the ordinary. If this happened out of nowhere, it suggests you may have a filesystem issue. I would reflash Raspberry pi OS and install again on a different SD card.

closed 2020-08-26 by cajun1689