kizniche/Mycodo #829: Initial Install - after admin creation - code 500

kkillion43, 2020-08-27:
I am not entirely sure, but appears it was looking for a 'pi' user?

**OS type/version**

- Distributor ID:	Ubuntu
- Description:	Ubuntu 20.04.1 LTS
- Release:	20.04

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/ubuntu/Mycodo/env/lib/python3.8/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/ubuntu/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1933, in page_data
    elif not dpkg_package_exists('ow-shell'):
  File "/home/ubuntu/Mycodo/mycodo/utils/system_pi.py", line 145, in dpkg_package_exists
    _, _, stat = cmd_output(cmd)
  File "/home/ubuntu/Mycodo/mycodo/utils/system_pi.py", line 311, in cmd_output
    pw_record = pwd.getpwnam(user)
KeyError: "getpwnam(): name not found: 'pi'"

kizniche, 2020-08-27:
Mycodo is not designed to install on Ubuntu. If you wish to try it on an unsupported platform, I would recommend using [docker](https://github.com/kizniche/Mycodo/wiki/Mycodo-and-Docker).

kkillion43, 2020-08-27:
Ah, ok no problem. I'll go that route and thanks man.

closed 2020-08-27 by kkillion43