kizniche/Mycodo #831: Error 500: Internal Server Error while adding dashboard widgets

bndaquino, 2020-08-30:
Hi Kyle,

I just got an Error 500: Internal Server Error while fiddling around adding a few widgets to my dashboard, and now I can't access Mycodo at all. I think I was using version 8.1.0. I'm pasting the traceback below. Thanks in advance for any help you might be able to give me.

Error (Full Traceback):

```
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 769, in page_dashboard
    y_axes=y_axes)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/asyncsupport.py", line 76, in render
    return original_render(self, *args, **kwargs)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1005, in render
    return concat(self.root_render_func(self.new_context(vars)))
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 22, in root
    <script type="text/javascript" src="/static/js/modules/solid-gauge.js"></script>
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 298, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 113, in block_body
    </div>
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard_options/mod_measurement.html", line 85, in root
    dict_units[dict_measure_units[measurement_id]]['unit'] -%}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: collections.OrderedDict object has no element None
```

kizniche, 2020-08-30:
I won't be able to help with the issue unless you can list the exact steps to reproduce the issue. You can delete your dashboards and widgets from the Configure -> Diagnostics page.

bndaquino, 2020-08-30:
To be more precise, I was adding a Measurement widget to the dashboard for a BME280 temperature sensor. The error came up after I saved my changes and the page was reloaded. Prior to this, I had been adding measurement widgets for BME280 humidity MH-Z16 CO2 without any issues. I'm not sure what other information might be relevant here, but please let me know if there's anything more specific you need to know.

Would deleting my dashboard/widgets help at all? And if so, is there a way I can do that in the Mycodo file directory on my pi (as I can't access the web UI to delete them from the Configure -> Diagnostics page)?

Thanks again for your help and sorry if these are somewhat basic questions. I'm just not really sure what to do here.

kizniche, 2020-08-30:
You should be able to access the web interface. You just can't access the /dash endpoint

bndaquino, 2020-08-30:
I just cleared my dashboard from the Diagnostics page and everything seems to be working. I'm in the process of adding my widgets back -- I'll let you know if I can recreate the issue. Thanks again.

bndaquino, 2020-08-31:
I upgraded to version 8.7.2, and I've found that I can consistently reproduce the Error 500 by trying to add widgets with an Average (Past, Single Channel) Math module, using a BME280 temperature channel as the data source.

Strangely, the error only occurs when I use the BME280 temperature channel average as the data source -- using the raw, unaveraged BME280 temperature or the BME280 humidity (averaged or unaveraged), works fine. Additionally, the error only occurs when I add a Gauge (Solid) or Measurement widget -- Gauge (Angular) and Indicator are fine. I'm using two different BME280s at the same time on different I2C buses, and both reproduce the error identically.

My intended behavior was for the Math module to track the average of the past hour of data. Here's my configuration for reference:

[image]

kizniche, 2020-08-31:
What happens when you use the default, unconcerned Input measurement (Celsius)?

bndaquino, 2020-08-31:
The behavior is the same.

kizniche, 2020-08-31:
Thanks for all the information. I'll investigate the cause.

kizniche, 2020-09-08:
I could not replicate the issue until I converted the input measurement to a non-default unit. working on a fix now.

closed 2020-09-09 by kizniche