kizniche/Mycodo #832: Error 502: Bad Gateway Error

ML1999, 2020-09-02:
Error 502: Bad Gateway Error after Reinstall
-----------------------------------------------------------------------------------
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
uname -a
Linux pi3b 5.4.61-v7+ #1339 SMP Tue Sep 1 18:45:30 BST 2020 armv7l GNU/Linux
--------------------------------------------------------------------------------------------------------------
Hardware	: BCM2835
Revision	: a22082
Serial		: 00000xxxxxxxxxxx
Model		: Raspberry Pi 3 Model B Rev 1.2
--------------------------------------------------------------------------------------------------------------
 $ /opt/vc/bin/vcgencmd version
Sep  1 2020 19:01:18 
Copyright (c) 2012 Broadcom
version d3115dbb5ea353b3a0e5e6a3b52e95cafc7249ee (clean) (release) (start)
--------------------------------------------------------------------------------------------------------------------
Steps: removed org version

137  rm -rf Mycodo
138  curl -L https://kizniche.github.io/Mycodo/install | bash

Console output and /home/pi/Mycodo/install/setup.log attached.
[console_out_put.txt](https://github.com/kizniche/Mycodo/files/5160204/console_out_put.txt)
[setup.log](https://github.com/kizniche/Mycodo/files/5160200/setup.log)

ML1999, 2020-09-02:
#### Restarting the Mycodo daemon
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/scripts/restart_daemon.py", line 12, in <module>
    from mycodo.config import DAEMON_PID_FILE
  File "/home/pi/Mycodo/mycodo/config.py", line 10, in <module>
    from flask_babel import lazy_gettext
ModuleNotFoundError: No module named 'flask_babel'
Job for mycodo.service failed because the control process exited with error code.
See "systemctl status mycodo.service" and "journalctl -xe" for details.
#### Mycodo Installer finished Tue  1 Sep 22:34:34 MDT 2020
#### Total install time: 12 minutes and 28 seconds

*********************************
** Mycodo finished installing! **
*********************************

mnz454, 2020-09-02:
I am also having this issue upon install.

My install reporting: 
ModuleNotFoundError: No module named 'Pyro5'

kizniche, 2020-09-02:
If your problems stem from not having the proper libraries, this is likely an issue with the package management system pip/pypi or your system/internet connection, and not Mycodo.

kizniche, 2020-09-02:
It appears to be a bug in setuptools (issue [#2353] (https://github.com/pypa/setuptools/issues/2353)

kizniche, 2020-09-02:
Until setuptools releases a new version with a fix, what you can do is issue the following commands to fix your broken install:

```
sudo ~/Mycodo/env/bin/pip install setuptools==49.5.0
sudo bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
sudo shutdown now -r
```

griffira, 2020-09-02:
Hi,

I tried this on the new install I mentioned in #833:

> sudo ~/Mycodo/env/bin/pip install setuptools==49.5.0
> sudo bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
> sudo shutdown now -r

I'm getting the below now. Should I try to install Pyro5 and other modules as they come up? Or, do I just need to wait for a permanent fix??

Thanks Again!

pi@raspberrypi:~ $ sudo service mycodo status
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-09-02 16:48:02 CDT; 18min ago
  Process: 465 ExecStartPre=/bin/bash /var/mycodo-root/install/influxdb_wait_until_running.sh (code=exited, status=0/SUC
  Process: 601 ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py (code=exited, status=1

Sep 02 16:47:59 raspberrypi systemd[1]: Starting Mycodo server...
Sep 02 16:48:02 raspberrypi python[601]: Traceback (most recent call last):
Sep 02 16:48:02 raspberrypi python[601]:   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 40, in <module>
Sep 02 16:48:02 raspberrypi python[601]:     from Pyro5.api import serve
Sep 02 16:48:02 raspberrypi python[601]: ModuleNotFoundError: No module named 'Pyro5'
Sep 02 16:48:02 raspberrypi systemd[1]: mycodo.service: Control process exited, code=exited, status=1/FAILURE
Sep 02 16:48:02 raspberrypi systemd[1]: mycodo.service: Failed with result 'exit-code'.
Sep 02 16:48:02 raspberrypi systemd[1]: Failed to start Mycodo server.

kizniche, 2020-09-02:
Please provide the output from the command you listed.

griffira, 2020-09-02:
As requested: (thanks!)

pi@raspberrypi:~ $ sudo ~/Mycodo/env/bin/pip install setuptools==49.5.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting setuptools==49.5.0
  Using cached setuptools-49.5.0-py3-none-any.whl (803 kB)
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 50.0.3
    Uninstalling setuptools-50.0.3:
      Successfully uninstalled setuptools-50.0.3
Successfully installed setuptools-49.5.0
pi@raspberrypi:~ $ sudo bash ~/Mycodo/mycodo/scripts/upgrade_post.sh

#### Running initialization

#### Creating mycodo user
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `dialout'.
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `i2c'.
The user `mycodo' is already a member of `kmem'.
The user `mycodo' is already a member of `video'.
The user `pi' is already a member of `mycodo'.
The user `mycodo' is already a member of `pi'.

#### Compiling mycodo_wrapper
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function ‘upgrade_commands’:
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:17:5: warning: ‘strncat’ specified bound 255 equals destination size [-Wstringop-overflow=]
     strncat(full_cmd, "/upgrade_commands.sh ", sizeof(full_cmd));
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function ‘main’:
/home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:35:4: warning: ‘strncat’ specified bound 255 equals destination size [-Wstringop-overflow=]
    strncat(restoreScript, "/upgrade_commands.sh backup-create", sizeof(restoreScript));
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

#### Creating symlinks to Mycodo executables

#### Creating files and directories

#### Setting permissions

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Checking python 3 virtualenv
#### Virtualenv already exists, skipping creation

#### Updating apt repositories
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done

#### Installing prerequisite apt packages and update pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'apache2' is not installed, so not removed
Package 'python-cffi-backend' is not installed, so not removed
Package 'python3-cffi-backend' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libjs-sphinxdoc libjs-underscore libpython-all-dev python-all python-all-dev python-asn1crypto python-configparser
  python-cookies python-crypto python-dbus python-entrypoints python-enum34 python-funcsigs python-jwt
  python-keyrings.alt python-mock python-pbr python-responses python-wheel python-xdg python3-asn1crypto
  python3-cookies python3-dbus python3-entrypoints python3-jwt python3-responses
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
[... 190 lines cut ...]
Appending to crontab:
-----------------------------------------------
@reboot /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove

#### Setting permissions

#### Restarting the Mycodo daemon
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/scripts/restart_daemon.py", line 12, in <module>
    from mycodo.config import DAEMON_PID_FILE
  File "/home/pi/Mycodo/mycodo/config.py", line 10, in <module>
    from flask_babel import lazy_gettext
ModuleNotFoundError: No module named 'flask_babel'
Job for mycodo.service failed because the control process exited with error code.
See "systemctl status mycodo.service" and "journalctl -xe" for details.

#### Restarting nginx
#### Reloading mycodoflask
mycodoflask.service is not active, cannot reload.

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
^C

kizniche, 2020-09-02:
Okay. I edited my previous comment with a solution that should work. Follow it again and see if there are still issues.

ML1999, 2020-09-03:
**It worked for me.** Steps:

**1)** sudo vi ~/Mycodo/install/requirements.txt 
 Add line  setuptools==49.5.0  then save exit (:wq!)

**2)** sudo bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
You will get the same error #### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
Just Ctrl C to drop to a prompt

**3)** sudo shutdown now -r

**4)** log back in after the reboot and check the mycodo service

pi@pi3b:~ $ sudo service mycodo status
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-09-02 20:05:13 MDT; 1min 2s ago
  Process: 564 ExecStartPre=/bin/bash /var/mycodo-root/install/influxdb_wait_until_running.sh (code=exited, status=0/SUC
  Process: 603 ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py (code=exited, status=0
 Main PID: 639 (python)
    Tasks: 9 (limit: 2063)
   CGroup: /system.slice/mycodo.service
           └─639 /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py

Sep 02 20:04:59 pi3b systemd[1]: Starting Mycodo server...
Sep 02 20:05:13 pi3b systemd[1]: Started Mycodo server.
Sep 02 20:05:13 pi3b mycodo_daemon[639]: Starting daemon.
lines 1-13/13 (END)

griffira, 2020-09-03:
**Worked perfectly for me as well. Thanks so much for the very prompt responses.**

> Okay. It looks like the version of setuptools we specified is being uninstalled for the newer version, so we'll have to add it to >the requirements.txt file. I edited my previous comment with a solution that should work. Follow it again and see if there are >still issues.

griffira, 2020-09-03:
I did notice a message about the database version while poking around for the first time. Don't know if this could be a potential problem or not:

Mycodo Version: 8.7.2
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Database Version: 90wvmtnznxb8 (Incorrect Version. Should be d66e33093e8e) 
Daemon Status: Running
Daemon Process ID: 930
Daemon RAM Usage: 42.728 MB
Daemon Virtualenv: Yes
Frontend Process ID: 506
Frontend RAM Usage: 52.784 MB
Frontend Virtualenv: Yes

kizniche, 2020-09-03:
Can anyone else verify the same behavior (reporting those database versions) that installed Mycodo via this method?

It is impossible for Mycodo to even run if it's indeed at that version, since that is the base version for 7.0.0. I'm not sure what's going on there, but issuing the following commands may fix it:

```
cd ~/Mycodo/databases
../env/bin/alembic upgrade head
../env/bin/python ./alembic_post.py
```

ML1999, 2020-09-04:
> Can anyone else verify the same behavior (reporting those database versions) who installed Mycodo via this method?
> ```

In the UI under system info  it  complained (red) about the DB being an older version.  I updated it and now it is happy(green) Database Version: d66e33093e8e

I am now having an issue with no data coming through the GPIO pins, not sure if it is related just starting to dig in.

griffira, 2020-09-04:
Ok,I got this when I ran the second command. Looks like there's some database inconsistencies:

pi@raspberrypi:~/Mycodo/databases $ sudo ../env/bin/alembic upgrade head
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
INFO  [alembic.runtime.migration] Running upgrade 90wvmtnznxb8 -> 8e5a8351ad7a, Add Math option order_of_use
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1278, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 593, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: duplicate column name: order_of_use

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "../env/bin/alembic", line 8, in <module>
    sys.exit(main())
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 577, in main
    CommandLine(prog=prog).main(argv=argv)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 571, in main
    self.run_cmd(cfg, options)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 551, in run_cmd
    **dict((k, getattr(options, k, None)) for k in kwarg)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/command.py", line 298, in upgrade
    script.run_env()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/script/base.py", line 489, in run_env
    util.load_python_file(self.dir, "env.py")
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/util/pyfiles.py", line 98, in load_python_file
    module = load_module_py(module_id, path)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/util/compat.py", line 184, in load_module_py
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "alembic/env.py", line 69, in <module>
    run_migrations_online()
  File "alembic/env.py", line 64, in run_migrations_online
    context.run_migrations()
  File "<string>", line 8, in run_migrations
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/runtime/environment.py", line 846, in run_migrations
    self.get_context().run_migrations(**kw)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/runtime/migration.py", line 520, in run_migrations
    step.migration_fn(**kw)
  File "/home/pi/Mycodo/databases/alembic/versions/8e5a8351ad7a_add_math_option_order_of_use.py", line 21, in upgrade
    batch_op.add_column(sa.Column('order_of_use', sa.Text))
  File "/usr/lib/python3.7/contextlib.py", line 119, in __exit__
    next(self.gen)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/operations/base.py", line 354, in batch_alter_table
    impl.flush()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/operations/batch.py", line 83, in flush
    fn(*arg, **kw)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/ddl/impl.py", line 237, in add_column
    self._exec(base.AddColumn(table_name, column, schema=schema))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/alembic/ddl/impl.py", line 140, in _exec
    return conn.execute(construct, *multiparams, **params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1014, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/sql/ddl.py", line 72, in _execute_on_connection
    return connection._execute_ddl(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1076, in _execute_ddl
    compiled,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1318, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1512, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 178, in raise_
    raise exception
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1278, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 593, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) duplicate column name: order_of_use
[SQL: ALTER TABLE math ADD COLUMN order_of_use TEXT]
(Background on this error at: http://sqlalche.me/e/13/e3q8)

kizniche, 2020-09-06:
You likely don't have a database schema issue if your system runs fine. However, the database version is an issue for the upgrade system, because it will prevent you from upgrading to any future release. I can't explain how you would have a database mismatch issue with a fresh install. In any case, I would recommend deleting your database and recreating the database. This will delete all your settings, users, etc.

```
sudo rm -rf ~/Mycodo/databases/mycodo.db
sudo service mycodoflask restart
```

Then load the web interface and it will create a new database and prompt you to create an admin user. You should now have the correct database version.

griffira, 2020-09-07:
Thank you!! This worked flawlessly.  I now have this when I look at the "about" Mycodo:

Mycodo Version: 8.7.2
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Database Version: d66e33093e8e
Daemon Status: Running
Daemon Process ID: 31334
Daemon RAM Usage: 38.308 MB
Daemon Virtualenv: Yes
Frontend Process ID: 31223
Frontend RAM Usage: 52.648 MB
Frontend Virtualenv: Yes

ML1999, 2020-09-08:
@griffira are you able to upgrade from the UI?

I performed the same steps and my upgrade is broken

pi@pi3b:~ $ sudo rm -rf ~/Mycodo/databases/mycodo.db
pi@pi3b:~ $ sudo service mycodoflask restart
pi@pi3b:~ $ sudo service mycodo restart
---------------------------------------------------------------

Mycodo Version: 8.7.2
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Database Version: d66e33093e8e
Daemon Status: Running
Daemon Process ID: 19893
Daemon RAM Usage: 42.472 MB
Daemon Virtualenv: Yes
Frontend Process ID: 19872
Frontend RAM Usage: 52.724 MB
Frontend Virtualenv: Yes
-------------------------------------------------------------------

https://192.168.x.x/admin/upgrade

Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 423, in admin_upgrade
    is_internet=False)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 2, in top-level template code
    {% set active_page = "upgrade" %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 325, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 52, in block "body"
    {% elif current_latest_major_version != current_release.split('.')[0] and
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 471, in getattr
    return getattr(obj, attribute)
**jinja2.exceptions.UndefinedError: 'current_release' is undefined**

kizniche, 2020-09-09:
version 8.8.0 released. Also, setuptools has been updated that should fix the install issue.

closed 2020-09-09 by kizniche