kizniche/Mycodo #833: Install keeps failing on brand new Raspberry Pi 4

griffira, 2020-09-02:
Hello,

I continue to have problems with the install. I even re-imaged my SD card to ensure I was starting out fresh then, tried to install Mycodo immediately thereafter, to no avail. If it means anything, I remember the "ModuleNotFoundError: No module named 'setuptools._distutils'" error coming up before on the install attempts before I wiped the SD card.

Thanks in advance for any help...I've got a brand new Atlas pH probe kit I'm itching to play with ;-)

From the setup log at the first sign of trouble:

Collecting bcrypt==3.1.7
  Downloading bcrypt-3.1.7.tar.gz (42 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
ERROR: Exception:
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 216, in _main
    status = self.run(options, args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 182, in wrapper
    return func(self, options, args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 325, in run
    reqs, check_supported_wheels=not options.target_dir
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 183, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 388, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 340, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 483, in prepare_linked_requirement
    req, self.req_tracker, self.finder, self.build_isolation,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 91, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pep517/wrappers.py", line 161, in get_requires_for_build_wheel
    'config_settings': config_settings
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pep517/wrappers.py", line 265, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 86, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 6, in <module>
    import distutils.core
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/_distutils_hack/__init__.py", line 83, in create_module
    return importlib.import_module('setuptools._distutils')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'setuptools._distutils'

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pylibftdi==0.19.0
  Downloading pylibftdi-0.19.0-py2.py3-none-any.whl (32 kB)
Requirement already up-to-date: RPi.GPIO in /usr/lib/python3/dist-packages (from -r /home/pi/Mycodo/install/requirements-rpi.txt (line 3)) (0.7.0)
Collecting smbus2==0.3.0
  Downloading https://www.piwheels.org/simple/smbus2/smbus2-0.3.0-py2.py3-none-any.whl (9.1 kB)
Installing collected packages: pylibftdi, smbus2
[... 188 lines cut ...]
#### Installing and configuring nginx web server
Failed to disable unit: Unit file mycodoflask.service does not exist.
Synchronizing state of nginx.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable nginx
Created symlink /etc/systemd/system/multi-user.target.wants/mycodoflask.service → /home/pi/Mycodo/install/mycodoflask.service.
Created symlink /etc/systemd/system/mycodoflask.service → /home/pi/Mycodo/install/mycodoflask.service.

#### Restarting nginx
#### Restarting mycodoflask

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...

ALSO:

pi@raspberrypi:~ $ sudo journalctl -u mycodoflask | tail -n 50
-- Logs begin at Thu 2019-02-14 04:11:59 CST, end at Wed 2020-09-02 15:23:08 CDT. --
Sep 02 15:20:15 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
Sep 02 15:20:15 raspberrypi systemd[3939]: mycodoflask.service: Failed to execute command: No such file or directory
Sep 02 15:20:15 raspberrypi systemd[3939]: mycodoflask.service: Failed at step EXEC spawning /var/mycodo-root/env/bin/gunicorn: No such file or directory
Sep 02 15:20:15 raspberrypi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=203/EXEC
Sep 02 15:20:15 raspberrypi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.

kizniche, 2020-09-02:
Duplicate of #832

closed 2020-09-02 by kizniche

kizniche, 2020-09-02:
I just posted in the original issue a workaround that will allow the install to complete.