kizniche/Mycodo #835: Refresh Custom widget for development purpose

Gargam, 2020-09-04:
### Describe the problem/bug

I'm trying to develop a custom widget but I can't find a way to refresh / see result of the widget in the dashboard to get instant feedback.
I have tried to modify it in place (in the custom_widget directory) and to restart the daemon without any success.
The only thing that worked was deleting and reinstalling the widget.

Is there a recommended way to achieve this or some cache to remove or a debug mode to force refresh each time ?

### Versions:

 - Mycodo Version: 8.7.2
 - Raspberry Pi Version: B+
 - Raspbian OS Version: Buster lite

kizniche, 2020-09-04:
Widgets are displayed by the frontend (Flask), so you'll need to restart the frontend from the Configuration dropdown menu.

kizniche, 2020-09-04:
Additionally, the widget HTML code needs to be regenerated, which occurs upon import of the module from the Configure -> Widget menu. This can also be accomplished from the mycodo_commands.py (use --help to see exact command).

Gargam, 2020-09-04:
Hi again, 

Yup I have tried to restart the frontend as well without success.

Calling the command with web-server-reload && generate-widget-html is working !

Thanks.

closed 2020-09-09 by kizniche