kizniche/Mycodo #836: Cannot set a PID Controller in PID: Pause action

dylandn, 2020-09-07:
# Describe the problem/bug

I cannot set a trigger function to pause a PID.
I have a "Trigger: Timer (Daily Point)" function, I've added a "PID: Pause" action to the function. It shows up at the bottom of the trigger. I then set the PID for this action to Pause to the desired PID, and click save. Upon reloading and returning to that Trigger function, the PID: Pause action shows a blank entry where the desired PID should be listed. 

There are 3 PID controllers in this particular setup. I cannot set the PID: Pause to pause any of them.

### Versions:

 - Mycodo Version: 8.7.2
 - Raspberry Pi Version: 4
 - Raspbian OS Version: latest

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

Create a new Trigger Timer 
Add a new action to the new Trigger: PID: Pause
Set the Controller to the new PID: Pause action, click save
see blank Controller field 

### Expected behavior

PID: Pause action should show saved PID controller

dylandn, 2020-09-07:
Addition note: PID: Resume action _will_ save the selected PID controller and display it.

kizniche, 2020-09-07:
Thanks for the bug report. A fix will be available in the next release.

closed 2020-09-09 by kizniche