kizniche/Mycodo #837: Remote Control Board Interface (eg ESP32)

gaffo, 2020-09-07:
**Is your feature request related to a problem? Please describe.**

My feature request: Consider how a remote control system might interoperate (or maybe you have) with Mycodo. In systems like openhab they can expose or push state (settings), and can have state changes (on off) pushed to them. 

**Describe the solution you'd like**

I'd like to know, if I were to mess around with it, what would be the right way to approach a more IoT style physical object control scheme, where the devices were not physically connected to the GPIO of the device.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Pushing data directly to Influx and allowing "Inputs" from Influx as an input.
Pulling data from Influx as control settings (on / off).

Doing the same with MQTT.

**Additional context**
I built out a mushroom monitoring system similar to your (https://kylegabriel.com/projects/2015/04/mushroom-cultivation-revisited.html) based on an ESP32. Mine was built where the ESP32 just pushes to prometheus and then I check things in Graphana. This let me see how off my inkbird controllers were. Graphana/Prom are running on a RPI4. I was about to start building hub software when I came across Mycodo.

Other note:

You're actually not far off from a industrial control system and I like how you've used python for extensibility here, cool design!

kizniche, 2020-09-08:
For remote inputs/outputs, I'd recommend using MQTT. The Mosquito MQTT server runs nicely on a Raspberry Pi and there are already an MQTT input and output that can be used, in addition to the ability to create your own input/output modules you can import. You could enter data directly into influxdb, but you would have to have the measurement and tags correct for them to be displayed/used by Mycodo. I would recommend creating a custom Input instead, as this is more likely to work as a long-term solution (through version updates).

gaffo, 2020-09-09:
Thank you, I'll start there!

closed 2020-09-09 by gaffo