kizniche/Mycodo #838: Saving measurment as String

Btazbt, 2020-09-07:
# Describe the problem/bug

This may be more of an New Feature Request, but I can' tell.  I've created a custom input for a valve to be 'open' or 'closed' based on a GPIO state.  I see the following error:

> 2020-09-07 09:53:18,132 - ERROR - mycodo.inputs.gpio_openclosed_14e7210f - InputModule raised an exception when taking a reading: could not convert string to float : 'Open'

To create my custom input I had also created a custom measurement Valve State with the unit being Position.    I can't seem to allow mycodo to save a measurement as a string.  Is this not supported or am I doing something wrong? 

`
# Measurements
measurements_dict = {
    0: {
        'measurement': 'gpio_state',
        'unit': 'bool'
    },
    1: {
        'measurement': 'valve_state',
        'unit': 'valve'
    }
}
`

`
        self.value_set(0, self.gpio.input(self.location))
        if self.value_get(0) == 0 :
            self.value_set(1, 'Open')
        else :
            self.value_set(1, 'Closed')
        return self.return_dict
`
 

### Versions:

 - Mycodo Version: 8.7.2
 - Raspberry Pi Version: 3B+

### Reproducibility

Repeats with each measurement of my custom input

### Expected behavior

string value of 'open' or 'closed' is saved.  Ultimately, I'd like a GPIO boolean 0/1 state to display as open/closed or to be able to label the 0/1 values as open/closed.

kizniche, 2020-09-08:
Only values can be stored in the time-series database.

Btazbt, 2020-09-08:
thanks. i suspected as much. I'll have to look into the display side of things to see if I can show the data differently.

closed 2020-09-08 by Btazbt