kizniche/Mycodo #842: SSD1306 OLED Display not seemingly usable on RPI4

jasonlaw0213, 2020-09-10:
Hey there!  I've got an SSD1306 I2C 128x64 (4 pins -- VCC, GND and the two I2C pins) OLED display that I'm trying to connect and work.  I've had no success, initially thinking it was a defective OLED-- but I now think I've determined the problem.

### Describe the problem/bug

The 128x6 I2C 4-pin display is not functional using the Setup->LCD option nor the examples in the Adafruit library directly, it mis-detects the board as a beagleboard -- seemingly because that library is deprecated and isn't setup to support the new processor (family, chip, dunno) in the Raspi4.

(This library has been deprecated! We are leaving this up for historical and research purposes but archiving the repository.

We are now only supporting the use of our CircuitPython libraries for use with Python.

Check out this guide for info on using OLEDs with the CircuitPython library: https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring)

### Versions:

 - Mycodo Version: [e.g. 8.0.3] 8.8.1
 - Raspberry Pi Version: [e.g. 3B+] 4
 - Raspbian OS Version: [e.g. Buster Lite] Buster Lite

### Reproducibility

This is continuously reproducible -- No pin setting, removal of reset pin entry (since no RST pin on OLED nor wired) resolves it.

After installing the new adafruit oled library (blinka/circuitpython_ssd1306) the adafruit example .py's now work flawlessly within python3:
python3 ssd1306_stats.py

However, within the Mycodo environment, it seemingly leverages the deprecated library or something as it does not light up a single pixel

Help getting this new oled library supported would be greatly appreciated!

Regards,

kizniche, 2020-09-11:
I've been meaning to add the new circuitpython library as a new Mycodo LCD module. I'll try to come up with something soon.

kizniche, 2020-10-20:
I just added 128x32 and 128x64 PiOLED using the Adafruit CircuitPython library for both I2C and SPI interfaces. Let me know if you are able to test and if it works. You will need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to acquire the new code.

dexilimes, 2020-10-21:
Hi there, I am also having the same issue. I tried upgrading to master, and running the OLED screen with the new CircuitPython settings. Unfortunately it did not activate. 
This is what the debug log puts out:

File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 297, in initialize_variables
    self.lcd_out = LCD_Pioled_Circuitpython(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 60, in __init__
    self.disp.begin()
AttributeError: 'SSD1306_I2C' object has no attribute 'begin'
2020-10-21 18:44:09,119 - INFO - mycodo.controllers.controller_lcd_37bf78de - Activated in 389.8 ms

kizniche, 2020-10-27:
I just pushed a potential fix. Let me know if it works if you test.

dexilimes, 2020-10-28:
Still getting the same error:
2020-10-28 12:11:31,311 - ERROR - mycodo.controllers.controller_lcd_1bbcf30c - initialize_variables() Exception: 'SSD1306_I2C' object has no attribute 'begin'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 297, in initialize_variables
    self.lcd_out = LCD_Pioled_Circuitpython(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 60, in __init__
    self.disp.begin()
AttributeError: 'SSD1306_I2C' object has no attribute 'begin'

kizniche, 2020-10-28:
I deleted that line, so it appears you did not upgrade to master ([1](https://github.com/kizniche/Mycodo/issues/842#issuecomment-712942814))

dexilimes, 2020-10-28:
i just updated to master, but got the same error:
2020-10-28 13:47:08,862 - ERROR - mycodo.controllers.controller_lcd_1bbcf30c - initialize_variables() Exception: 'SSD1306_I2C' object has no attribute 'begin'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 297, in initialize_variables
    self.lcd_out = LCD_Pioled_Circuitpython(lcd_dev)
  File "/var/mycodo-root/mycodo/devices/lcd_pioled_circuitpython.py", line 60, in __init__
    def lcd_init(self):
AttributeError: 'SSD1306_I2C' object has no attribute 'begin'
2020-10-28 13:47:08,862 - INFO - mycodo.controllers.controller_lcd_1bbcf30c - Activated in 498.3 ms

Then when I decativated the LED again
2020-10-28 13:47:12,195 - DEBUG - mycodo.daemon - Could not deactivate LCD controller with ID 1bbcf30c-807f-4b3d-bdce-ef7d7d27ff78, it's not active.

dexilimes, 2020-10-28:
I checked the lcd_pioled_cicrcuitpython.py file and the begin has been deleted.

dexilimes, 2020-10-28:
I rebooted the system and the oled is now working, but,  its not displaying the correct data. 
I had the following settings:
line 1:humidity input
line 2: temp input
line 3:IP adress
line 4:blank

but the OLED is displaying:
mycodo 8.8.7
stop (128x32) OLED (sd13....... (i cant see the rest)

dexilimes, 2020-10-28:
Forget the last comment, it jumps to the correct settings after that screen. 
Thanks so much for getting this to work! your an absolute legend!

It is displaying the sensor name AM2315/AM2320 71.1%. am I able to change this to say what is being measured, rather than the sensor name? i.c. Humidity

dexilimes, 2020-10-28:
My apologies, I got it to work by making the measurements a different input name on the input menu. 

Thanks heaps for all your help - Mycodo is an amazing resource

jasonlaw0213, 2020-10-29:
As always, @kizniche , you are a gentleman and a scholar.

The OLED I alluded to in my initial post activates now.

However, no data values for the different inputs, outputs is displayed.  I’ve reset the raspi/Mycodo multiple times and toyed with the values, renamed my outputs to shorten them.

Any thoughts?

[image]

kizniche, 2020-10-29:
Is the Max Age for each line longer than the Period for the corresponding Input? The LCD Line will only look for a measurement in the past x seconds (Max Age).

jasonlaw0213, 2020-10-29:
The input (DHT22 as example) Period is 15 seconds.
LCD Module Period is 20 seconds.
The LCD data lines Max Ages are default 360 seconds.

The outputs (relay controlled outlets) are on/off GPIO but I want to be able to track some of their statuses / on durations -- no max ages or periods on those.

Do I have my age/period values skewed?  Sorry to come to you for troubleshooting, I guess I'm not as adept as I thought I was.

kizniche, 2020-10-29:
There's an ongoing power outage due to a hurricane in the southeast US, so I won't be able to investigate until it's restored.

jasonlaw0213, 2020-10-29:
Oh, Please!!! 

I don't think any of us expect expedient troubleshooting responses within the realm of hours or days and we're definitely not entitled to the amount of help you've provided throughout all these issue threads!

Let alone in the middle of an emergency, natural or otherwise!

I hope you're staying safe and can ride this out calmly without any real impact to you.

grux77, 2020-10-30:
I also managed to activate my OLED 128x32 SD1306 with the master-version.
With "[I2C] 128x32 OLED (SD1306, CircuitPython)".
Gorgeous reading!
Now you can throw away "128x32 OLED (SD1306, Adafruit_SSD1306)".

lu-krin, 2020-10-31:
I don't know anything about python but I used to be a developer so I did a little digging and solved the issue "no attribute begin" issue tonight like this:

Updated file /Mycodo/mycodo/devices/lcd_pioled_circuitpython.py - see # comments for my changes

```
# coding=utf-8
import logging
import time

import adafruit_ssd1306
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont
from adafruit_extended_bus import ExtendedI2C

logger = logging.getLogger("mycodo.device.lcd_pioled")

class LCD_Pioled_Circuitpython:
    """Output to the PiOLED I2C LCD"""

    def __init__(self, lcd_dev):
        self.logger = logging.getLogger(
            "{}_{}".format(__name__, lcd_dev.unique_id.split('-')[0]))

        self.disp = None
        self.interface = 'I2C' #  --------------------------------------------- changed - hardcoded
        self.lcd_x_characters = lcd_dev.x_characters
        self.lcd_y_lines = lcd_dev.y_lines
        self.lcd_type = '128x64_pioled_circuit_python' #  --------------------------------------------- changed - hardcoded

        if self.interface == 'I2C':
            if self.lcd_type == '128x32_pioled_circuit_python':
                self.disp = adafruit_ssd1306.SSD1306_I2C(
                    128, 32,
                    ExtendedI2C(lcd_dev.i2c_bus),
                    addr=int(str(lcd_dev.location), 16))
            elif self.lcd_type == '128x64_pioled_circuit_python':
                self.disp = adafruit_ssd1306.SSD1306_I2C(
                    128, 64,
                    ExtendedI2C(lcd_dev.i2c_bus),
                    addr=int(str(lcd_dev.location), 16))

        elif self.interface == 'SPI':
            if self.lcd_type == '128x32_pioled_circuit_python':
                import Adafruit_GPIO.SPI as SPI
                self.disp = adafruit_ssd1306.SSD1306_SPI(
                    128, 32,
                    spi=SPI.SpiDev(lcd_dev.spi_bus, lcd_dev.spi_device),
                    dc=lcd_dev.pin_dc,
                    reset=lcd_dev.pin_reset,
                    cs=lcd_dev.pin_cs)
            elif self.lcd_type == '128x64_pioled_circuit_python':
                import Adafruit_GPIO.SPI as SPI
                self.disp = adafruit_ssd1306.SSD1306_SPI(
                    128, 64,
                    spi=SPI.SpiDev(lcd_dev.spi_bus, lcd_dev.spi_device),
                    dc=lcd_dev.pin_dc,
                    reset=lcd_dev.pin_reset,
                    cs=lcd_dev.pin_cs)

        if not self.disp:
            self.logger.error("Unable to set up display. Check the LCD settings.")
        else:
            self.logger.error("Pass")  #  --------------------------------------------- changed - self.disp.begin() didn't do anything and was throwing errors

    def lcd_init(self):
        """ Initialize LCD display """
        try:
            self.disp.fill(0) #  --------------------------------------------- changed to function used in https://learn.adafruit.com/adafruit-pioled-128x32-mini-oled-for-raspberry-pi/usage
            self.disp.show() #  --------------------------------------------- changed as per URL above
        except Exception as err:
            self.logger.error(
                "Could not initialize LCD. Check your configuration and wiring. Error: {err}".format(err=err))

    def lcd_write_lines(self,
                        message_line_1,
                        message_line_2,
                        message_line_3,
                        message_line_4,
                        message_line_5=None,
                        message_line_6=None,
                        message_line_7=None,
                        message_line_8=None):
        """ Send strings to display """
        x = 0
        top = -2  # padding
        font = ImageFont.load_default()

        image = Image.new('1', (self.disp.width, self.disp.height))

        draw = ImageDraw.Draw(image)
        draw.rectangle((0, 0, self.disp.width, self.disp.height), outline=0, fill=0)

        draw.text((x, top), message_line_1, font=font, fill=255)
        draw.text((x, top + 8), message_line_2, font=font, fill=255)
        draw.text((x, top + 16), message_line_3, font=font, fill=255)
        draw.text((x, top + 24), message_line_4, font=font, fill=255)

        if message_line_5 is not None:
            draw.text((x, top + 32), message_line_5, font=font, fill=255)
        if message_line_6 is not None:
            draw.text((x, top + 40), message_line_6, font=font, fill=255)
        if message_line_7 is not None:
            draw.text((x, top + 48), message_line_7, font=font, fill=255)
        if message_line_8 is not None:
            draw.text((x, top + 54), message_line_8, font=font, fill=255)

        self.disp.image(image)
        self.disp.show() # --------------------------------------------- changed as per URL above
        time.sleep(0.1)

    def lcd_backlight(self, state):
        """ backlight not supported """
        pass

```

kizniche, 2020-10-31:
I just want to point out this was fixed 3 days ago in https://github.com/kizniche/Mycodo/commit/f4b3e560c03f48412bd4327200291579eb42013a, which I reference how to get the latest code in https://github.com/kizniche/Mycodo/issues/842#issuecomment-717581607 and https://github.com/kizniche/Mycodo/issues/842#issuecomment-717640472

Since this is getting a lot of attention, I'll make a v8.8.8 bugfix release tonight.

kizniche, 2020-10-31:
v8.8.8 is released. Please let me know if this issue persists after you upgrade. Thanks,

closed 2020-10-31 by kizniche

jasonlaw0213, 2020-11-06:
Hey, Kyle!  Hope you're doing well and survived the storms/outages.

Thanks for the update, but I still have a "NO DATA" issue for all my Data sensors aside from CPU/GPU temp.   The rest of the Data sensors/inputs/etc keep showing "NO DATA" on the OLED despite tweaking the Max Age values to try and get it to display info for these devices.

I can't tell if it's a bug/issue or something I need to resolve on my side.

Any thoughts to troubleshoot?

kizniche, 2021-03-26:
@jasonlaw0213 Your last comment must have escaped me since this issue was closed. Are you still experiencing this issue in the latest release? If so, reopen this issue and we'll continue investigating.