kizniche/Mycodo #843: TypeError: '>' not supported between instances of 'float' and 'NoneType' in self.loop()

dylandn, 2020-09-11:
# Describe the problem/bug
When enabling either of my PID controllers, the mycodo log is flooded with error messages, nearly 10 errors per second. 

With one PID controller activated, the log shows: 
2020-09-10 20:40:31,523 - ERROR - mycodo.controllers.controller_pid_df606f36 - loop() Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 84, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 146, in loop
    elif time.time() > self.timer:
TypeError: '>' not supported between instances of 'float' and 'NoneType'
2020-09-10 20:40:31,624 - ERROR - mycodo.controllers.controller_pid_df606f36 - loop() Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 84, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 146, in loop
    elif time.time() > self.timer:
TypeError: '>' not supported between instances of 'float' and 'NoneType'
2020-09-10 20:40:31,725 - ERROR - mycodo.controllers.controller_pid_df606f36 - loop() Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 84, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 146, in loop
    elif time.time() > self.timer:
TypeError: '>' not supported between instances of 'float' and 'NoneType'
2020-09-10 20:40:31,827 - ERROR - mycodo.controllers.controller_pid_df606f36 - loop() Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 84, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 146, in loop
    elif time.time() > self.timer:
TypeError: '>' not supported between instances of 'float' and 'NoneType'

### Versions:

 - Mycodo Version: 8.8.1
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: latest

### Reproducibility

When both PIDs are deactivated, the log is quiet

### Expected behavior
No errors.

kizniche, 2020-09-11:
You appear to be missing a crucial part of your log. Please post the excerpt that includes the PID controller being activating and the errors immediately proceeding activation.

dylandn, 2020-09-11:
2020-09-10 20:55:26,685 - ERROR - mycodo.controllers.controller_pid_5dc5f93f - initialize_variables() Exception: 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 179, in initialize_variables
    self.raise_output_id = pid.raise_output_id.split(",")[0]
AttributeError: 'NoneType' object has no attribute 'split'
2020-09-10 20:55:26,686 - INFO - mycodo.controllers.controller_pid_5dc5f93f - Activated in 112.9 ms
2020-09-10 20:55:26,687 - ERROR - mycodo.controllers.controller_pid_5dc5f93f - loop() Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 84, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 146, in loop
    elif time.time() > self.timer:
TypeError: '>' not supported between instances of 'float' and 'NoneType'

kizniche, 2020-09-11:
Okay. There appears to be a bug in the PID controller. If you need it's use, I'd recommend restoring a pre-8.8.0 backup until a fix can be released.

kizniche, 2020-09-11:
The bug appears to be that both raise and lower outputs need to be set. Your unset output can be configured with a dummy output to allow the PID to operate, until the fix is made.

dylandn, 2020-09-11:
Indeed, setting the unused output to a dummy makes the error stop. Thank you!

dylandn, 2020-09-11:
I'm not sure if the next error messages are related to this problem, but now my PID controller spits out new error messages:
2020-09-10 21:44:27,090 - ERROR - mycodo.outputs.pwm_gpio_c97f126a - Cannot manipulate Output c97f126a-a41f-4846-9667-55581df2bcb0: output channel doesn't exist: 32ab959d-7e8f-4722-bc4c-f97ec046b68d
2020-09-10 21:44:37,109 - ERROR - mycodo.outputs.pwm_gpio_c97f126a - Cannot manipulate Output c97f126a-a41f-4846-9667-55581df2bcb0: output channel doesn't exist: 32ab959d-7e8f-4722-bc4c-f97ec046b68d
2020-09-10 21:44:47,156 - ERROR - mycodo.outputs.pwm_gpio_c97f126a - Cannot manipulate Output c97f126a-a41f-4846-9667-55581df2bcb0: output channel doesn't exist: 32ab959d-7e8f-4722-bc4c-f97ec046b68d
2020-09-10 21:44:57,195 - ERROR - mycodo.outputs.pwm_gpio_c97f126a - Cannot manipulate Output c97f126a-a41f-4846-9667-55581df2bcb0: output channel doesn't exist: 32ab959d-7e8f-4722-bc4c-f97ec046b68d

I have 2 PID controllers, one for temperature & one for humidity. Raising temperature is controlled with a on/off shell script output, lowering humidity is controlled by a PWM output. Both outputs seem work by themselves, but the PID controllers cant change them.

kizniche, 2020-09-11:
Just pushed a fix. I'll try to get the next release out ASAP, as this is a major bug.

kizniche, 2020-09-13:
Fixed in latest release

closed 2020-09-13 by kizniche