kizniche/Mycodo #844: Error adding gauges with > 4 stops to dashboard

rdrake, 2020-09-11:
### Describe the problem/bug

Attempting to add an angular gauge to a dashboard with > 4 stops resulted in an internal server error.

```
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 600, in page_dashboard
    unmet_dependencies = utils_dashboard.widget_add(form_base, request.form)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 202, in widget_add
    new_widget, dict_widgets[widget_name])
  File "/home/pi/Mycodo/mycodo/widgets/widget_gauge_angular.py", line 51, in execute_at_creation
    color = color_list[i + 1]
IndexError: list index out of range
```

### Versions:

 - Mycodo Version: 8.8.1
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Buster Lite

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behaviour:

1. Open a dashboard
2. Click on **Add Dashboard Widget** -> **Gauge (Angular)**
3. Change the number of steps to 5 (or more)
4. Click **Create**

### Expected behaviour

Widget added without error

### Additional context

Attempting to add a solid gauge with > 4 stops also results in the same error (line 49).  Modifying an existing widget to add additional stops does work.

kizniche, 2020-09-11:
Thanks for the bug report. In the meantime before a fix is made, you can delete your [unaccessible] dashboards and widgets from the Configure -> Diagnostic page.

kizniche, 2020-09-16:
v8.8.3 released with the fix

closed 2020-09-16 by kizniche