kizniche/Mycodo #845: Failed install on fresh system. 804 Error

slushmonk3y, 2020-09-14:
I am not super competent at this, but it is a fresh install of Raspbian and Mycodo. I am not sure what could be going wrong. I was hoping the new version of Mycodo would work, but I am still having the same error.

### Describe the problem/bug
I cant get it to install correctly. It mostly installs but it doesn't seem to be able to access the user table.  When the install gets to the "Generating widget HTML files" portion it gives this error:

2020-09-12 16:55:10,804 Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/databases/mycodo.db', error='(sqlite3.OperationalError) no such table: misc

When you open the web client and type in your info to create an admin account, it doesn't  seem to store the account.

Here is the setup log:
[setup.log](https://github.com/kizniche/Mycodo/files/5220686/setup.log)

### Versions:
 - Mycodo Version: 8.8.1
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: 10 (Buster)

kizniche, 2020-09-14:
I believe 8.8.1 has an install bug that was fixed with the release of 8.8.2. The current version is 8.8.3. Please try installing the latest version and see if the issue persists.

slushmonk3y, 2020-09-15:
OK. I deleted the Mycodo directory and reinstalled. 

It appears to be getting a similar error, but now it says "304 Error raised in session_scope." But still no such table.

[setup.log](https://github.com/kizniche/Mycodo/files/5221842/setup.log)

slushmonk3y, 2020-09-15:
oh yeah, that was Mycodo 8.8.2

kizniche, 2020-09-15:
Okay, I think I see the issue. I need to perform some tests.

slushmonk3y, 2020-09-15:
Awesome, thank you for the help.

kizniche, 2020-09-16:
v8.8.3 released with the fix

closed 2020-09-16 by kizniche

slushmonk3y, 2020-09-16:
Your a machine! It worked.