kizniche/Mycodo #848: AttributeError: 'Query' object has no attribute 'id' with Triggers & Conditional Controller (?)

dylandn, 2020-09-16:
### Describe the problem/bug
Errors in the log when my triggers attempt to activate/deactivate a controller.
I have two Triggers: Timer (Daily Point) and a Conditional controller. At separate times of the day, one trigger activates the  controller, the other deactivates the controller. 

An error is produced at every trigger:
2020-09-14 18:00:00,885 - ERROR - mycodo.trigger_action_adae23d7 - Error triggering action:
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/function_actions.py", line 1046, in trigger_action
    message = action_activate_controller(cond_action, message)
  File "/var/mycodo-root/mycodo/utils/function_actions.py", line 666, in action_activate_controller
    id=controller_entry.id,
AttributeError: 'Query' object has no attribute 'id'
2020-09-15 06:00:00,854 - ERROR - mycodo.trigger_action_49392521 - Error triggering action:
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/function_actions.py", line 1048, in trigger_action
    message = action_deactivate_controller(cond_action, message)
  File "/var/mycodo-root/mycodo/utils/function_actions.py", line 691, in action_deactivate_controller
    id=controller_entry.id,
AttributeError: 'Query' object has no attribute 'id'

### Versions:

 - Mycodo Version: 8.8.2
 - Raspberry Pi Version: 4
 - Raspbian OS Version: latest

kizniche, 2020-09-18:
I suspect you don't actually have your controller set for your Function Action. Please provide a screenshot of your Trigger Settings. I also just pushed a commit that adds debug lines to determine if you have it set. If you find you do have your Action set, and want to test further for the issue, [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), enable ```Log Level: Debug``` for your Trigger, and if the same error occurs, there will be a log line just prior to the error traceback that may indicate the issue.

dylandn, 2020-09-19:
You're right, I had a blank controller field that I filled in since. It's only happened that one time and not since. 
I can reproduce the error by setting the controller blank and triggering it.

closed 2020-09-21 by kizniche