kizniche/Mycodo #849: Error 502: Bad Gateway Error - When saving pH & EC function

kpslc, 2020-09-16:
This has happened as long as I've been using the system. Hoped the update to 8.8.3 would fix it. Not a big deal as the save occurs. Just have to go back in the browser history to redisplay the function page.

The interesting thing is the URL on the error page is exactly the same URL as the function page when it is opened properly.

I looked at the mycodo front end log file last night when this occurred after updating the system and there were no errors relating to this issue. Checked it again now after seeing the error again, it is unchanged from last night, meaning no new errors. That file is attached.
[mycodoflask log.txt](https://github.com/kizniche/Mycodo/files/5233122/mycodoflask.log.txt)

kizniche, 2020-09-16:
Please provide detailed steps to reproduce the issue.

kpslc, 2020-09-16:
I can make it happen every time when I change the eC & PH Function by clicking the main save button under the function. Typically I'm making a small change to the .py code and clicking "save". Clicking save on any condition or action inside the function does not cause the error.

kizniche, 2020-09-16:
There is no "EC & PH Function". If you mean "Controller: Conditional", I an unable to replicate the behavior by adding a Controller: Conditional, then clicking Save.

If you do not provide a detailed list of steps to reproduce the issue, there is nothing I can do.

kpslc, 2020-09-16:
Sorry. I have a Controller: Conditional setup for the pH and EC regulation. 

I just setup a new "test" conditional controller. Hitting save is okay, no errors. 
Next I copied in my python code (attached as a.txt file to allow upload) and hit save. Error occurred. This is the code that is running on my system. 
Next to test if it is a file length problem, I cut out all the code except a few comments at the top and hit save. This time it saved without the 502 error. It did offer a warning that my code was incomplete, but no 502 error.

Hope that helps. Again, it isn't a huge deal and the edits are saved before the 502 error.

[pHec Check WFUV.py.txt](https://github.com/kizniche/Mycodo/files/5233839/pHec.Check.WFUV.py.txt)

kizniche, 2020-09-16:
Thank you. I was able to replicate the issue with your example code. The actual error appeared in the nginx log:

```
2020/09/16 13:49:53 [error] 27351#27351: *2375 upstream sent too big header while reading response header from upstream, client: 10.48.44.99, server: , request: "POST /function HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/function", host: "10.48.44.193", referrer: "https://10.48.44.193/function"
```

And was alleviated by [increasing the proxy buffer size](https://ma.ttias.be/nginx-proxy-upstream-sent-big-header-reading-response-header-upstream/).

Saving of the form appears to be working, even with the error (it occurs after the submitted data is saved). The fix will be available in the next release.

kizniche, 2020-09-16:
Thank you. I was able to replicate the issue with your example code. The actual error appeared in the nginx log:

```
2020/09/16 13:49:53 [error] 27351#27351: *2375 upstream sent too big header while reading response header from upstream, client: x.x.x.x, server: , request: "POST /function HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/function", host: "x.x.x.x", referrer: "https://x.x.x.x/function"
```

And was alleviated by [increasing the proxy buffer size](https://ma.ttias.be/nginx-proxy-upstream-sent-big-header-reading-response-header-upstream/).

Saving of the form appears to be working, even with the error (it occurs after the submitted data is saved). The fix will be available in the next release.

closed 2020-09-16 by kpslc