kizniche/Mycodo #852: BME280 setup issue

DarthPleurotus, 2020-09-22:
I recently upgraded Mycodo and have been fiddling around with some of the sensors I have attached, including two BME280 temp/humidity sensors. I don't know specifically when, but at some point since upgrading one of my BME280s stopped working. Judging by the live data tab, Mycodo isn't receiving any data from the sensor.

The sensor in question is connected to I2C bus 3 at address 0x76. I tried `i2cdetect -y 3` and the sensor showed up normally. However, when I deactivate the sensor via the Mycodo interface I get the error 

> Error: Could not deactivate Input controller with ID 2fe61fcf-ab12-4366-8ce2-dfe355c4d71e: 'NoneType' object has no attribute 'stop_input'

This seems to indicate that Mycodo does not detect the sensor. Perhaps most confusingly, when I try to edit any of the setup information for either of my BME280s (in particularly I am trying to turn on debugging for the troublesome one), trying to save my edits produces the following error (and changes aren't saved):

> Error: Modify Input: Expecting value: line 1 column 1 (char 0)

### Versions:

 - Mycodo Version 8.8.3 (current stable)
 - Raspberry Pi 3B
 - Raspbian OS Version: Buster

kizniche, 2020-09-22:
> one of my BME280s stopped working

If one is working and another is not, this indicates the issue is not with Mycodo.

DarthPleurotus, 2020-09-22:
I am using the Adafruit library. I tried deleting and re-adding the input with the same library. That seemed to fix the main issue! 

There is still the odd little mystery of the second error message I mentioned. Now that I re-added the input for the first (troublesome) BME280 sensor, it no longer has this issue, but I still get the error message when trying to edit info for the second BME280 sensor. I will simply delete and add the input for this sensor too, so it's no problem for me; I just figured I would note that here in case someone else runs into a similar error message. Thanks!

closed 2020-09-22 by DarthPleurotus