kizniche/Mycodo #853: Custom Widget overriding baseline widget

Btazbt, 2020-09-25:
### Describe the problem/bug

I created a custom Indicator Widget  using the widget_indicator.py as the starting point.  My widget simple replaced the 'button-red.png' statements with 'button-closed.png' and I've placed in the flask img folder the additional image.  On a dashboard if I add the baseline widget it shows up correctly.  If I remove it and add the custom widget it shows up correctly.  But  if I add the baseline widget (which displays correctly), and then when I add my own widget then both widgets use my 'button-closed.png' image.  When I remove my custom widget the baseline widget goes back to the regular image.

kizniche, 2020-09-25:
I won't be able to diagnose any issue without seeing your code.

Btazbt, 2020-09-25:
Just figured out how to add files...

Here is with the baseline indicator, alone
[image]
Here is with the baseline indicator and the custom indicator.  Both now use the custom indicator graphic
[image]

this is the file (convert to .py) that I've added as my custom widget.

[widget_ocindicator.txt](https://github.com/kizniche/Mycodo/files/5285527/widget_ocindicator.txt)

kizniche, 2020-09-25:
Since your widget module uses the same function name "getLastDataIndicator", you will need to rename all instances in your module to prevent conflicts with the built-in indicator module that creates a function of the same name.

kizniche, 2020-09-25:
You will also need to rename all instances of repeatLastDataIndicator() in your module as well.

Btazbt, 2020-09-25:
thanks for the quick feedback.  maybe I'm not understanding the issue.  I tried simply renaming all getLastDataIndicator to getLastDataIndicator-BT and same with repeatLastDataIndicator, but there is no change.  What did you mean by renaming the instance?

kizniche, 2020-09-25:
Did you also reimport the module from the import section of the settings?

Btazbt, 2020-09-26:
no, after the first import I've just been replacing the python file in the custom folder.  Does it need to be re-imported with each change?

kizniche, 2020-09-26:
No, there are alternate methods to update the Mycodo environment if you're manually editing Mycodo system files, but it's not something I will go into here. By not using the widget import system, it makes my diagnosing your issue take longer and require more effort, since you're neither using the method of importing widgets that I've documented in the manual, nor mentioning you're deviating from the import process I've built in. There are processes that occur during import that are crucial to the preparation of widget modules and the Mycodo environment,and your issue is likely a result of your deviation from the manual. Please delete your custom widget from the widget settings page and reimport it to test the code changes I recommended.

Btazbt, 2020-09-26:
I completely understand that not following the written directions can cause unthought of issues....I say the same thing at work.   I'm just dangerous enough to somewhat know what I'm doing, but don't know all the backend of things.  I'll toy with this a little bit and see what I can do.  I'm certain the problem is me and not the overall mycodo work now that you have given me a few things to check into.  I really appreciate the work you have done and quick responses to this thread.  I'll check back in on this issue once I've had some time to toy with things.  thanks again.

kizniche, 2020-09-26:
Well, if I wasn't clear, I'm 99% certain your issue is caused by not using the import feature. It is what is expected to happen if you don't use it.

closed 2020-10-02 by kizniche

Btazbt, 2020-10-06:
I'm happy to report that I've resolved the issue.  I took an upgrade and noticed the baseline widget was updated so I started over from stratch on my custom one.  This time using the import with each change.  Took me a bit to realize that the upgrade removed my custom images (or perhaps I did, but I don't think so) that I placed in the flask static/img folder.  Any suggestions on where to place custom images so that they don't get removed during an update?

kizniche, 2020-10-06:
Any location not in ~/Mycodo won't be touched during an upgrade.

Btazbt, 2020-10-06:
Naturally, I was just looking to see if there were any existing options withing Mycodo to keep everything in one place.  Something similar to the custom folders.