kizniche/Mycodo #854: Waveshare AD DA input (ADS1256) unable to get readings

gecastan, 2020-09-28:
Hi there,

First off thanks for the great work.
Successfully used Mycodo with ADS1115 and upgraded to a more complete hat (easier to wire) : Waveshare AD DA (ADS1256) 
Sadly I m unable to get readings from it using Mycodo (sample test from the device without mycodo is working fine tho, so wiring is OK)

Product link : https://www.amazon.com/Raspberry-Expansion-High-Precision-Functions-Interface/dp/B071JGZM4B/ref=pd_lpo_sbs_147_img_1?_encoding=UTF8&psc=1&refRID=86DDDF05ZWS6N44E88JJ

As reference, board is working properly following the official documentation : https://www.waveshare.com/wiki/High-Precision_AD/DA_Board
And the sample test can be found here : https://www.waveshare.com/wiki/File:High-Precision-AD-DA-Board-Code.7z
And here is the output (chan 0 = potentiometer, chan1 reserved for something else and I have a soil moisture sensor on channel 7 which shows voltage going down when put into water as intended)

`Waveshare/RaspberryPI/ADS1256/python3 $ python3 main.py
ID Read success
0 ADC = 5.000000
1 ADC = 4.478990
2 ADC = 1.193992
3 ADC = 1.192906
4 ADC = 1.193935
5 ADC = 1.192706
6 ADC = 1.193436
7 ADC = 2.656148
`

On the the issue with Mycodo :

To add the device I proceeded as follow
- Setup --> Data
- Selected Texas Instrument ADS1256, Waveshare Analog to Digital Converter
- Installed required libraries from the UI

Here is the output from /var/log/mycodo/mycodo.log :
`2020-09-28 01:31:59,536 - ERROR - mycodo.controllers.controller_input_5ff575b2 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-09-28 01:32:05,346 - ERROR - mycodo.controllers.controller_input_774f54df - Error while attempting to read input: 'NoneType' object has no attribute 'next'`

Before creating this issue, i tried to find some help on the wiki / previous issues but I m currently stuck.
Any help to debug this out is welcome.

Thanks

kizniche, 2020-09-29:
Did you configure the appropriate pins?

gecastan, 2020-09-29:
What pins are you referencing to? 
- Hardware and connection : Setup is working fine with the example.
- Config tuning :  If you're refering to some tuning to the python config files, it was my understanding that since it was an already configured and available device from the dropdown list, no config was required.
Maybe I missed a step in the setup. 
Documentation is quite nice but I m having hard time debugging to test a specific device with the custom python environment.

gecastan, 2020-09-30:
Even when deactivating the device from the UI, it was polling for measurements all the time.
I removed the device and restarted the whole since to get some proper logs and it turns out it cannot find the "ADS1256_definitions' module

Full output from :

```
2020-10-01 00:46:14,235 - INFO - mycodo.daemon - Mycodo daemon v8.8.3 starting
2020-10-01 00:46:14,866 - INFO - mycodo.controllers.controller_output - Activated in 226.7 ms
2020-10-01 00:46:15,367 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-10-01 00:46:15,367 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-10-01 00:46:15,368 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-10-01 00:46:15,368 - INFO - mycodo.daemon - All activated Input controllers started
2020-10-01 00:46:15,369 - INFO - mycodo.daemon - All activated Math controllers started
2020-10-01 00:46:15,369 - INFO - mycodo.daemon - All activated PID controllers started
2020-10-01 00:46:15,369 - INFO - mycodo.daemon - All activated LCD controllers started
2020-10-01 00:46:15,369 - INFO - mycodo.daemon - All activated Custom controllers started
2020-10-01 00:46:15,720 - INFO - mycodo.controllers.controller_widget - Activated in 349.9 ms
2020-10-01 00:46:16,722 - INFO - mycodo.daemon - Mycodo daemon started in 2.485 seconds
2020-10-01 00:46:16,729 - INFO - mycodo.daemon - 38.19 MB RAM in use
2020-10-01 00:47:51,787 - ERROR - mycodo.controllers.controller_input_42b371bc - initialize_variables() Exception: No module named 'ADS1256_definitions'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 338, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/var/opt/hosting/app/Mycodo/mycodo/inputs/ads1256.py", line 106, in __init__
    self.initialize_input()
  File "/var/opt/hosting/app/Mycodo/mycodo/inputs/ads1256.py", line 109, in initialize_input
    from ADS1256_definitions import POS_AIN0
ModuleNotFoundError: No module named 'ADS1256_definitions'
2020-10-01 00:47:51,798 - INFO - mycodo.controllers.controller_input_42b371bc - Activated in 328.2 ms
2020-10-01 00:47:51,798 - ERROR - mycodo.controllers.controller_input_42b371bc - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-10-01 00:48:06,725 - ERROR - mycodo.controllers.controller_input_42b371bc - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
```

kizniche, 2020-09-30:
```ModuleNotFoundError: No module named 'ADS1256_definitions'```

Here is the issue. I will investigate.

kizniche, 2020-10-01:
I updated both [PiPyADC-py3](https://github.com/kizniche/PiPyADC-py3) and [mycodo/inputs/ads1256.py](https://github.com/kizniche/Mycodo/commit/6353d185c86d69ce07f945085e7a0be42d8e1db2#diff-86e714670551a553a417bddfdfb0f8f0). You'll need to upgrade your python package. The easiest way would be to uninstall the current version from the terminal and let Mycodo install the new version:

```
~/Mycodo/env/bin/pip uninstall PiPyADC-Py3
```

Then restart the frontend from the web UI Config -> Restart Frontend menu. Then go to the Data page and add a new ADS1x15 Input and it should prompt you to install the new library.

kizniche, 2020-10-01:
I also see the original repository I forked and ported to Python 3 is not Python 3-compatible: https://github.com/ul-gh/PiPyADC

gecastan, 2020-10-01:
Thanks for the quick fix i ll test this out tonight

gecastan, 2020-10-01:
Hey again,

I proceeded as instructed but still no luck :
* Removed the device
* Uninstalled the previous PiPyADC-Py3 module
* Updated Mycodo/mycodo/inputs/ads1256.py
* Restarted the frontend
* Installed requirements (PiPyADC fetched from your updated master branch)
* Installed Waveshare ADS1256 device which prompted me to install PiPyADC
* Still unable to load module

Dependency logs:
```
[2020-10-01 17:50:30] Dependency installation beginning. Installing: pipyadc_py3

[2020-10-01 17:50:30] 
[2020-10-01 17:50:30] #### Installing/updating git+https://github.com/kizniche/PiPyADC-py3.git (pip-pypi)
[2020-10-01 17:50:33] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2020-10-01 17:50:33] Collecting git+https://github.com/kizniche/PiPyADC-py3.git
[2020-10-01 17:50:33]   Cloning https://github.com/kizniche/PiPyADC-py3.git to /tmp/pip-req-build-a5l7gc1a
[2020-10-01 17:50:37] Building wheels for collected packages: PiPyADC-py3
[2020-10-01 17:50:37]   Building wheel for PiPyADC-py3 (setup.py): started
[2020-10-01 17:50:39]   Building wheel for PiPyADC-py3 (setup.py): finished with status 'done'
[2020-10-01 17:50:39]   Created wheel for PiPyADC-py3: filename=PiPyADC_py3-1.2-py3-none-any.whl size=28099 sha256=e8ad9fb5000d4056c729eb0b4333bf2c81c9a5ac57d4d430c9990d1fcb4298e2
[2020-10-01 17:50:39]   Stored in directory: /tmp/pip-ephem-wheel-cache-5x2mhqpl/wheels/15/29/fc/86419861e1a22f210bf9739870e3df210b047315de3f6c5d51
[2020-10-01 17:50:39] Successfully built PiPyADC-py3
[2020-10-01 17:50:43] Installing collected packages: PiPyADC-py3
[2020-10-01 17:50:43] Successfully installed PiPyADC-py3-1.2

[2020-10-01 17:50:43] End install of pipyadc_py3

[2020-10-01 17:50:44] 
[2020-10-01 17:50:44] #### Setting permissions

[2020-10-01 17:50:46] #### All Dependencies have been installed.
```
Daemon logs:
```
2020-10-01 17:50:55,672 - INFO - mycodo.daemon - Mycodo daemon v8.8.3 starting
2020-10-01 17:50:56,291 - INFO - mycodo.controllers.controller_output - Activated in 200.6 ms
2020-10-01 17:50:56,793 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-10-01 17:50:56,793 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-10-01 17:50:56,794 - INFO - mycodo.daemon - All activated Input controllers started
2020-10-01 17:50:56,794 - INFO - mycodo.daemon - All activated Math controllers started
2020-10-01 17:50:56,794 - INFO - mycodo.daemon - All activated PID controllers started
2020-10-01 17:50:56,794 - INFO - mycodo.daemon - All activated LCD controllers started
2020-10-01 17:50:56,795 - INFO - mycodo.daemon - All activated Custom controllers started
2020-10-01 17:50:57,164 - INFO - mycodo.controllers.controller_widget - Activated in 369.0 ms
2020-10-01 17:50:58,166 - INFO - mycodo.daemon - Mycodo daemon started in 2.487 seconds
2020-10-01 17:50:58,169 - INFO - mycodo.daemon - 38.40 MB RAM in use
2020-10-01 17:52:24,279 - ERROR - mycodo.controllers.controller_input_ae22b2e4 - initialize_variables() Exception: No module named 'ADS1256_definitions'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 338, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/var/opt/hosting/app/Mycodo/mycodo/inputs/ads1256.py", line 106, in __init__
    self.initialize_input()
  File "/var/opt/hosting/app/Mycodo/mycodo/inputs/ads1256.py", line 111, in initialize_input
    from ADS1256_definitions import POS_AIN0
ModuleNotFoundError: No module named 'ADS1256_definitions'
2020-10-01 17:52:24,286 - INFO - mycodo.controllers.controller_input_ae22b2e4 - Activated in 357.0 ms
2020-10-01 17:52:24,287 - ERROR - mycodo.controllers.controller_input_ae22b2e4 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 391, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
```

My updated "Mycodo/mycodo/inputs/ads1256.py" 
```
    def initialize_input(self):
        import glob
        from pipyadc_py3 import ADS1256
        from ADS1256_definitions import POS_AIN0
        from pipyadc_py3.ADS1256_definitions import POS_AIN1
        from pipyadc_py3.ADS1256_definitions import POS_AIN2
        from pipyadc_py3.ADS1256_definitions import POS_AIN3
        from pipyadc_py3.ADS1256_definitions import POS_AIN4
        from pipyadc_py3.ADS1256_definitions import POS_AIN5
        from pipyadc_py3.ADS1256_definitions import POS_AIN6
        from pipyadc_py3.ADS1256_definitions import POS_AIN7
        from pipyadc_py3.ADS1256_definitions import NEG_AINCOM

        # Input pin for the potentiometer on the Waveshare Precision ADC board
        POTI = POS_AIN0 | NEG_AINCOM

        # Light dependant resistor
        LDR = POS_AIN1 | NEG_AINCOM
```

gecastan, 2020-10-01:
Well, silly me, spotted the error.
Turns out you missed a line :-)
spotted it and fixed it manually it's now working properly.
I ll make a pull request 

Switch 
        from ADS1256_definitions import POS_AIN0
to
        from pipyadc_py3.ADS1256_definitions import POS_AIN0

I m now getting proper outputs.

Cheers mate

closed 2020-10-02 by kizniche