kizniche/Mycodo #855: Selectable I2C address resetting

David-OConnor, 2020-09-29:
### Describe the problem/bug

Setting the I2C address does not take effect, and resets upon Activation. For 

### Versions:

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Setup -> Data -> Input -> Anyleaf: AnyLeaf pH (Or any other input with I2C selectable address). + -> Change the Address field. Activate. The sensor will read from the default address, and the selection box will reset to that. This happens regardless of the value of the `i2c_address_editable` field.

### Expected behavior

The address stays where you set it, and the Pi reads that I2C address.

kizniche, 2020-09-29:
Did you press the Save button?

David-OConnor, 2020-09-29:
Oops - looks like that was it! Working on my end. Going to pass words, and confirm it's working for the initial reporter!

David-OConnor, 2020-09-29:
Sorted. Thank you!

closed 2020-09-29 by David-OConnor

kizniche, 2020-09-29:
You're welcome. My easiest issue all week 😅