kizniche/Mycodo #856: Delete Custom Input fails

kkillion43, 2020-09-30:
I have been using the sensor for about a month now and thought it would be awesome to get it going on mycodo (very thankful for this opensource!).

I built a custom input and had to adjust the **Input Information** dict and **measurements_dict**

So doing this and reimporting the custom input, I would get an Error:

**Error: Import Input: 'input_name_unique' is not unique, there is already an input with that name (5n1_rs485_sensor)**
...and couldn't change it despite the effort

Then I tried to delete the imported module in the browser and would get this error. Thankful since it tried to delete the file and not just in the database. It's weird though, bc the file is there and path is correct.

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 639, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_settings.py", line 174, in settings_input
    utils_settings.settings_input_delete(form_input_delete)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_settings.py", line 642, in settings_input_delete
    os.remove(full_path_file)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/inputs/custom_inputs/5n1_rs485_sensor.py'

kizniche, 2020-09-30:
I'll see if I can replicate this behavior.

kizniche, 2020-10-08:
I'm trying to understand the issue. You first imported an input module. You then tried to import another with the name unique name and it prevented you from importing it. You then tried to remove the first import module to be able to import the second module with the same unique name, and it wouldn't let you remove the first imported module? Is that correct? You're currently stuck with an imported module you can't remove from the web interface?

kkillion43, 2020-10-08:
Hey Kyle,

Thanks for taking the time.

I first imported the input module and then realized I needed to change the "measurements_dict" of pH and NPK units to ion_concentration.

So, I changed the measurements_dict to correct it. 

I felt like I needed to reimport the updated input module and it gave me unique ID not so unique. So I tried to delete and reimport but wasn't able.

kizniche, 2020-10-08:
Can you provide the file that is unable to be deleted? You can't attach python files, but you can rename it to txt and attach it.

kkillion43, 2020-10-08:
Absolutely. I used your K30 example as boiler plate. 

[custom_input_file.txt](https://github.com/kizniche/Mycodo/files/5348302/custom_input_file.txt)

kizniche, 2020-10-10:
Is this the exact input you have imported? I ask because I am prevented from importing it into my system, due to syntax error:

```
Error: Import Input: Could not load uploaded file as a python module:
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_settings.py", line 514, in settings_input_import
    input_info = load_module_from_file(full_path_tmp, 'inputs')
  File "/home/pi/Mycodo/mycodo/utils/modules.py", line 17, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 860, in get_code
  File "<frozen importlib._bootstrap_external>", line 791, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/inputs/tmp_inputs/tmp_input_testing.py", line 202
    def initialize_input(self):
                              ^
IndentationError: unindent does not match any outer indentation level
```

kkillion43, 2020-10-10:
Hey Kyle ... thanks for getting back. 

That was a typo error on my part when saving. I had that part uncommented/commented since I do my own init of the sensor. I wasn't sure if myCodo was looking for an **initialize_input** function like in testing so I just `pass`. 

In the code, you just have to delete one space before the `def initialize_input(self)` accounting for the comment hashtag.

Let me know if this one works,
[custom_input_file.txt](https://github.com/kizniche/Mycodo/files/5359641/custom_input_file.txt)

kizniche, 2020-10-10:
I would like the exact input module you currently have imported in your system, without any modification. SFTP download and only change extension to txt.

kkillion43, 2020-10-10:
This is the file,
[custom_input_file.txt](https://github.com/kizniche/Mycodo/files/5359769/custom_input_file.txt)

kizniche, 2020-10-20:
I'm still unable to import. This is the traceback:

```
Error: Import Input: Could not load uploaded file as a python module:
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_settings.py", line 514, in settings_input_import
    input_info = load_module_from_file(full_path_tmp, 'inputs')
  File "/home/pi/Mycodo/mycodo/utils/modules.py", line 17, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/inputs/tmp_inputs/tmp_input_testing.py", line 18, in <module>
    import minimalmodbus as mm
ModuleNotFoundError: No module named 'minimalmodbus'
```

It appears you're importing modules that have not yet been installed to the virtualenvironment. All dependencies that need to be installed for the input should be included in "dependencies_module", and imports should occur within a function or class that doesn't get executed until after the Input is successfully added (and the dependencies are automatically installed). I needed to modify your Input to fix that import error.

After resolving that issue, I am able to import the Input Module, add the Input, save settings for the Input, delete the Input, and delete the Input Module without any issue.

kkillion43, 2020-10-20:
Thanks Kyle. I very much appreciate you taking the time to go through that process.

This is really interesting and got me thinking.

I am curious and wanted to see what you thought about running a **pipreqs** on the **custom_inputs** location within the venv prior to the Input being successfully added. You could run through that list of requirements not already satisfied or just
see if `pip install -r requirements.txt` for the user works out. 

This could maybe preserve the more native import everyone is accustomed to.

A Pseudo Example, 
[image]

closed 2020-10-20 by kkillion43

kizniche, 2020-10-21:
There are a few reasons I set it up this way:

1. The current pytests don't have a virtualenv with every single package required for every Input to run, as this would require a massive amount of bandwidth and processing every time I wanted to run tests and every time I committed code (TravisCI does it automatically).
2. There is also the apt repository that dependencies can be installed from, so the user has the option to use either pypi or apt.
3. The user may not want to import all modules at the top of the file.

It's a slightly more complex system but it provides more flexibility, specifically because it permits users to use apt.