kizniche/Mycodo #857: Low cost control of chemical metering with steeper motor pumps

alonrab, 2020-09-30:
Feature: Use stepper motor peristaltic pumps seem less costly than DC brush/brushless motor and give much longer motor life/ high presicion
Also It is needed to mechanically open close farm equipment levers/valves 

The pump can use any  bipolar stepper motor with controller. Although some nicer controllers seem to use I2C, the simplest setup use GPIO. 
 
Can it be added to the MyCodo system so a number of pumps can be operated with conditional function? I will use it for pH control and chemical metering.

The stepper motor use Polulu DRV 8825 controller

Function 1: turn pump on/off 
Function 2: Ramp pump speed
Function 3: use step count to do a specific amount of turns (e.g. half a turn to the right)

Problem encountered: the stepper controller get really hot so need to put the SLEEP GPIO off when not in use
[https://www.pololu.com/product/2987/pictures]
[](https://www.pololu.com/product/2987/pictures)

[image]
[image]

The code example I used was made by:
https://www.rototron.info/raspberry-pi-stepper-motor-tutorial/

Many thanks!

kizniche, 2020-09-30:
I've wanted to add stepper motor support to Mycodo for a while, but I never had a motor/board to test with. I'll take a look at the code you provided.

kizniche, 2020-12-13:
I believe I found a library that does pretty much the exact same thing as the linked one: https://pypi.org/project/rpi-python-drv8825/

Though I think I found an issue with their library:

https://github.com/dimschlukas/rpi_python_drv8825/blob/14e888f3947dc5a2c8821ca988485fba33f3808a/rpi_python_drv8825/stepper.py#L28

This constant 0.005 should in fact be set from fullstep_delay when the class is initialized:

https://github.com/dimschlukas/rpi_python_drv8825/blob/14e888f3947dc5a2c8821ca988485fba33f3808a/rpi_python_drv8825/stepper.py#L5

I just pushed code with a new stepper motor output. It's currently untested, but if you would like to test, you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to try it out. I would only test whether it moves clockwise (positive value) or counter clockwise (negative value) when entering a value in the Set Value field when you expand the output. If it works, and if you can confirm that the fullstep_delay option does not work as I suspect because it was mistakenly set as a constant rather than from the value passed to the class, I can just put the class within the Mycodo output module.

alonrab, 2020-12-14:
Got it , I'm away from the lab and my stepper pumps till early Jan due to
the covid mess but will test and update then.
Thanks!

kizniche, 2020-12-14:
I just bought a motor and driver so I can start experimenting myself. I also think it would be nice to be able to have output constrained within bounds, for things like duct vents (e.g. 0 - 360 degree value), track movement (0 - 1000 steps from end to end), etc. Currently this logic needs to be done by the user in a function/Conditional, but it would be nice to integrate it into a PID, for instance.

Magnum-Pl, 2020-12-15:
I just tried the stepper output and I got this error when trying to send a value.

`2020-12-15 22:12:14,950 - INFO - mycodo.outputs.stepper_ drv8825_62be8638 - Initialized in 59.1 ms
2020-12-15 22:12:34,286 - ERROR - mycodo.daemon - Could not turn output on: 'float' object cannot be interpreted as an integer
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 738, in output_on
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 286, in output_on_off
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/outputs/base_output.py", line 258, in output_on_off
    output_channel=output_channel)
  File "/home/pi/Mycodo/mycodo/outputs/stepper_ drv8825.py", line 210, in output_switch
    self.stepper.run(amount, True)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpi_python_drv8825/stepper.py", line 36, in run
    for i in range(steps):
TypeError: 'float' object cannot be interpreted as an integer`

kizniche, 2020-12-16:
I just pushed a fix that casts the amount as an integer.

Magnum-Pl, 2020-12-16:
After updating, I'm able to send a positive amount without getting the error, but only if I remove the Mode 1, 2 & 3 pins. This may be because I'm using the A4988 stepper instead of DRV8825, but their pinouts are identical and I think they can be used interchangeably in most cases. One difference I noticed between the two are the microstep resolutions. The DRV8825 has three settings for 1/32 that the A4988 doesn't have. The only resolution this affects on the A4988 is 1/16, so all the other resolutions seem to work fine as long as I don't have those microstep pins connected. The speed still changes when I change the resolution in the output options even with the resolution pins disconnected. 

[This page](https://lastminuteengineers.com/a4988-stepper-motor-driver-arduino-tutorial/) has tables for the resolutions of both drivers. 

If I send a negative amount, I get a red notification that says "object - Object", but no error in the log.
[image]

kizniche, 2020-12-16:
I just committed a fix for sending negative values. Can you check with ```gpio readall``` that your mode pins are set to output and are set correctly HIGH/LOW based on how you configured them in the output module?

Magnum-Pl, 2020-12-20:
That fixed it. Im able to turn the motor in either direction by sending positive or negative values. The mode pins are all set to output and they are set correctly for all the step resolution options. Changing the step resolution changes the number of steps required for one revolution. The torque is a lot higher than I was able to get while using an Arduino. 

I tried this with the mod pins disconnected and the step resolution changes the speed instead of the resolution. Can you add an option to adjust the speed?

kizniche, 2020-12-20:
The Full Step Delay allows you to change the step delay, which determines the speed.

kizniche, 2020-12-20:
For instance, changing it from 0.005 to 0.0025 will double the speed.

kizniche, 2020-12-20:
I hooked up my new stepper motor today and got to do some testing. I found it was less accurate the faster it was set. Perhaps this is because of the poor accuracy of millisecond timing with Python and/or Pi, and an arduino would serve it better for smaller step delays.

kizniche, 2020-12-22:
I just pushed an update to the stepper motor driver that added a bunch of new options, but I also renamed the unique name, so if you wish to upgrade to master to test the new code, you will first need to delete any stepper motor outputs you currently have, otherwise strange issues may present themselves.

kizniche, 2020-12-22:
[image]

Magnum-Pl, 2021-01-05:
Hey, Kyle. I finally got around to testing the new stepper options and I got an error when trying to add the stepper output. The output page gives me this error: 

`Error: Add Output: Daemon response: add_mod_output() Error: 6da6d065-cd54-47a8-9676-bc14c95ec1c8: name 'db_retrieve_table_daemon' is not defined`

These errors are from the daemon:

`ERROR - mycodo.controllers.controller_output - Could not initialize output 47a9a58b-0aae-4d8b-846e-b8a0819a1250
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 144, in all_outputs_initialize
    self.output[each_output.unique_id] = output_loaded.OutputModule(each_output)
  File "/home/pi/Mycodo-master/mycodo/outputs/stepper_bipolar_generic.py", line 216, in __init__
    output_channels = db_retrieve_table_daemon(
NameError: name 'db_retrieve_table_daemon' is not defined`

`ERROR - mycodo.controllers.controller_output - Output 47a9a58b-0aae-4d8b-846e-b8a0819a1250 not found`

kizniche, 2021-01-05:
That issue must have been introduced the last time I updated the module. I haven't tested it since then. I'll push a fix shortly.

kizniche, 2021-01-06:
Just pushed a fix that should resolve the import issues

Magnum-Pl, 2021-01-15:
Everything works great now. Thanks

closed 2021-03-09 by kizniche