kizniche/Mycodo #859: Minimum Off Duration not working

frodus17, 2020-10-01:
### Describe the problem/bug

I have the minimum off duration set to 600 (60s x 10 minutes = 600 seconds). Under no circumstances is it followed. It's mostly 2-3 minutes and it's costing me a lot in electricity to constantly run the chamber and is not great to cycle the compressor so often. I want it to remain off for 10 minutes. It used to work, until I upgraded from a previous version. I don't recall which one. It turns things on and off and regulates right, but something seems like it broke.

### Versions:

 - Mycodo Version: 8.8.4
 - Raspberry Pi Version: 3B+ I think
 - Raspbian OS Version: Buster

### Reproducibility

I let the PID run. Set the min off time to 600. It varies between 2-6 minutes most times (120-360s)

### Expected behavior

It should honor the minimum off time. It wants to constantly kick on/off the compressor. Sometimes because of the length of time the PID loop calculates, it turns on almost right after it turns off. It doesn't seem to be taking this value into consideration. It looks like it's only looking at min on time, and immediately turning on if it thinks it needs to.

### Screenshots

[image]

[image]

### Additional context

Is there anything that should be added to make it easier to address this issue?

frodus17, 2020-10-01:
Also, I'm not sure why there are some blips showing on the on/off(GPIO) duration in seconds. It isn't pulsing on for 5 seconds for 5 times before it turns on for longer. It's on during that entire time around 20:48 for instance.

kizniche, 2020-10-01:
Have you physically verified it is actually turning on and off and not just continually staying on, or are you basing this only on the graph?

kizniche, 2020-10-01:
I'll test tomorrow. It also looks like you cut off the duration y-axis so I can't interpret the graph too well.

frodus17, 2020-10-01:
I added the graph.
Yes, I physically verified it's on when it turns the output on, except the graph has these blips, which it doesn't do physically. It's on from the first blip to the last long vertical red line.

kizniche, 2020-10-01:
I was able to replicate this behavior. I'm investigating the cause/fix now.

kizniche, 2020-10-02:
I just found the issue and will be pushing a fix and making a release shortly.

```
2020-10-01 19:58:14,888 - INFO - mycodo.controllers.controller_pid_5656e3ef - PID Settings: Device ID: 3ca3da6e-aa23-4144-bea3-323da90c529e, Measurement ID: 36a77284-c4d6-4b99-bf25-a8c6a156e153, Direction: raise, Period: 30.0, Setpoint: 150.0, Band: 0.0, Kp: 1.0, Ki: 0.0, Kd: 0.0, Integrator Min: -100.0, Integrator Max 100.0, Output Raise: 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900, Output Raise Channel: 0, Output Raise Type: on_off, Output Raise Min On: 0.0, Output Raise Max On: 0.0, Output Raise Min Off: 120.0, Output Raise Always Min: False, Output Lower: None, Output Lower Channel: None, Output Lower Type: None, Output Lower Min On: 0.0, Output Lower Max On: 0.0, Output Lower Min Off: 0.0, Output Lower Always Min: False, Setpoint Tracking Type: , Setpoint Tracking ID: 
2020-10-01 19:58:14,889 - INFO - mycodo.controllers.controller_pid_5656e3ef - Activated in 132.9 ms
2020-10-01 19:58:15,976 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - Latest (CH0, Unit: F): 140.2664 @ 2020-10-01 19:57:41
2020-10-01 19:58:15,977 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - PID: Input: 140.2664, Output: P: 9.733599999999996, I: 0.0, D: 0.0, Out: 9.733599999999996
2020-10-01 19:58:16,276 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - Setpoint: 150.0 Output: 9.733599999999996 to output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0
2020-10-01 19:58:16,286 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - output_on_off(on, 0, sec, 9.73, 120.0, True)
2020-10-01 19:58:16,300 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - Output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0 (GPIO 20) on for 9.7 seconds. Output returned: success
2020-10-01 19:58:26,070 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - output_on_off(off, 0, None, 0.0, 0.0, True)
2020-10-01 19:58:26,071 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - Output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0 (GPIO 20) OFF at 2020-10-01 19:58:26. Output returned: success
2020-10-01 19:58:46,017 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - Latest (CH0, Unit: F): 140.2664 @ 2020-10-01 19:57:41
2020-10-01 19:58:46,018 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - PID: Input: 140.2664, Output: P: 9.733599999999996, I: 0.0, D: 0.0, Out: 9.733599999999996
2020-10-01 19:58:46,367 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - Setpoint: 150.0 Output: 9.733599999999996 to output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0
2020-10-01 19:58:46,377 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - output_on_off(on, 0, sec, 9.73, 120.0, True)
2020-10-01 19:58:46,377 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - Output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0 (GPIO 20) instructed to turn on, however the output has been instructed to stay off for 99.65 more seconds.
2020-10-01 19:59:16,025 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - Latest (CH0, Unit: F): 130.5824 @ 2020-10-01 19:59:11
2020-10-01 19:59:16,026 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - PID: Input: 130.5824, Output: P: 19.417599999999993, I: 0.0, D: 0.0, Out: 19.417599999999993
2020-10-01 19:59:16,270 - DEBUG - mycodo.controllers.controller_pid_5656e3ef - Setpoint: 150.0 Output: 19.417599999999993 to output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0
2020-10-01 19:59:16,281 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - output_on_off(on, 0, sec, 19.42, 120.0, True)
2020-10-01 19:59:16,282 - DEBUG - mycodo.outputs.on_off_gpio_3b1a0897 - Output 3b1a0897-b879-4ecc-a9f7-3f53d2fd7900 CH0 (GPIO 20) instructed to turn on, however the output has been instructed to stay off for 69.75 more seconds.
```

kizniche, 2020-10-02:
v8.8.5 just released.

Sorry about the issue. It was introduced in 8.8, where the output system was refactored to allow multi-channel outputs. I realize this is a significant bug that has the potential to cause damage, so I apologize. Test coverage is lacking and I sometimes don't catch bugs like these that don't immediately present themselves to me.

closed 2020-10-02 by kizniche

frodus17, 2020-10-02:
Kyle, Thank you, dang that was fast. I completely get it, one guy doing all the testing and coverage of all the bugs can get hard. I try to keep updates, so I can help you out where possible, but I haven't had the kegerator running while we updated our garage. Just brewed. I'm thinking I'll update the controls on the other chamber I have, so I get rid of the old system.