kizniche/Mycodo #861: wrong values ph/orp

ddralle, 2020-10-05:
Hello,
I've been tinkering for a few days now and unfortunately I can't get any further.

I use the ORP and PH interfaces from ATLAS via I2C.
Devices are found and can be controlled.

the values of the PH probe cannot be calibrated, totally wrong values come out when calibrating.

if you ventilate the temperature massively, you can influence the value, i think that's a mistake.

I have already exchanged the measuring probes, without success.

the same problem with the REDOX probe, totally wrong values.

self test, different hardware with ATLAS IOT, perfect values.

I already connected a 1WIRE temperature probe to test it, which also does not lead to success.

With the atlas IOT software, I noticed that the order of the 3 point calibration is different, maybe that's because of that
-> mycodo 4,7,10
-> atlas 7,4,10

what can be the reason?

kizniche, 2020-10-05:
It works for me. Can you provide more evidence to support your hypothesis?

kizniche, 2020-10-05:
> the order of the 3 point calibration is different

This could be an issue.

From the [datasheet](https://atlas-scientific.com/files/pH_EZO_Datasheet.pdf): "Issuing the cal,mid command after the EZO pH circuit has been calibrated will clear the other calibration points. Full calibration will have to be redone."

kizniche, 2020-10-05:
Your title mentions ORP yet you do not mention it in the issue as being an issue. Please explain.

I have changes to the calibration you can test, but I'll await your response about ORP.

kizniche, 2020-10-05:
Also, what does "ventilate the temperature" mean? If you're not using a temperature compensation measurement, temperature changes will alter the pH to a higher degree.

ddralle, 2020-10-05:
So I use the ORP - REDOX measurement.
this measurement doesn't fit at all, the pH measurement is much more important.

i use your tool for a garden pool.

at the moment it fails because of the PH.

as hardware -> "EZO pH Circuit" + probe (third party)

software -> ATLAS IOT -> example:
PH 4 test -> 4.02
PH 7 test -> 7.08
PH 10 test -> 7.04
-> calibrated -> perfect values (6.86, 9.01) test liquid

mycodo -> example:
PH 4 test -> 6.02
PH 7 test -> 9.14
PH 10 test -> 14.08
-> after calibration -> 6.68 - 10.01 // 9.01 - 14.45
Totally wrong values, the test liquid 4 is then rated at 6.02

that doesn't fit, I don't get the values properly calibrated or displayed, I've already tried a lot, I think that's a software problem

kizniche, 2020-10-05:
I changed the calibration order to mid, low, high. To test the new code, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch). Let me know if that changes anything.

kizniche, 2020-10-05:
Can you list the steps you took to add and calibrate the sensors? Be as detailed as possible.

ddralle, 2020-10-05:
yes i'm testing now

ddralle, 2020-10-05:
ok start testing:
8.8.5
calibration temp 20.5c
4.0 - 0.000
7.0 - 7,299
10.0-5805

master:
calibration temp 20.5c
7.0 - 2,793
4.0 - 3.957
10.0-10.008

second calibration 20.5c

7.0-7.029
4.0 - 4.040
10.0-10.040

reference:
6.86 - 7.009
9:00 am - 14,000

REDOX
468 - 441,600mV
650 - 627,000mV

I would say the values are better but cannot be used.

I noticed that the IOT software tests every second, does that make a difference?

In general, the PH measurement has become significantly better

kizniche, 2020-10-06:
All calibrations are stored on the EZO board chip. Calibration is merely giving the board a calibration command after the probe has acclimated to a particular pH solution (4, 7, 10). It's literally that simple. Measurements don't need to be returned to the user during calibration, but I coded that in to give feedback to indicate when the probe has acclimated to the solution. There can't possibly be anything different about Mycodo's calibration than any other software, as it's only sending a calibration command that indicates what solution the probe is currently in.

ddralle, 2020-10-06:
I had this assumption too.
so the calibration is saved on the chip.

i will do another test.

I start the IOT again and use it to calve and record the values.

after calibration, I start mycodo and then the live values must be identical.

I will report

ddralle, 2020-10-06:
ok i did the test, here are the results:

fresh start IOT ATLAS without calibration (calibration of mycodo)
4.00 - 8.81
7.00-7.02
9.00-14.00
10.00-14.00
6.86-7.08

calibration newly created with ATLAS software:
7.00 - 7.00
4.00-4.00
10.00-10.00

Test carried out with ATLAS software:
4.00-4.01
6.86-6.84
7.00-6.99
9.00 - 8.74
10.00-9.98

sd card moved,
fresh start MYCODO with calibration from IOT ATLAS

4.00-3.960
6.86-6.961
7.00 - 7.059
9.00 - 8.877
10.00-10.159

the atlas software is set to 25c by default (emulated value)
mycodo uses the 1wire thermometer according to software 20.5c
minimum values ​​can certainly be neglected, but i think the values ​​are still too imprecise.

What is striking about the ATLAS software is the renewal of the values ​​every second.

With the mycodo software, I initially had an interval of 15 seconds, so the probes are not at all reliable, much better with an interval of 1 second, it feels not yet optimal.

Does it help you?

more tests how can i help you?

kizniche, 2020-10-06:
I don't know what to tell you. There are no calculations occurring in Mycodo. It asks the chip for the pH and relays the value the cup returns to the user.

kizniche, 2020-10-06:
Your issue is probably better taken up with Atlas Scientific.

ddralle, 2020-10-06:
i'm the first, there has never been a problem like this

I know that, time for a beer :-)

can we somehow recreate that with a user?
do you agree with me that the values are not clean?

ok i will open a ticket for them, i will report,
thank you in advance for your help.

apart from this error, the software is really great!

kizniche, 2020-10-07:
I performed a calibration in Mycodo and took some measurements of my calibration solutions. I also cleared the calibration prior to calibrating. It appears to have calibrated successfully and the values appear accurate and consistent. I id not measure the calibration solution temperatures, so I was probably within 5C of the actual temperature. I also did not set up a temperature sensor to use as the compensation measurement. So, with that in mind, the very slight error could be due to the lack of proper temperature compensation or calibration temperature accuracy. Even without the temperature measurements, it seems my measurements are still very accurate.

```
timestamp (UTC),pH
2020-10-07 00:18:38.159,9.943
2020-10-07 00:19:05.090,9.953
2020-10-07 00:19:35.123,9.955
2020-10-07 00:20:05.038,9.956
2020-10-07 00:20:35.068,9.962
2020-10-07 00:21:05.096,9.963
2020-10-07 00:21:35.124,9.962
2020-10-07 00:22:05.038,9.961
2020-10-07 00:22:35.074,9.967
2020-10-07 00:23:05.090,9.965
2020-10-07 00:23:35.100,9.968
2020-10-07 00:24:05.113,9.967
2020-10-07 00:24:35.050,9.968
2020-10-07 00:25:05.103,9.968
2020-10-07 00:25:35.020,9.962
2020-10-07 00:26:05.045,9.966
2020-10-07 00:26:35.062,9.966
2020-10-07 00:27:05.086,9.964
2020-10-07 00:27:35.124,9.966
2020-10-07 00:28:05.036,10.164
2020-10-07 00:28:35.058,4.114
2020-10-07 00:29:05.090,4.083
2020-10-07 00:29:35.111,4.077
2020-10-07 00:30:05.036,4.071
2020-10-07 00:30:35.047,4.065
2020-10-07 00:31:05.070,4.062
2020-10-07 00:31:35.096,4.057
2020-10-07 00:32:05.123,4.057
2020-10-07 00:32:35.023,4.052
2020-10-07 00:33:05.030,4.055
2020-10-07 00:33:35.058,4.05
2020-10-07 00:34:05.066,4.05
2020-10-07 00:34:35.096,4.047
2020-10-07 00:35:05.098,4.047
2020-10-07 00:35:35.120,4.044
2020-10-07 00:36:05.060,6.866
2020-10-07 00:36:35.064,6.981
2020-10-07 00:37:05.078,6.983
2020-10-07 00:37:35.079,6.984
2020-10-07 00:38:05.123,6.982
2020-10-07 00:38:35.021,6.984
2020-10-07 00:39:05.096,6.984
2020-10-07 00:39:35.135,6.984
2020-10-07 00:40:05.068,6.982
2020-10-07 00:40:35.087,6.981
2020-10-07 00:41:05.103,6.981
2020-10-07 00:41:35.127,6.98
2020-10-07 00:42:05.053,6.981
2020-10-07 00:42:35.076,6.981
```

[image]

kizniche, 2020-10-07:
Perhaps the Atlas code is measuring every second and producing a rolling average. This can be mimicked in Mycodo with a Math controller that averages the past X seconds of measurements. You could also set the measurement Period very low, but I don't recommend this.

closed 2021-02-02 by kizniche