kizniche/Mycodo #863: Anyleaf ph, install dependencies

mircololi, 2020-10-07:
### Describe the problem/bug

Try to install the following  dependencies : "anyleaf "and "adafruit_extended_bus"

### Versions:

 - Mycodo Version:  8.8.5
 - Raspberry Pi Version: zero\4B
 - Raspbian OS Version: RASPBIAN LITE

Hi, first of all I want to thank you for this wonderful software!
I tried to install dependencies for "anyleaf ph" with no success in a raspberry ZERO W and 4B.
Below I send the report

### Screenshots

[2020-10-07 08:36:14] Dependency installation beginning. Installing: anyleaf,adafruit_extended_bus

[2020-10-07 08:36:14] 
[2020-10-07 08:36:14] #### Installing/updating anyleaf (pip-pypi)
[2020-10-07 08:36:16] Collecting anyleaf
[2020-10-07 08:36:16]   Using cached anyleaf-0.1.8.1-py3-none-any.whl (6.7 kB)
[2020-10-07 08:36:16] Collecting adafruit-circuitpython-max31865>=2.2.3
[2020-10-07 08:36:16]   Using cached adafruit-circuitpython-max31865-2.2.4.tar.gz (20 kB)
[2020-10-07 08:36:20] Collecting adafruit-circuitpython-ads1x15>=2.2.2
[2020-10-07 08:36:20]   Using cached adafruit-circuitpython-ads1x15-2.2.3.tar.gz (23 kB)
[2020-10-07 08:36:23] Collecting filterpy>=1.4.5
[2020-10-07 08:36:23]   Using cached filterpy-1.4.5.zip (177 kB)
[2020-10-07 08:36:25] Collecting Adafruit-Blinka
[2020-10-07 08:36:26]   Downloading Adafruit-Blinka-5.5.0.tar.gz (109 kB)
[2020-10-07 08:36:31] Collecting adafruit-circuitpython-busdevice
[2020-10-07 08:36:31]   Using cached adafruit-circuitpython-busdevice-5.0.1.tar.gz (22 kB)
[2020-10-07 08:36:34] Requirement already satisfied, skipping upgrade: numpy in /usr/lib/python3/dist-packages (from filterpy>=1.4.5->anyleaf) (1.16.2)
[2020-10-07 08:36:34] Requirement already satisfied, skipping upgrade: scipy in /usr/lib/python3/dist-packages (from filterpy>=1.4.5->anyleaf) (1.1.0)
[2020-10-07 08:36:35] Collecting matplotlib
[2020-10-07 08:36:35]   Using cached matplotlib-3.3.2.tar.gz (37.9 MB)
[2020-10-07 08:36:57] Collecting Adafruit-PlatformDetect>=2.18.0
[2020-10-07 08:36:57]   Downloading Adafruit-PlatformDetect-2.18.0.tar.gz (28 kB)
[2020-10-07 08:37:01] Collecting Adafruit-PureIO>=1.1.6
[2020-10-07 08:37:01]   Using cached Adafruit_PureIO-1.1.6.tar.gz (26 kB)
[2020-10-07 08:37:04] Collecting pyftdi>=0.40.0
[2020-10-07 08:37:04]   Using cached pyftdi-0.52.0-py3-none-any.whl (150 kB)
[2020-10-07 08:37:05] Collecting certifi>=2020.06.20
[2020-10-07 08:37:05]   Using cached certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
[2020-10-07 08:37:05] Collecting cycler>=0.10
[2020-10-07 08:37:05]   Using cached cycler-0.10.0-py2.py3-none-any.whl (6.5 kB)
[2020-10-07 08:37:05] Collecting kiwisolver>=1.0.1
[2020-10-07 08:37:05]   Using cached kiwisolver-1.2.0.tar.gz (52 kB)
[2020-10-07 08:37:09] Collecting pillow>=6.2.0
[2020-10-07 08:37:10]   Using cached Pillow-7.2.0.tar.gz (39.1 MB)
[2020-10-07 08:37:22] Requirement already satisfied, skipping upgrade: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in ./env/lib/python3.7/site-packages (from matplotlib->filterpy>=1.4.5->anyleaf) (2.4.7)
[2020-10-07 08:37:22] Requirement already satisfied, skipping upgrade: python-dateutil>=2.1 in ./env/lib/python3.7/site-packages (from matplotlib->filterpy>=1.4.5->anyleaf) (2.8.1)
[2020-10-07 08:37:24] Collecting pyusb>=1.0.0
[2020-10-07 08:37:24]   Using cached pyusb-1.1.0.tar.gz (78 kB)
[2020-10-07 08:37:24]   Installing build dependencies: started
[2020-10-07 08:37:25]   Installing build dependencies: finished with status 'error'
[2020-10-07 08:37:25]   ERROR: Command errored out with exit status 1:
[2020-10-07 08:37:25]    command: /home/pi/Mycodo/env/bin/python /home/pi/Mycodo/env/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-x_xkviu5/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=30.3.0' wheel setuptools_scm
[2020-10-07 08:37:25]        cwd: None
[2020-10-07 08:37:25]   Complete output (42 lines):
[2020-10-07 08:37:25]   Traceback (most recent call last):
[2020-10-07 08:37:25]     File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
[2020-10-07 08:37:25]       "__main__", mod_spec)
[2020-10-07 08:37:25]     File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
[2020-10-07 08:37:25]       exec(code, run_globals)
[2020-10-07 08:37:25]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/__main__.py", line 26, in <module>
[2020-10-07 08:37:25]       sys.exit(_main())
[2020-10-07 08:37:25]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/cli/main.py", line 73, in main
[2020-10-07 08:37:25]       command = create_command(cmd_name, isolated=("--isolated" in cmd_args))
[2020-10-07 08:37:25]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/commands/__init__.py", line 104, in create_command
[... 85 lines cut ...]
[2020-10-07 08:37:42]       from . import certs
[2020-10-07 08:37:42]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/requests/certs.py", line 15, in <module>
[2020-10-07 08:37:42]       from pip._vendor.certifi import where
[2020-10-07 08:37:42]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/certifi/__init__.py", line 1, in <module>
[2020-10-07 08:37:42]       from .core import contents, where
[2020-10-07 08:37:42]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/certifi/core.py", line 12, in <module>
[2020-10-07 08:37:42]       from importlib.resources import path as get_path, read_text
[2020-10-07 08:37:42]     File "/usr/lib/python3.7/importlib/resources.py", line 11, in <module>
[2020-10-07 08:37:42]       from typing import Iterable, Iterator, Optional, Set, Union   # noqa: F401
[2020-10-07 08:37:42]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/typing.py", line 1359, in <module>
[2020-10-07 08:37:42]       class Callable(extra=collections_abc.Callable, metaclass=CallableMeta):
[2020-10-07 08:37:42]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/typing.py", line 1007, in __new__
[2020-10-07 08:37:42]       self._abc_registry = extra._abc_registry
[2020-10-07 08:37:42]   AttributeError: type object 'Callable' has no attribute '_abc_registry'
[2020-10-07 08:37:42]   ----------------------------------------
[2020-10-07 08:37:42] ERROR: Command errored out with exit status 1: /home/pi/Mycodo/env/bin/python /home/pi/Mycodo/env/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-5vhs07ku/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=30.3.0' wheel setuptools_scm Check the logs for full command output.

[2020-10-07 08:37:43] End install of adafruit_extended_bus

[2020-10-07 08:37:43] 
[2020-10-07 08:37:43] #### Setting permissions

[2020-10-07 08:37:43] #### All Dependencies have been installed

Thank you for your help!

kizniche, 2020-10-07:
Try this:

```sudo ~/Mycodo/env/bin/pip3 uninstall typing```

Ref: https://github.com/ethereum/eth-abi/issues/131

mircololi, 2020-10-07:
I tried and it returns this message written in red:

WARNING:Skipping typing as it is not installed.

kizniche, 2020-10-07:
> [2020-10-07 08:37:42]     File "/home/pi/Mycodo/env/lib/python3.7/site-packages/typing.py", line 1007, in __new__
> [2020-10-07 08:37:42]       self._abc_registry = extra._abc_registry
> [2020-10-07 08:37:42]   AttributeError: type object 'Callable' has no attribute '_abc_registry'

I don't have /home/pi/Mycodo/env/lib/python3.7/site-packages/typing.py, so I can't replicate your issue. I'm not sure how you got that package in your environment, considering it's a built-in package, but that's what's causing your problem.

kizniche, 2020-10-07:
Delete and rebuild your environment:

```
sudo rm -rf ~/Mycodo/env
sudo bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh setup-virtualenv-full
sudo bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-dependencies
sudo service mycodo restart && sudo service mycodoflask restart
```

mircololi, 2020-10-08:
I tried with the commands but failed to fix. I reinstalled mycodo on the raspberry (zero), tried to add only anyleaf ph as input and nothing else.
While installing dependencies only "python3-numpy" and "python3-scipy" install correctly, while "anyleaf" and "adafruit_extended_bus" fail. Any ideas on how to fix this?

Thank you so much for this software and for your help!

[2020-10-08 15:27:57] Dependency installation beginning. Installing: python3-numpy,python3-scipy,anyleaf,adafruit_extended_bus

[2020-10-08 15:27:59] 
[2020-10-08 15:27:59] #### Installing/updating python3-numpy (apt)
[2020-10-08 15:28:05] Reading package lists...
[2020-10-08 15:28:08] Building dependency tree...
[2020-10-08 15:28:08] Reading state information...
[2020-10-08 15:28:11] The following additional packages will be installed:
[2020-10-08 15:28:11]   libblas3 libgfortran5 liblapack3
[2020-10-08 15:28:11] Suggested packages:
[2020-10-08 15:28:11]   gfortran python-numpy-doc python3-pytest python3-numpy-dbg
[2020-10-08 15:28:12] The following NEW packages will be installed:
[2020-10-08 15:28:12]   libblas3 libgfortran5 liblapack3 python3-numpy
[2020-10-08 15:28:18] 0 upgraded, 4 newly installed, 0 to remove and 21 not upgraded.
[2020-10-08 15:28:18] Need to get 3,676 kB of archives.
[2020-10-08 15:28:18] After this operation, 17.4 MB of additional disk space will be used.
[2020-10-08 15:28:18] Get:1 http://ftp.arnes.si/mirrors/raspbian/raspbian buster/main armhf libgfortran5 armhf 8.3.0-6+rpi1 [206 kB]
[2020-10-08 15:28:18] Get:2 http://raspbian.mirror.garr.it/mirrors/raspbian/raspbian buster/main armhf libblas3 armhf 3.8.0-2 [103 kB]
[2020-10-08 15:28:19] Get:4 http://ftp.arnes.si/mirrors/raspbian/raspbian buster/main armhf python3-numpy armhf 1:1.16.2-1 [1,873 kB]
[2020-10-08 15:28:21] Get:3 http://raspbian.mirror.garr.it/mirrors/raspbian/raspbian buster/main armhf liblapack3 armhf 3.8.0-2 [1,493 kB]
[2020-10-08 15:28:40] debconf: unable to initialize frontend: Dialog
[2020-10-08 15:28:40] debconf: (TERM is not set, so the dialog frontend is not usable.)
[2020-10-08 15:28:40] debconf: falling back to frontend: Readline
[2020-10-08 15:28:40] debconf: unable to initialize frontend: Readline
[2020-10-08 15:28:40] debconf: (This frontend requires a controlling tty.)
[2020-10-08 15:28:40] debconf: falling back to frontend: Teletype
[2020-10-08 15:28:40] dpkg-preconfigure: unable to re-open stdin: 
[2020-10-08 15:28:40] Fetched 3,676 kB in 20s (183 kB/s)
[2020-10-08 15:28:41] Selecting previously unselected package libgfortran5:armhf.
[2020-10-08 15:28:42] (Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 43793 files and directories currently installed.)
[2020-10-08 15:28:42] Preparing to unpack .../libgfortran5_8.3.0-6+rpi1_armhf.deb ...
[2020-10-08 15:28:42] Unpacking libgfortran5:armhf (8.3.0-6+rpi1) ...
[2020-10-08 15:28:43] Selecting previously unselected package libblas3:armhf.
[2020-10-08 15:28:43] Preparing to unpack .../libblas3_3.8.0-2_armhf.deb ...
[2020-10-08 15:28:43] Unpacking libblas3:armhf (3.8.0-2) ...
[2020-10-08 15:28:43] Selecting previously unselected package liblapack3:armhf.
[2020-10-08 15:28:43] Preparing to unpack .../liblapack3_3.8.0-2_armhf.deb ...
[2020-10-08 15:28:43] Unpacking liblapack3:armhf (3.8.0-2) ...
[2020-10-08 15:28:46] Selecting previously unselected package python3-numpy.
[2020-10-08 15:28:46] Preparing to unpack .../python3-numpy_1%3a1.16.2-1_armhf.deb ...
[2020-10-08 15:28:46] Unpacking python3-numpy (1:1.16.2-1) ...
[2020-10-08 15:28:50] Setting up libgfortran5:armhf (8.3.0-6+rpi1) ...
[2020-10-08 15:28:50] Setting up libblas3:armhf (3.8.0-2) ...
[2020-10-08 15:28:50] update-alternatives: using /usr/lib/arm-linux-gnueabihf/blas/libblas.so.3 to provide /usr/lib/arm-linux-gnueabihf/libblas.so.3 (libblas.so.3-arm-linux-gnueabihf) in auto mode
[... 220 lines cut ...]
[2020-10-08 15:44:00]     hashes=hashes,
[2020-10-08 15:44:00]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 130, in get_http_url
[2020-10-08 15:44:00]     link, downloader, temp_dir.path, hashes
[2020-10-08 15:44:00]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 282, in _download_http_url
[2020-10-08 15:44:00]     for chunk in download.chunks:
[2020-10-08 15:44:00]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/cli/progress_bars.py", line 168, in iter
[2020-10-08 15:44:00]     for x in it:
[2020-10-08 15:44:00]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/network/utils.py", line 88, in response_chunks
[2020-10-08 15:44:00]     decode_content=False,
[2020-10-08 15:44:00]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/urllib3/response.py", line 576, in stream
[2020-10-08 15:44:00]     data = self.read(amt=amt, decode_content=decode_content)
[... cut ...]
[2020-10-08 15:44:00]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/urllib3/response.py", line 442, in _error_catcher
[2020-10-08 15:44:00]     raise ReadTimeoutError(self._pool, None, "Read timed out.")
[2020-10-08 15:44:00] pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.

[2020-10-08 15:44:02] End install of adafruit_extended_bus

[2020-10-08 15:44:03] 
[2020-10-08 15:44:03] #### Setting permissions

[2020-10-08 15:44:08] #### All Dependencies have been installed.

kizniche, 2020-10-08:
I don't see your previous error, do you? I only see read timeouts, which could mean an issue with your internet connection or server you're trying to connect to.

mircololi, 2020-10-08:
Now the error is another. I tried to connect to a different network and tried again to install the two missing dependencies but without success.
What log can I send you to help understand the problem?

Sorry for the few details I can give you but I use google translate and I wouldn't want to write wrong!

[2020-10-08 17:03:11] Dependency installation beginning. Installing: anyleaf,adafruit_extended_bus

[2020-10-08 17:03:13] 
[2020-10-08 17:03:13] #### Installing/updating anyleaf (pip-pypi)
[2020-10-08 17:03:30] Collecting anyleaf
[2020-10-08 17:03:30]   Using cached anyleaf-0.1.8.1-py3-none-any.whl (6.7 kB)
[2020-10-08 17:03:31] Collecting adafruit-circuitpython-max31865>=2.2.3
[2020-10-08 17:03:32]   Using cached adafruit-circuitpython-max31865-2.2.4.tar.gz (20 kB)
[2020-10-08 17:04:00] Collecting filterpy>=1.4.5
[2020-10-08 17:04:00]   Using cached filterpy-1.4.5.zip (177 kB)
[2020-10-08 17:04:11] Collecting adafruit-circuitpython-ads1x15>=2.2.2
[2020-10-08 17:04:11]   Using cached adafruit-circuitpython-ads1x15-2.2.3.tar.gz (23 kB)
[2020-10-08 17:04:41] Collecting Adafruit-Blinka
[2020-10-08 17:04:41]   Using cached Adafruit-Blinka-5.5.1.tar.gz (110 kB)
[2020-10-08 17:05:13] Collecting adafruit-circuitpython-busdevice
[2020-10-08 17:05:13]   Using cached adafruit-circuitpython-busdevice-5.0.1.tar.gz (22 kB)
[2020-10-08 17:05:40] Requirement already satisfied, skipping upgrade: numpy in /usr/lib/python3/dist-packages (from filterpy>=1.4.5->anyleaf) (1.16.2)
[2020-10-08 17:05:40] Requirement already satisfied, skipping upgrade: scipy in /usr/lib/python3/dist-packages (from filterpy>=1.4.5->anyleaf) (1.1.0)
[2020-10-08 17:05:47] Collecting matplotlib
[2020-10-08 17:05:50]   Using cached matplotlib-3.3.2.tar.gz (37.9 MB)
[2020-10-08 17:07:33] Collecting Adafruit-PlatformDetect>=2.18.0
[2020-10-08 17:07:33]   Using cached Adafruit-PlatformDetect-2.18.0.tar.gz (28 kB)
[2020-10-08 17:08:01] Collecting Adafruit-PureIO>=1.1.6
[2020-10-08 17:08:01]   Using cached Adafruit_PureIO-1.1.6.tar.gz (26 kB)
[2020-10-08 17:08:30] Collecting pyftdi>=0.40.0
[2020-10-08 17:08:30]   Using cached pyftdi-0.52.0-py3-none-any.whl (150 kB)
[2020-10-08 17:08:31] Requirement already satisfied, skipping upgrade: RPi.GPIO in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-max31865>=2.2.3->anyleaf) (0.7.0)
[2020-10-08 17:08:32] Collecting rpi_ws281x>=4.0.0
[2020-10-08 17:08:32]   Using cached rpi_ws281x-4.2.4.tar.gz (60 kB)
[2020-10-08 17:08:42] Collecting sysv_ipc
[2020-10-08 17:08:42]   Using cached sysv_ipc-1.0.1.tar.gz (102 kB)
[2020-10-08 17:08:59] Collecting certifi>=2020.06.20
[2020-10-08 17:08:59]   Using cached certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
[2020-10-08 17:09:00] Collecting cycler>=0.10
[2020-10-08 17:09:00]   Using cached cycler-0.10.0-py2.py3-none-any.whl (6.5 kB)
[2020-10-08 17:09:03] Collecting kiwisolver>=1.0.1
[2020-10-08 17:09:03]   Using cached kiwisolver-1.2.0.tar.gz (52 kB)
[2020-10-08 17:09:40] Collecting pillow>=6.2.0
[2020-10-08 17:09:44]   Using cached Pillow-7.2.0.tar.gz (39.1 MB)
[2020-10-08 17:10:27] Requirement already satisfied, skipping upgrade: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in ./env/lib/python3.7/site-packages (from matplotlib->filterpy>=1.4.5->anyleaf) (2.4.7)
[2020-10-08 17:10:27] Requirement already satisfied, skipping upgrade: python-dateutil>=2.1 in ./env/lib/python3.7/site-packages (from matplotlib->filterpy>=1.4.5->anyleaf) (2.8.1)
[2020-10-08 17:10:27] Requirement already satisfied, skipping upgrade: pyserial>=3.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-max31865>=2.2.3->anyleaf) (3.4)
[2020-10-08 17:10:29] Collecting pyusb>=1.0.0
[2020-10-08 17:10:29]   Using cached pyusb-1.1.0.tar.gz (78 kB)
[2020-10-08 17:10:29]   Installing build dependencies: started
[2020-10-08 17:11:31]   Installing build dependencies: still running...
[2020-10-08 17:11:34]   Installing build dependencies: finished with status 'done'
[2020-10-08 17:11:34]   Getting requirements to build wheel: started
[2020-10-08 17:11:39]   Getting requirements to build wheel: finished with status 'done'
[2020-10-08 17:11:39] ERROR: Exception:
[2020-10-08 17:11:39] Traceback (most recent call last):
[2020-10-08 17:11:39]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/packaging/requirements.py", line 98, in __init__
[2020-10-08 17:11:39]     req = REQUIREMENT.parseString(requirement_string)
[2020-10-08 17:11:39]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pyparsing.py", line 1955, in parseString
[2020-10-08 17:11:39]     raise exc
[2020-10-08 17:11:39]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pyparsing.py", line 3250, in parseImpl
[2020-10-08 17:11:39]     raise ParseException(instring, loc, self.errmsg, self)
[2020-10-08 17:11:39] pip._vendor.pyparsing.ParseException: Expected W:(abcd...), found '_'  (at char 0), (line:1, col:1)
[2020-10-08 17:11:39] 
[2020-10-08 17:11:39] During handling of the above exception, another exception occurred:
[2020-10-08 17:11:39] 
[2020-10-08 17:11:39] Traceback (most recent call last):
[2020-10-08 17:11:39]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 3101, in __init__
[2020-10-08 17:11:39]     super(Requirement, self).__init__(requirement_string)
[... cut ...]
[2020-10-08 17:15:37]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 3103, in __init__
[2020-10-08 17:15:37]     raise RequirementParseError(str(e))
[2020-10-08 17:15:37] pip._vendor.pkg_resources.RequirementParseError: Parse error at "'_'": Expected W:(abcd...)

[2020-10-08 17:15:39] End install of adafruit_extended_bus

[2020-10-08 17:15:41] 
[2020-10-08 17:15:41] #### Setting permissions

[2020-10-08 17:15:46] #### All Dependencies have been installed.

AKAMEDIASYSTEM, 2020-10-11:
I'm encountering the same error trying to install dependency `adafruit_sgp30`.

Fresh 8.8.5 install on an unmodified RasPi 3B+ running the Raspbian LITE image from 8-20-20.

My dependency install fails at:

`[2020-10-11 21:14:52] #### Installing/updating adafruit-circuitpython-sgp30 (pip-pypi)
[2020-10-11 21:14:56] Collecting adafruit-circuitpython-sgp30
[2020-10-11 21:14:56]   Using cached adafruit-circuitpython-sgp30-2.3.1.tar.gz (20 kB)
[2020-10-11 21:15:05] Collecting Adafruit-Blinka
[2020-10-11 21:15:05]   Using cached Adafruit-Blinka-5.5.1.tar.gz (110 kB)
[2020-10-11 21:15:13] Collecting adafruit-circuitpython-busdevice
[2020-10-11 21:15:13]   Using cached adafruit-circuitpython-busdevice-5.0.1.tar.gz (22 kB)
[2020-10-11 21:15:22] Collecting Adafruit-PlatformDetect>=2.18.0
[2020-10-11 21:15:22]   Using cached Adafruit-PlatformDetect-2.18.0.tar.gz (28 kB)
[2020-10-11 21:15:32] Requirement already satisfied, skipping upgrade: Adafruit-PureIO>=1.1.6 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sgp30) (1.1.6)
[2020-10-11 21:15:32] Collecting pyftdi>=0.40.0
[2020-10-11 21:15:32]   Using cached pyftdi-0.52.0-py3-none-any.whl (150 kB)
[2020-10-11 21:15:32] Requirement already satisfied, skipping upgrade: RPi.GPIO in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sgp30) (0.7.0)
[2020-10-11 21:15:33] Collecting rpi_ws281x>=4.0.0
[2020-10-11 21:15:33]   Using cached rpi_ws281x-4.2.4-cp37-cp37m-linux_armv7l.whl (115 kB)
[2020-10-11 21:15:34] Collecting sysv_ipc
[2020-10-11 21:15:34]   Using cached sysv_ipc-1.0.1.tar.gz (102 kB)
[2020-10-11 21:15:39] Collecting pyusb>=1.0.0
[2020-10-11 21:15:39]   Using cached pyusb-1.1.0.tar.gz (78 kB)
[2020-10-11 21:15:39]   Installing build dependencies: started
[2020-10-11 21:15:57]   Installing build dependencies: finished with status 'done'
[2020-10-11 21:15:57]   Getting requirements to build wheel: started
[2020-10-11 21:15:59]   Getting requirements to build wheel: finished with status 'done'
[2020-10-11 21:15:59] ERROR: Exception:
[2020-10-11 21:15:59] Traceback (most recent call last):
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/packaging/requirements.py", line 98, in __init__
[2020-10-11 21:15:59]     req = REQUIREMENT.parseString(requirement_string)
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pyparsing.py", line 1955, in parseString
[2020-10-11 21:15:59]     raise exc
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pyparsing.py", line 3250, in parseImpl
[2020-10-11 21:15:59]     raise ParseException(instring, loc, self.errmsg, self)
[2020-10-11 21:15:59] pip._vendor.pyparsing.ParseException: Expected W:(abcd...), found '_'  (at char 0), (line:1, col:1)
[2020-10-11 21:15:59] 
[2020-10-11 21:15:59] During handling of the above exception, another exception occurred:
[2020-10-11 21:15:59] 
[2020-10-11 21:15:59] Traceback (most recent call last):
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 3101, in __init__
[2020-10-11 21:15:59]     super(Requirement, self).__init__(requirement_string)
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_vendor/packaging/requirements.py", line 102, in __init__
[2020-10-11 21:15:59]     requirement_string[e.loc : e.loc + 8], e.msg
[2020-10-11 21:15:59] pip._vendor.packaging.requirements.InvalidRequirement: Parse error at "'_'": Expected W:(abcd...)
[2020-10-11 21:15:59] 
[2020-10-11 21:15:59] During handling of the above exception, another exception occurred:
[2020-10-11 21:15:59] 
[2020-10-11 21:15:59] Traceback (most recent call last):
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 228, in _main
[2020-10-11 21:15:59]     status = self.run(options, args)
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 182, in wrapper
[2020-10-11 21:15:59]     return func(self, options, args)
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 324, in run
[2020-10-11 21:15:59]     reqs, check_supported_wheels=not options.target_dir
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 183, in resolve
[2020-10-11 21:15:59]     discovered_reqs.extend(self._resolve_one(requirement_set, req))
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 388, in _resolve_one
[2020-10-11 21:15:59]     abstract_dist = self._get_abstract_dist_for(req_to_install)
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 340, in _get_abstract_dist_for
[2020-10-11 21:15:59]     abstract_dist = self.preparer.prepare_linked_requirement(req)
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 483, in prepare_linked_requirement
[2020-10-11 21:15:59]     req, self.req_tracker, self.finder, self.build_isolation,
[2020-10-11 21:15:59]   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 91, in _get_prepared_distribution
[... cut ...]
...pip module `typing` i not installed, but it does seem likely a pip issue is the root cause.  (I also tried the reset-Mycodo's-environment fix mentioned in this thread with no change in outcome).

I attempted to roll `pip` back to 20.0.2, per [this](https://github.com/pypa/pip/issues/8272#issuecomment-631252067) issue, but experienced the same outcome (plus a warning about using an outdated version of pip)

Any ideas would be much appreciated - it appears to be a string-parsing issue, but without digging into the sourcecode for pip I'm not sure how best to proceed.

Thank you!

AKA

mircololi, 2020-10-12:
I tried to install raspbian desktop instead of rapsbian lite on raspberry 4 and the dependencies install correctly. I hope this information can be useful in some way ...... thanks again for Mycodo !!!!

2020-10-12 11:49:10] Dependency installation beginning. Installing: python3-scipy,anyleaf,adafruit_extended_bus

[2020-10-12 11:49:10] 
[2020-10-12 11:49:10] #### Installing/updating python3-scipy (apt)
[2020-10-12 11:49:11] Lettura elenco dei pacchetti...
[2020-10-12 11:49:12] Generazione albero delle dipendenze...
[2020-10-12 11:49:12] Lettura informazioni sullo stato...
[2020-10-12 11:49:13] I seguenti pacchetti sono stati installati automaticamente e non sono piÃ¹ richiesti:
[2020-10-12 11:49:13]   libjs-sphinxdoc libjs-underscore libpython-all-dev python-all python-all-dev
[2020-10-12 11:49:13]   python-asn1crypto python-configparser python-cookies python-crypto
[2020-10-12 11:49:13]   python-dbus python-entrypoints python-enum34 python-funcsigs python-jwt
[2020-10-12 11:49:13]   python-keyrings.alt python-mock python-pbr python-responses python-wheel
[2020-10-12 11:49:13]   python-xdg python3-asn1crypto python3-cookies python3-dbus
[2020-10-12 11:49:13]   python3-entrypoints python3-jwt python3-responses
[2020-10-12 11:49:13] Usare "apt autoremove" per rimuoverli.
[2020-10-12 11:49:13] I seguenti pacchetti aggiuntivi saranno inoltre installati:
[2020-10-12 11:49:13]   python3-decorator
[2020-10-12 11:49:13] Pacchetti suggeriti:
[2020-10-12 11:49:13]   python-scipy-doc
[2020-10-12 11:49:13] I seguenti pacchetti NUOVI saranno installati:
[2020-10-12 11:49:13]   python3-decorator python3-scipy
[2020-10-12 11:49:13] 0 aggiornati, 2 installati, 0 da rimuovere e 0 non aggiornati.
[2020-10-12 11:49:13] Ã necessario scaricare 8.925 kB di archivi.
[2020-10-12 11:49:13] Dopo quest'operazione, verranno occupati 38,0 MB di spazio su disco.
[2020-10-12 11:49:13] Scaricamento di:1 http://raspbian.mirror.garr.it/mirrors/raspbian/raspbian buster/main armhf python3-decorator all 4.3.0-1.1 [14,5 kB]
[2020-10-12 11:49:14] Scaricamento di:2 http://ftp.arnes.si/mirrors/raspbian/raspbian buster/main armhf python3-scipy armhf 1.1.0-7 [8.910 kB]
[2020-10-12 11:49:26] debconf: impossibile inizializzare il frontend: Dialog
[2020-10-12 11:49:26] debconf: (TERM non Ã¨ impostata per cui il frontend a dialoghi non puÃ² essere usato.)
[2020-10-12 11:49:26] debconf: sarÃ  usato il frontend: Readline
[2020-10-12 11:49:26] debconf: impossibile inizializzare il frontend: Readline
[2020-10-12 11:49:26] debconf: (Questo frontend richiede un terminale di controllo.)
[2020-10-12 11:49:26] debconf: sarÃ  usato il frontend: Teletype
[2020-10-12 11:49:26] dpkg-preconfigure: impossibile riaprire stdin: 
[2020-10-12 11:49:26] Recuperati 8.925 kB in 12s (755 kB/s)
[2020-10-12 11:49:26] Selezionato il pacchetto python3-decorator non precedentemente selezionato.
[2020-10-12 11:49:26] (Lettura del database... 
(Lettura del database... 5%
(Lettura del database... 10%
(Lettura del database... 15%
(Lettura del database... 20%
(Lettura del database... 25%
(Lettura del database... 30%
(Lettura del database... 35%
(Lettura del database... 40%
(Lettura del database... 45%
(Lettura del database... 50%
(Lettura del database... 55%
(Lettura del database... 60%
(Lettura del database... 65%
(Lettura del database... 70%
(Lettura del database... 75%
(Lettura del database... 80%
(Lettura del database... 85%
(Lettura del database... 90%
(Lettura del database... 95%
(Lettura del database... 100%
(Lettura del database... 93548 file e directory attualmente installati.)
[2020-10-12 11:49:26] Preparativi per estrarre .../python3-decorator_4.3.0-1.1_all.deb...
[2020-10-12 11:49:26] Estrazione di python3-decorator (4.3.0-1.1)...
[2020-10-12 11:49:27] Selezionato il pacchetto python3-scipy non precedentemente selezionato.
[2020-10-12 11:49:27] Preparativi per estrarre .../python3-scipy_1.1.0-7_armhf.deb...
[2020-10-12 11:49:27] Estrazione di python3-scipy (1.1.0-7)...
[2020-10-12 11:49:31] Configurazione di python3-decorator (4.3.0-1.1)...
[2020-10-12 11:49:31] Configurazione di python3-scipy (1.1.0-7)...

[2020-10-12 11:49:37] End install of python3-scipy

[2020-10-12 11:49:37] 
[... 54 lines cut ...]
[2020-10-12 11:50:27] End install of anyleaf

[2020-10-12 11:50:27] 
[2020-10-12 11:50:27] #### Installing/updating Adafruit_Extended_Bus (pip-pypi)
[2020-10-12 11:50:28] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2020-10-12 11:50:29] Collecting Adafruit_Extended_Bus
[2020-10-12 11:50:29]   Downloading https://www.piwheels.org/simple/adafruit-extended-bus/adafruit_extended_bus-1.0.0-py3-none-any.whl (4.9 kB)
[2020-10-12 11:50:29] Requirement already satisfied, skipping upgrade: Adafruit-Blinka in ./env/lib/python3.7/site-packages (from Adafruit_Extended_Bus) (5.5.1)
[2020-10-12 11:50:29] Requirement already satisfied, skipping upgrade: rpi-ws281x>=4.0.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->Adafruit_Extended_Bus) (4.2.4)
[2020-10-12 11:50:29] Requirement already satisfied, skipping upgrade: pyftdi>=0.40.0 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->Adafruit_Extended_Bus) (0.52.0)
[2020-10-12 11:50:29] Requirement already satisfied, skipping upgrade: Adafruit-PureIO>=1.1.6 in ./env/lib/python3.7/site-packages (from Adafruit-Blinka->Adafruit_Extended_Bus) (1.1.6)
[... cut ...]
[2020-10-12 11:50:29] Requirement already satisfied, skipping upgrade: pyusb>=1.0.0 in ./env/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->Adafruit_Extended_Bus) (1.1.0)
[2020-10-12 11:50:30] Installing collected packages: Adafruit-Extended-Bus
[2020-10-12 11:50:31] Successfully installed Adafruit-Extended-Bus-1.0.0

[2020-10-12 11:50:31] End install of adafruit_extended_bus

[2020-10-12 11:50:31] 
[2020-10-12 11:50:31] #### Setting permissions

[2020-10-12 11:50:32] #### All Dependencies have been installed.

David-OConnor, 2020-10-12:
I'm glad this worked out, and that's useful information about the Raspbian distro. I'm suspicious this is caused by the Lite version of Raspian missing some system dependencies that Scipy relies on. (And perhaps these are installable using apt, to prevent the need to install a new OS).

AKAMEDIASYSTEM, 2020-10-17:
Thanks for sharing the solution! Using the non-LITE Raspian distro fixed this issue for me as well.

kizniche, 2020-10-20:
All of my Raspbian installs are Lite, but they don't suffer from this issue, so it's likely something introduced recently. If I get some time, I'll install the Lite version and experiment to discover any conflicts/dependencies that aren't accounted for.

kizniche, 2020-12-20:
I've been trying to track down this issue, but haven't made any conclusions yet. However, I have found that I can install Adafruit_Extended_Bus to the Mycodo virtual env when I run ```pip install``` as a user, but it fails with the aforementioned error when it's run as root (user used when installing dependencies). I've never encountered this before and there doesn't appear to be an obvious issue. I'll keep looking for a solution, but if anyone has better knowledge than I about the workings of pip and can shed some light on this, let me know.

Magnum-Pl, 2020-12-20:
I had the same issue when trying to install Adafruit_Extended_Bus for the MCP4728 in [#893](https://github.com/kizniche/Mycodo/issues/893#issuecomment-743885972). I was able to get it to work after installing pyusb as a user.

kizniche, 2020-12-21:
I'm closer to finding the issue, but in the meantime, I found a fix for the several sensors that have pyusb as a dependency. I found that the latest version (1.1.0) won't install as root, but the next lower version (1.0.2) will. Since Adafruit_Extended_Bus needs >= 1.0.0, specifying version pyusb==1.0.2 in the dependency line seems to work.

The install will fail with this command:

```bash
sudo ~/Mycodo/env/bin/pip3 install pyusb==1.1.0
```

The install will succeed with this command:

```bash
sudo ~/Mycodo/env/bin/pip3 install pyusb==1.0.2
```

kizniche, 2020-12-21:
Alright, I might put this issue to bed. I'm not really getting anywhere with getting 1.1.0 installed. However, I did find apt has the package python3-usb, which I think might be what's installed by default on the Desktop but not the Lite version of the Raspberry Pi OS. If so, that may explain why this issue doesn't present itself on the Desktop version, as it installs pyusb 1.0.2.

kizniche, 2021-01-24:
The latest release of pyusb (1.1.1, Jan 20, 2021) seems to have fixed the install issue.

MILO-GM, 2021-01-24:
Thank you very much Kyle!

closed 2021-04-24 by kizniche