kizniche/Mycodo #864: Editing configs while the Input is active

David-OConnor, 2020-10-08:
**Is your feature request related to a problem? Please describe.**
Is there a technical reason we need to deactivate an input to edit the config?

**Describe the solution you'd like**
Being able to edit input settings (ie from the + dropdown menu) while a probe is active

**Additional context**
This would facilitate the troubleshooting and experimenting process. I'm suspicious there's a technical reason making this inadvisable. What do you think? Thank you.

kizniche, 2020-10-09:
Since inputs/sensors can have vastly different requirements for their initialization and operation, changing settings while they're running can cause a host of issues. Rather than allow only a subset to have setting changed while running, I made them all need to be disabled before settings can be changed.

kizniche, 2020-10-09:
There is also modifications that would need to be made to support changing settings while an input is running. Either:

1. Reinitilize input after changing settings, akin to deactivating and reactivating.
2. Lock conducting a measurement to prevent querying input settings while settings are modified (thereby preventing an operation from proceeding that may use partial settings from the pre-saved configuration and partial settings from post-save.

kizniche, 2020-10-09:
Depending on the approach taken, there may be additional issues such as changing the communication address (I2C address, for instance), that would necessitate reinitializing the sensor class, which is more conducive in a system that deactivated/activated the input (1) rather than merely pausing its operation (2).

David-OConnor, 2020-10-12:
Thank you for the detailed information! It sounds like the the most straightforward solution is to leave the existing behavior. It's a simple safety check, and trying to modify could introduce bugs.

closed 2020-10-12 by David-OConnor